USC-ACTLab/crazyswarm #147: [crazyswarm_server-1] process has died when using the hover_swarm.launch

lkyfly0765, 2019-04-06:
We use the optitrack as mocap, and we have configured the Motive correctly. We follow the instructions on _https://crazyswarm.readthedocs.io/en/latest/usage.html_ . The related yaml files has been changed correctly and the launch file has been configured as : 

   firmware: "bitcraze" # one of "crazyswarm", "bitcraze"
      broadcast_address: "E7E7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      # genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      # genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      # genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      # genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      # genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 1 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_local_ip: "169.254.106.255"
      optitrack_server_ip: "169.254.41.1:3883"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      # save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False

And then I start the launch file `eric@eric-ThinkPad-S2:~$ roslaunch crazyswarm hover_swarm.launch `

eric@eric-ThinkPad-S2:~$ roslaunch crazyswarm hover_swarm.launch 
... logging to /home/eric/.ros/log/7b882882-5764-11e9-a618-34f64bc13130/roslaunch-eric-ThinkPad-S2-6968.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://eric-ThinkPad-S2:35681/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
[... 118 lines cut ...]
 * /rosversion: 1.12.14

NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[crazyswarm_server-1]: started with pid [6985]
process[joy-2]: started with pid [6986]
process[crazyswarm_teleop-3]: started with pid [6987]
process[rviz-4]: started with pid [6994]
[ INFO] [1554442479.080369264]: Wait for services...
[ INFO] [1554442479.095308661]: Manager ready.
[ INFO] [1554442479.265023748]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
[crazyswarm_server-1] process has died [pid 6985, exit code -11, cmd /home/eric/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/eric/.ros/log/7b882882-5764-11e9-a618-34f64bc13130/crazyswarm_server-1.log].
log file: /home/eric/.ros/log/7b882882-5764-11e9-a618-34f64bc13130/crazyswarm_server-1*.log

Everything is ok, but it goes wrong at last:[crazyswarm_server-1] process has died 

What's wrong with it ? How to figure it out?
Thanks!

whoenig, 2019-04-07:
Looks like you are getting a segfault in the crazyswarm_server. To debug it, add "launch-prefix="xterm -e gdb --args"" to https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L8. In the xterm window, type and enter "r" (for run), and after the crash occurred type and enter "bt" to get a stacktrace on where the crash occurred.

It's likely related to some wrong/missing configuration that does not have proper error checking.

lkyfly0765, 2019-04-08:
> Looks like you are getting a segfault in the crazyswarm_server. To debug it, add "launch-prefix="xterm -e gdb --args"" to [...]

lkyfly0765, 2019-04-08:
Hi whoenig, I have tried GDB debug process, and it says 
[image]
What's wrong with it ?

whoenig, 2019-04-08:
Type "bt" and press enter to see a stack trace. If this does not contain line numbers, you'll need to recompile with debug symbols: "catkin_make clean" followed by "catkin_make -DCMAKE_BUILD_TYPE=Debug" in the ros_ws folder.

lkyfly0765, 2019-04-09:
Here is the stact trace.
[image]

lkyfly0765, 2019-04-09:
Also I am doubt about the `broadcast_address: "E7E7E7E7E7"` what should the broadcast_address to be set? Every CF's radio address is not same.

whoenig, 2019-04-10:
Which version/commit are you using? Those line numbers actually don't point to code in the latest master branch.

The broadcast address should not be changed from the default value (it's hard-coded in the firmware).

lkyfly0765, 2019-04-10:
I am using the latest version of the crazyswarm ros-package. We just change our package from Crazyflie_ros of your to crazyswarm because we are going to control more cfs. So I just clone the package at about a week a ago.  
I will try to delete the package and clone it again. 

I have changed the broadcast add to the “E7E7E7E7E7” in the launch file, will this cause something wrong ? Actually at the first time I use this launch file,the broadcast add had not been changed at that time, it also appears to the process has died. 

| |
Eric Liu
|
|
邮箱：lkyfly0765@126.com
|

Signature is customized by Netease Mail Master

On 04/10/2019 08:44, whoenig wrote:

Which version/commit are you using? Those line numbers actually don't point to code in the latest master branch.

The broadcast address should not be changed from the default value (it's hard-coded in the firmware).

whoenig, 2019-04-11:
I couldn't find the lines even for the version from a week ago. You could re-clone, or attach the source code file (crazyswarm_server.cpp) here. If you change the broadcast address it should not crash (however, broadcast communication might not work correctly.) The crash is likely caused by some incorrect setting in crazyflieTypes.yaml or allCrazyflies.yaml that causes an out-of-bounds error.

lkyfly0765, 2019-04-11:
The allCrazyflies.yaml is:
`crazyflies:
  - id: 1
    channel: 99
    initialPosition: [1.0, 0.0, 0.0]
    type: default
  - id: 2
    channel: 99
    initialPosition: [0.0, 0.0, 0.0]
    type: default
  - id: 3
    channel: 99
    initialPosition: [-1.0, 0.0, 0.0]
    type: default`

And the crazyflieTypes.yaml is not changed.Just the default configuration.
Also we are facing another issue that when we use the `python chooser.py` to get the battery info of each cf, but in the xterm window it feedback as below(we have three cfs for test):
` eric@eric-ThinkPad-S2:~/crazyswarm/ros_ws/src/crazyswarm/scripts$ python chooser.py 
[rospack] Error: package 'crazyflie_tools' not found
[rospack] Error: package 'crazyflie_tools' not found
[rospack] Error: package 'crazyflie_tools' not found
`
This problem occurs also when we try to reboot or sysoff the cfs. It feedback the same info `[rospack] Error: package 'crazyflie_tools' not found`. What caused this?

I am still testing the hover_swarm.launch file.So I am not sure whether the previous issue has been solved or not. Here I attach the crazyswarm_server.cpp code:
`#include "ros/ros.h"
#include <tf/transform_broadcaster.h>
#include <tf_conversions/tf_eigen.h>
#include <ros/callback_queue.h>

#include "crazyflie_driver/AddCrazyflie.h"
#include "crazyflie_driver/LogBlock.h"
#include "crazyflie_driver/GenericLogData.h"
#include "crazyflie_driver/UpdateParams.h"
#include "crazyflie_driver/UploadTrajectory.h"
#undef major
#undef minor
#include "crazyflie_driver/Takeoff.h"
#include "crazyflie_driver/Land.h"
#include "crazyflie_driver/GoTo.h"
#include "crazyflie_driver/StartTrajectory.h"
#include "crazyflie_driver/SetGroupMask.h"
#include "crazyflie_driver/FullState.h"
#include "crazyflie_driver/Position.h"
#include "std_srvs/Empty.h"
#include <std_msgs/Empty.h>
#include "geometry_msgs/Twist.h"
#include "sensor_msgs/Imu.h"
#include "sensor_msgs/Temperature.h"
#include "sensor_msgs/MagneticField.h"
#include "std_msgs/Float32.h"

#include <sensor_msgs/Joy.h>
#include <sensor_msgs/PointCloud.h>

//#include <regex>
#include <thread>
#include <mutex>
#include <condition_variable>

#include <crazyflie_cpp/Crazyflie.h>

// debug test
#include <signal.h>
#include <csignal> // or C++ style alternative

// Motion Capture
#ifdef ENABLE_VICON
#include "libmotioncapture/vicon.h"
#endif
[... 1838 lines cut ...]

  XmlRpc::XmlRpcValue crazyflies;
  nGlobal.getParam("crazyflies", crazyflies);
  ROS_ASSERT(crazyflies.getType() == XmlRpc::XmlRpcValue::TypeArray);

  std::set<int> cfIds;
  for (int32_t i = 0; i < crazyflies.size(); ++i)
  {
    ROS_ASSERT(crazyflies[i].getType() == XmlRpc::XmlRpcValue::TypeStruct);
    XmlRpc::XmlRpcValue crazyflie = crazyflies[i];
    int id = crazyflie["id"];
    int channel = crazyflie["channel"];
    if (cfIds.find(id) != cfIds.end()) {
      ROS_FATAL("CF with the same id twice in configuration!");
      return 1;
    }
    cfIds.insert(id);
  }

  // ROS_INFO("All CFs are ready!");

  server.run();

  return 0;
}`

lkyfly0765, 2019-04-13:
> I couldn't find the lines even for the version from a week ago. You could re-clone, or attach the source code file (crazyswarm_server.cpp) here. If you change the broadcast address it should not [...]

During my test, for Optitrack users, in hover_swarm.launch file L44 https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L44, you have to configure the 'optitrack_local_ip:' to your ros laptop's IP add, or just comment out this line. But I'm not sure will it connect to the Motive correctly if I comment it out.

lkyfly0765, 2019-04-20:
Here we find the solution of this problem. It's for Optitrack users.
For the [crazyswarm-server-1] Process has died problems, please check the following options:

- 1:

In hover_swarm.launch file L44 https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L44, please confirm that the  'optitrack_local_ip:' to your ros laptop's IP address, and Don't comment this line or it will cause the connection problems of your computer with VRPN server.

- 2:

 The crazyflies.yaml & allCrazyflies.yaml is correcting configured. Importantly, the IDs must be set start from number 1.

- 3:

 The firmware version is update to the lastst version. You can flash the firmware using the chooser.py in ~/ros_ws/src/scripts.
Importantly, you should confirm that BOTH STM32 & nRF51 firmware has been updated correctly.

- 4:

For optitrack users, if you choose "motionCapture" as "object_tracking_type", you should create rigid body in Motive software. 

But, IMPORTANTLY,Rigid Bodies must be created in order of cf2's ID number in allCrazyflies.yaml file.That is, cf1 must be created as the first one, and cf2 as the second one and cf3...    A name of rigid body is also important. We highly recommended that you name the rigidbodies as cf1,cf2,cf3 and so on. However, its order is important! #reference https://github.com/USC-ACTLab/libmotioncapture/pull/3

- 5:
 For more cfs, the different markers on each cf may be very difficult. We highly recommended you to use "libobjecttracker" as"object_tracking_type" to avoid insufficient space to place the markers. Importantly, if you choose "libobjecttracker", you should not create any rigid body in Motive software. And you should config the crazyflieTypes.yaml by using the` $roslaunch crazyswarm mocap_helper.launch`

closed 2019-04-20 by lkyfly0765