USC-ACTLab/crazyswarm #148: Real Time Battery Status Feedback

hanzheteng, 2019-04-09:
Hi @whoenig,

During our experiments, the batteries for crazyflie can only support 3-5 mins flight time. (Original battery 4-5 mins and Turnigy 3-4 mins.) Therefore, I am thinking about how to get a real-time battery status feedback. 

I remember you mentioned somewhere (couldn't find it again) that you are using LED extension board to show the battery status. I think it's a good idea. Which firmware version would support this feature? Do I need to set some parameters about this?

Also, is there a way to get real time battery feedback through commend line? The crazyflie_tools package provides a way to check the battery level, but it can only be done when the robot is on the ground, since the channel will be occupied for communications of the python script commends.

I found a `setLEDColor` function in the `crazyflie.py` file, but couldn't find any other function for showing the battery status.

Thanks,
@hanzheteng

whoenig, 2019-04-10:
The LED effect is part of the crazyswarm firmware (not yet in the official bitcraze firmware). It is enabled by default (by setting this variable to 16): https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L28.

By default, logging is disabled (i.e. the basestation only communicates with the CFs using broadcasts). You can enable logging here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L54. Configure the variable name as "pm.vbat" here: https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L19. You will be able to see the battery voltage by using "rostopic echo /cf<id>/log1". Note that if you enable logging you will not be able to communicate with as many CFs as before.

hanzheteng, 2019-04-10:
Very clear and useful. Thanks!

closed 2019-04-10 by hanzheteng