USC-ACTLab/crazyswarm #149: Position controller instability at yaw angles close to 180 degrees

williamdwarke, 2019-04-16:
I'm currently experiencing a very bizarre yet consistent behavior where the low-level position controller goes unstable when I command yaw angles close to 180 degrees. I'm using a custom trajectory generator along with the sendPositionSetpoint function, but the issue persists even when I simply command a position hold at any location with a yaw angle of 180 +/- 15 degrees or so. I've tested this with both the PID and Mellinger server config parameters, and have troubleshooted everything I can on the server side so I'm pretty sure the problem is somewhere in the crazyflie firmware, which I have yet to dissect. Any pointers or troubleshooting ideas regarding this issue?

jpreiss, 2019-04-16:
I believe this is due to an error in the EKF state estimator implementation in the firmware that @whoenig has identified and has an upcoming fix. (@whoenig, if there is any remaining coding effort needed to implement this, I can help).

whoenig, 2019-04-17:
@jpreiss I don't recall having found a big in the yaw state estimation like that. The last thing I did on the EKF was implementing the position-only updates. Could you elaborate?

@williamdwarke I have seen this issue too and I do think it's a firmware problem. Which state estimator are you using (kalman or kalmanUSC)?

williamdwarke, 2019-04-17:
Ah, I can confirm that switching from KalmanUSC to Kalman removes the instability. Additionally, using the Mellinger controller with an initial yaw of 180 degrees causes instability during takeoff for some reason, but switching to PID completely fixes it. So far I've had good luck using Kalman and PID with this setup and will report back if I encounter any more bugs/instability. Thanks so much for your help!

jpreiss, 2020-06-13:
Closing this as we have added the necessary features to the official firmware's `kalman` estimator and no longer recommend using our firmware with `kalmanUSC`.

closed 2020-06-13 by jpreiss