USC-ACTLab/crazyswarm #151: Problems running multiple waypoint CSVs

J-Durotifa, 2019-04-29:
Hi,

I'm trying to fly a drone swarm using the "csv_sequence.py" script. The drones fly properly when there's just one "waypoint" file in each folder. Every time I have 2 or more csv files in a drone folder, it seems like the axes becomes flipped and the drones start flying in the wrong direction, sometimes crashing.

I would like to know if this is a normal occurrence or if I've just been doing something wrong. I want to be able to assign different waypoints to different drones instead of using the figure8 script where all the drones follow the same trajectory.

whoenig, 2019-04-30:
If you want one trajectory for each drone, you can look at swap6v.py.

@jpreiss might be able to comment on csv_sequence.py (I have not used it before).

jpreiss, 2019-04-30:
`csv_sequence.py` is not tested very thoroughly yet. Its purpose is similar to `swap6v.py`, but instead of uploading polynomial trajectories to the crazyflies, it evaluates the polynomial trajectories on the PC and sends high-rate control setpoints to the crazyflies in real time. The reason to do this is if your trajectories are too long to fit in the crazyflie's memory. It will not work for larger numbers of crazyflies due to increased radio bandwidth requirements. I would suggest using `swap6v` instead unless you have long trajectories.

jpreiss, 2019-04-30:
`csv_sequence.py` also has some irrelevant code for setting LED colors that is from a specific multi robot planning experiment... it should be removed.

J-Durotifa, 2019-04-30:
> `csv_sequence.py` is not tested very thoroughly yet. Its purpose is similar to `swap6v.py`, but instead of uploading polynomial trajectories to the crazyflies, it evaluates the polynomial [...]

From what I've seen, csv_sequence.py does upload the trajectory to the drone.
This line in the code is what does it:
    # upload the trajectories
    for cf, seq in zip(crazyflies, seqs):
        for i, traj in enumerate(seq):
            cf.uploadTrajectory(i, 0, traj)

I'd try commenting it out to see if it does the trick.

The trajectory generator doesn't allow me to set the time between waypoints for curves and shapes like circles, so my aim is to be able to fly a drone swarm with 7 degree polynomial CSVs and another CSV with GOTO commands so I can have some sort of control.

J-Durotifa, 2019-04-30:
> `csv_sequence.py` also has some irrelevant code for setting LED colors that is from a specific multi robot planning experiment... it should be removed.

I can't seem to find this section in the code

jpreiss, 2019-04-30:
ah, you are using an older version. please start with swap6v.py for your project

J-Durotifa, 2019-04-30:
Ah I see. I would check that. 

I want to be able to fly multiple trajectories per drone. Is this something I would be able to do with swap6v.py

whoenig, 2019-04-30:
Yes, as long as all trajectories can be uploaded while the CF is on the ground and fit in its memory.

closed 2020-02-25 by jpreiss