USC-ACTLab/crazyswarm #152: Optitrack mocap/vrpn network issue

UrizielTSD, 2019-05-17:
Hello,

we are trying to use Crazyswarm with the Optitrack tracking System. But are running in some issues.

First our setup:
- Windows 10, running optitrack Motive 2.1.1 
-- Manually configured IPv4 Controller (Ip: 192.168.105.1, subnetmask: 255.255.255.0)
- Ubuntu 16.05, running Crazyswarm, reinstalled this morning with the new git repo Version
--  Manually configured IPv4 Controller (Ip: 192.168.105.2 subnetmask: 255.255.255.0)
- Both are connected with each other directly with an Ethernet cable

Situation 1 VRPN:
Using Motive 2.1.1 or 1.9.0 on VRPN Streaming Engine, we are able to make crazyswarm work with Optitrack. But the Connection is unstable:
- LED ring blinking red and green
- When taking off with niceHover.py or other scripts, crazyflie Flips over
-- We adjusted inside crazyswarm the coodinate System, that optitrack System Matches ROS.

Situation 2 mocap:
Using Motive 2.1.1 or 1.9.0 on NatNet Streaming Engine, we are unable to make crazyflie work with optitrack.
- Using Wireshark on both machine we can see the pakages being multicasted out of the Windows machine and reaching the Ubuntu Machine.
- Using another Windows machine (same IP configuration as Ubuntu) and running the NatNet SDK sample program:
-- On the same Windows machine as Motive the Connection works and data is streamed correctly and automatically.
-- On another Windows machine connected to the main Windows machine the program is able to connect to the NatNet Server and is able to manually fetch a data description but there are no coordinates or update of those.

Picutes for Situation 2:
Both Windows machine are running the Connection to Motive Server at the same time, i tried to only run second machine, but same result. Addtionally you can see in Wireshark that the pakages are reaching the second Windows machine.

- Main Windows machine running Motive: 
https://ibb.co/PwgVk0K

- Second Windows machine connected over Ethernet cable to the main Windows machine:
https://ibb.co/W35BMnJ

We would like to use mocap with Optitrack so solving Situation 2 is our priority at the Moment.
And ideas or help would be greatly appreciated!

whoenig, 2019-05-17:
Situtation 1: This is strange. This method should work well for a few CFs with unique marker configuration (and certainly for a single one). What do you mean by: "We adjusted inside crazyswarm the coodinate System"? All you need to do is change the streaming settings in Motive (to have Z-axis up).

Situation 2:
I just made the following repository public, which I have used to test Optitrack with Motive 2.0.0: https://github.com/whoenig/NatNetSDKCrossplatform. Please follow the build and run instructions here and report back if you are able to receive data. Note that proper optitrack support has been only integrated into the Crazyswarm a few weeks ago - if you use an older version of the crazyswarm, there is only support for older Motive versions.

UrizielTSD, 2019-05-17:
Hello,
Thanks for the quick Reply,
Situation 1:
I changed inside the Crazyswarm VRPN files the Axis Code to match optitrack. Is what I meant by that, switching the Axis is an Option for the NatNet Streaming Engine. Does this Option also affect the VRPN Streaming Engine ?

Situation 2:
As stated above I reinstalled the complete crazyswarm package from Github this morning, which includes all of this weeks commits because of the newer Version support. I will try your repo tomorrow and give you Feedback asap.

Just a Question about your Connection Settings, did you only set the static Ipv4 and subnetmask on your PCs or did you made any additionall changes ?
Thank you

whoenig, 2019-05-17:
VRPN: I didn't actually test it myself. My understanding was that this setting affects both, but I am not entirely sure.

Connection: This should be sufficient. I actually use DHCP (and have a router between the two PCs). Since you get data using windows/wireshark this is likely just a SW issue.

UrizielTSD, 2019-05-18:
Hello,

I was able to run your repo program:
[image]

I had to use a second machine running xubuntu 18.05, since boostlib1.65 is required which isnt supported by ros kinetic. 
But that shouldnt matter for the Output.

Additionally I was able to see live updates using my Windows machine, I used a newer Version of the NatNet SDK Client.
The Data apparently streamed correctly over the Network, but crazyswarm isnt Picking them up somehow.

UrizielTSD, 2019-05-20:
Hello,

today I was finally able to copple Optitrack NatNet Motive 2.1 with your newest crazyswarm build.
The Connection is much better, still there are sometimes latency Problems but I dont think they are related with the following:
- While using Motioncapture with rigidbodies, and running the niceHover script this happens:
<div style="width:100%;height:0px;position:relative;padding-bottom:56.250%;"><iframe src="https://streamable.com/s/6rvfk/goxemm" frameborder="0" width="100%" height="100%" allowfullscreen style="width:100%;height:100%;position:absolute;left:0px;top:0px;overflow:hidden;"></iframe></div>
- I configured Z Up in Motive
- I checked that Z was up in Rziz
- What I did observe was that X and Y where inversed

I also tried libobjecttracker, but that produces a similar result (the drone just slides out of the Tracker field).

Any recommendations on how to make the Flies, fly correctly ?

whoenig, 2019-05-20:
Looks like a coordinate frame problem to me. Make sure that the drones rigid body is configured in such a way that positive x is in front (where the antenna is), and positive y is on the left side. Check that this coordinate system is also correct in ROS (no need to fly - you can just manually move the drone around and verify that it matches.)

vchavezfs, 2019-05-23:
Hello again,

It looks like after the latest changes to optitrack support, I unable to get data from motive? Whenever I run the mocap_helper.launch file, nothing prints out. What I noticed are the parameters not being read when I run the launch file. The older build works fine, but it looks like there were quite some modifications to properly support optitrack and a firmware upgrade. I also noticed that the launch files were changed to just input the hostname rather than local ip and server ip in the older launch file. Any ideas? 

[image]

whoenig, 2019-05-23:
Could you try the standalone applications at https://github.com/whoenig/NatNetSDKCrossplatform and report back if those tools worked for you?
It is correct that the new code does not require on the "local IP" anymore (it simply uses localhost).

jpreiss, 2020-06-13:
@vchavezfs were you able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss