USC-ACTLab/crazyswarm #154: Commands ignored by crazyflie

medioman22, 2019-05-26:
Hello,

I have followed the instructions for using crazyswarm with OptiTrack but I am running through some issues.

Let me first describe my setup:

- I have 1 crazyflie 2.1
- 1 crazyradio PA
- OptiTrack mocap system with Motive 2.0.2
- Ubuntu 16.04 ROS kinetic

After some efforts I managed to make the drone take off with hover_swarm.launch, and I am quite satisfied by the stabilization achieved. However, there seems to be a communication problem between the cf and the crazyradio, because the drone does not execute all the instructions sent.

When I pressed start on the controller to take off, I always get the output on the console, but the drone actually takes off sometimes and not always.

Here is the output of the terminal:
```
~/Documents/github/crazyswarm/ros_ws$ roslaunch crazyswarm hover_swarm.launch 
... logging to /home/lis-laptop-15-20/.ros/log/754d29e6-7fcc-11e9-a40b-507b9d6b467f/roslaunch-lislaptop1520-ThinkPad-T450s-9110.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lislaptop1520-ThinkPad-T450s:43974/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
[... 159 lines cut ...]
[ WARN] [1558885176.893531232]: [cf1] Link Quality low (0.690000)
[ INFO] [1558885178.029765811]: Takeoff!
[ INFO] [1558885179.709442929]: Takeoff!
[ INFO] [1558885180.171974267]: Takeoff!
[ INFO] [1558885180.481736260]: Takeoff!

```

I sometimes get the message "[cf1] Link Quality low" even though I have only one cf, about 3 meters away (and I also get this message when I have the cf next to the crazyradio). What is strange is that using the python client app, I always have a perfect link quality.

I also noticed that when I set logging_enabled to false in the launch file, the drone does not respond to any command at all. Checking in the source of crazy_server I noticed that when logging is disabled, no ping is sent to the cf in the runSlow() function of CrazyflieGroup.

I have spent a lot of time trying to figure out how to use crazyswarm but I am not sure what to do now.

I have tried to use the python API in vain. I have tried using the ROS services for high-level commands as well as the topics for low-level commands also in vain. I experience the same issues, the ROS part works fine and I always get the output about the commands I send, but the cf does not always execute the commands.

What could be the problem ?
- communication with the cf
- firmware (cf and crazyradio) ? I have updated them several time and explained in the documentation of crazyswarm
- I saw that there could be problems with the USB3 ports, is that still an issue ? my laptop only has USB3 ports.
- Some sort of command buffer being filled in the drone or in the crazyradio ?

If someone has similar issues, did you find any workaround. There should be no such issue for controlling only one drone right ?

Best,

whoenig, 2019-05-27:
* For the radio, I suggest running `rosrun crazyflie_tools comCheck --uri <yourURI>`, which measures a lot of statistics for link quality. I don't think there is often an issue with USB3, and if there is it is HW-related and you would have trouble with the cfclient as well.
* For the other issues: How recently did you clone (or update) the crazyswarm package? 3 or 4 weeks ago a switched the communication to use the "safemode", but that implementation had a major bug which just got fixed about a week ago. This bug caused behavior similar to what you describe. There were no major bug fixes in the firmware recently (although, the latest crazyswarm uses new messages to broadcast the pose information, so if your crazyswarm clone was older, make sure that you do update the firmware as well.)
* Looks like you are also starting vrpn_client node. This is not needed for the crazyswarm (although running it shouldn't cause the issues you described.)

medioman22, 2019-05-28:
Thank you for such a quick reply.

- I have ran the command `rosrun crazyflie_tools comCheck --uri <yourURI>` which gives me very good results.
```
rosrun crazyflie_tools comCheck --uri radio://0/80/2M/E7E7E7E701
numPacketsReceived: 95.6 %
Avg. roundtrip time: 20.7856 ms
751.88 packets/s sent
link quality: 1
```
- I had already noticed the update on the repo and indeed the results were much better after it.
- Indeed I am running the vrpn_client node, to track other objects as part of my application.

After investigating, I managed to get satisfying results by sending several time the same high-level command. Basically, I update the command I send at a rate of 2 Hz, but i send the same command at a rate of 25 Hz to make sure the drone executes it.
However, for my specific application I would need to update the command at a frequency of about 10-15 Hz. I am using the Melinger controller as it is very stable, but I am aware that it means that the crazyflie has to recompute a trajectory onboard for each new command. For this reason, I have observed that the crazyflie can't remain stable if I send a new command at a higher rate.

For my application, I need to send continuously position setpoints to the drone, and the setpoints are not changing drastically but they are not known in advance (i.e. I cannot just update a trajectory and ask the drone to follow it). Basically instead of asking the drone to move 0.5m in 1second (which is suited for the high-level ROS services), I would rather like to ask it to move 5cm in 0.1second. However, I have issues with the two options that seem suited for this task:

1. Using the PID controller instead of the Mellinger controller. The PID controller is much less stable and the drone is drifting. Is there a parameter to adjust the roll and pitch inputs to correct this effect ?
2. Using the low-level ROS topics (/cf1/cmd_position). I am not entirely sure how they work. When I send some commands the drone starts flying but then the motors stop. Do I need to continuously send position commands ? I have seen on some other issues that the high-level and low-level commands cannot be mixed, is that correct ?

Finally, I still have the same question about the `enable_logging` parameter. When it is set to false, the drone does not respond at all to any command. Is that the intended behavior ?

Thanks

whoenig, 2019-05-28:
1. PID controller: Not sure, but it should actually behave similarly to the mellinger controller. The performance might depend on which state estimator you use. The PID controller is optimized for Kalman, while mellinger has been optimized for KalmanUSC.

2. Commands: Ideally you should send cmd_position or cmd_full_state, and not goTo commands for your application. Both of those do not do any onboard planning and use the provided setpoint directly instead. You do need to send the commands fairly frequently - there is a timeout on-board that is triggered after 500ms of no command received (at which point it will try to hover; after 2s the motors shut off). You are correct that mixing low- and high-level commands does not fully work yet. You can take-off using high-level, then moving using low-level, and land using low-level. You cannot switch from low-level to high-level, but you can switch from high-level to low-level.

3. Enable_logging should only have an effect on streaming data from the CF to the basestation. What you describe sounds like a bug to me. Do you have any more details on that?

medioman22, 2019-05-30:
Hello !

Thank you very much for your answer. I have looked into it, and manage to get something satisfying for my application:

1. PID controller: indeed after switching the estimator to Kalman (not KalmanUSC) the result is quite good. I have been able to send high-level commands at my desired rate and am quite happy with the results, as the drone trajectory is quite smooth even when I change the setpoints.

2. Thank you for the explanation, it did not understand that the low-level command had to be sent continuously. I will stick with the high-level commands for now as they provide a neat interface for take-off and landing, and I obtained satisfying results.

3. Basically, whenever l set Enable_logging to False, hover_swarm.launch works as usual and I can see all the commands being sent, however the drone nerver takes off. In fact, the LED "M4" which usually is green or flickering green does not light up, except at startup. Looking at the source code of the server, I noticed that at startup, a ping message is sent 50 times to each drone listed in the config file. This corresponds to the quick flickering of LED "M4" that I notice when I launch hover_swarm.launch. Here are the corresponding lines in crazyswarm_server.cpp, in CrazyflieGroup::readObjects():
```cpp
// Turn all CFs on
for (const auto& config : cfConfigs) {
    Crazyflie cf(config.uri);
    cf.syson();
    for (size_t i = 0; i < 50; ++i) {
         cf.sendPing();
    }
}
```

However, I noticed that when logging is disabled, no ping is sent at all. I read somewhere (forgot where) that a ping should be sent regularly to the cf to keep the connection working. However, in CrazyflieGroup::runSlow():
```cpp
while(ros::ok() && !m_isEmergency) {
      if (enableLogging) {
        for (const auto& cf : m_cfs) {
          cf->sendPing();
        }
      }
      m_slowQueue.callAvailable(ros::WallDuration(0));
      std::this_thread::sleep_for(std::chrono::milliseconds(10));
    }
```
the ping is not sent at all if Enable_logging is false.

whoenig, 2019-05-31:
About 3: Since the crazyswarm is mostly using broadcast, it is actually intended to not send a ping if enable_logging is set to false. What is not intended is that your CF is not doing anything in that particular case - this does sound like a bug. Note that for a larger swarm, you will need to set enable_logging to false, otherwise you can only control a few (~4) CFs/radio.

closed 2020-06-13 by jpreiss