USC-ACTLab/crazyswarm #155: No communication with  crazyflie and the crazyradio in crazyswarm

lincoln1587, 2019-05-31:
Hi, I got setup the crazyswam and vrpn, but I cannot successfully take off the cf2 in hover_swam.launch.
1. I have update the firmware using `make cload`
2. I can see the cf2 in rviz dynamically
3. After I push the take off button, nothing happened.
4. I can communicate using the cfclient
The launch file is set as 
 ```
     firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "E7E7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      # genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      # genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      # genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      # genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      # genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 3 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_host_name: "192.168.1.121"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
```
and I got the output as 

```
SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
[... 182 lines cut ...]
wroll: -29.6886 >= 20
wpitch: -33.9442 >= 20

[ WARN] [1559291606.512044829]: No updated pose for CF cf4 for 0.000367 s.
[ WARN] [1559291606.555412919]: Dynamic check failed for object cf4
wroll: -27.5046 >= 20
wpitch: -30.8324 >= 20

[ WARN] [1559291606.555456734]: No updated pose for CF cf4 for 0.000400 s.
[ INFO] [1559291613.663247774]: Takeoff!
[ INFO] [1559291619.561099744]: Land!
[ INFO] [1559291620.321771045]: Takeoff!
[ INFO] [1559291621.041713770]: Takeoff!
[ INFO] [1559291621.305450885]: Takeoff!
[ INFO] [1559291621.497124924]: Takeoff!
[ INFO] [1559291621.618048248]: Takeoff!
[ INFO] [1559291625.994520740]: Land!
[ INFO] [1559291632.330600580]: Land!
[ INFO] [1559291633.942937544]: Takeoff!

```

Also I notice that the green light in CF2 is not flashing. 

Thanks

whoenig, 2019-05-31:
Does it work if you set enable_logging to True in hover_swarm.launch? This might be related to issue #154.

lincoln1587, 2019-06-01:
@whoenig  Thanks for your suggestion. After set enable_logging to True, the communication is connected I think, based on the flashing green light. But, my cf2 is still not moving after the take off command. lol.

lincoln1587, 2019-06-02:
@whoenig I found that I can only command one CF2, by setting the broadcast address the same as the CF2 address based on the hovering_swarm launch. How to fix that ?

whoenig, 2019-06-03:
The broadcast address should not be changed - it is actually hardcoded in the firmware. However, if it works this way, this likely means that you didn't successfully update the NRF firmware (which is needed for broadcasts to work properly.)

jpreiss, 2020-06-13:
@lincoln1587 were you able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss