USC-ACTLab/crazyswarm #156: can't not takeoff the crazyflie

luozhexin, 2019-06-05:
Hello,

I have followed the instructions for using crazyswarm with OptiTrack but I am running through some issues.
when i using hover_swarm.launch,i press the button to send the command Takeoff,but it doesn't takeoff.I can see the red on the crazyflie,that means cf is in safety model.So i flash the fireware(STM32 and nrf51),but it still red.How do I know if I flash correctly？

whoenig, 2019-06-05:
If you get a red solid LED, than the firmware likely crashed. You can look at the console output (e.g., using cflient or the console tool in crazyflie_tools) to see if there are any further details on what went wrong. If you use a JLINK adapter or `crazyflie_tools flash` for flashing your firmware, the image will be verified.

luozhexin, 2019-06-06:
Thank you for your quickly reply.Follow your instructions，i see the output of the cfclinet console.But i have not idea what it means,so i took a PrtScr.
I would appreciate it if you could help me.

luozhexin, 2019-06-06:
sorry to disturb you again.I think that the detail information can help me better.Here are the results and the launch file.
I can see the cf in rviz dynamically ，but it still can’t takeoff.
I would appreciate it if you can help me.

whoenig, 2019-06-06:
@luozhexin, unfortunately, your attachments didn't make their way to github. Could you upload the additional information here?

luozhexin, 2019-06-06:
Here is the PrtScr
[image]
[image]
Here is the result and launch file
`/crazyflieTypes/default/batteryVoltageWarning: 2.8
 * /crazyflieTypes/default/batteryVoltateCritical: 2.7
 * /crazyflieTypes/default/bigQuad: True
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 3
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: False
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: False
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_z: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_xy: 50000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_xy: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_z: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_xy: 0.1
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_z: 0.5
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kp_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kp_z: 3
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kw_xy: 16000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/mass: 0.11
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/massThrust: 23000
 * /crazyflieTypes/medium/markerConfiguration: 1
[... 156 lines cut ...]
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: True
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  
  

</launch>
`

whoenig, 2019-06-06:
Your gyro self-test fails. You might want to contact bitcraze for help if this problem persists. This indicates a problem with your IMU sensor.

luozhexin, 2019-06-06:
I think there is no problem with CF. Maybe there is something wrong with the computer configuration. Does the unconnectedness in figure 1 affect the results?

luozhexin, 2019-06-06:
Beside，the red in the CF is flashing

luozhexin, 2019-06-07:
thanks again! I worked on this package today,this time crazyflies tools finally worked fine,i can get battery information, flash(stm32) ,flash(NRF) and reboot function perform well.So,i have got the right address of crazyflie,PA is fine i thought,the crazyflie is fine too(at least we can contact with it).
so my problem may still be related to the hover_swarm.launch file and the related flie. Today,i got an error when i shout down the terminal which is running  hover_swarm.launch,it outputs as below`
[crazyswarm_server-1] escalating to SIGTERM
`
so , besides hover_swarm.launch file ,which flies do i need to make some changes in order to connect to the crazyflie? forgiving me that i am a newer to this, thank you very much  for guiding me .

whoenig, 2019-06-08:
No other files need to be changed. From your log above, I don't see an attempt in the crazyswarm_server to connect to your CF. Did you use the tools from chooser.py?

ygtq99, 2019-06-08:
Hi，whoening
Sorry to disturb you.
I also faced the problem like this, when I roslaunch hover_swarm. launch and press the take off button(by the xbox), the crazyflie didn't  respond,only on the terminal display takeoff or land! and the crazyflie didn't seem to connect to crazyradio PA, because the crazyflie's lights didn't change.
The allCrazyflies.yaml and hover_swarm.launch have been changed,we have optitrack to get the coordinate information by using vprn/sample.launch(because we found that we can't fount the crazyflie in rviz just use the hover_swarm.launch)
And it seemed that the chooser.py  worked well，I can get the battery information，using flash(stm32) ,flash(NRF) and press the reboot function to restart the crazyflie.
And I want to tell you my experience during this period (Maybe you can see what problems did I do wrong)：
1.install Simulation and Physical Robots from
https://crazyswarm.readthedocs.io/en/latest/installation.html
and it seemed successful. 
2.After finding out crazyflie_tools was missing the packages，clone the package from https://github.com/whoenig/crazyflie_tools
and now we can use chooser.py .
3.With a crazyflie,crazyradio PA,xbox,we roslaunch crazyswarm hover_swarm.launch and 
and vrpn/sample.launch,now we can see the crazyflie in rviz and in the terminal it tell us to press the button to take off or land ,but when we press the 'take off' button,it just display  'take off' but no respond in crazyflie ,and the lights of crazyflie seemed no change.but when we using the older package like crazyflie_ros,the crazyflie works well.I think that the crazyradio PA can't connect witih crazyflie，but I don't know what‘s wrong with this.
In the last,I think the main reason is that the communication problems,the PA with crazyflie and the optitrack with my PC.One thing to note is that only by vrpn/sample.launch  we can see the crazyflie in rviz .
Thank you in advance!

jpreiss, 2019-06-08:
Did you flash your crazyflies with the Crazyswarm firmware as described in https://crazyswarm.readthedocs.io/en/latest/usage.html#crazyflie-preparation ?

ygtq99, 2019-06-13:
Oh,thanks @jpreiss 
I have followed the steps to flash the crazyflies in https://crazyswarm.readthedocs.io/en/latest/usage.html#crazyflie-preparation .
In the first ,I have modified `hover_swarm.launch`many times ,at last I think choose `vrpn` in the `motion_capture_type` is the easiest way to connect the crazyflie.(but we faced some problems later)

### hover_swarm.launch

>  firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "FFE7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 3 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      # optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      vrpn_host_name: "169.254.41.1"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False

By using `hover_swarm.launch`,we can easily see the crazyflie in the rviz ,and the `led-ring` changed to green.

1. 

At the first time,we opened another terminal to use `figure8_csv.py`,the crazyflie works well likes the 
`figure8_csv.py--sim`,but it's not very stable and we found `No updated pose for motion capture object cf3 [ WARN] ` ,and it's constantly refreshing on the terminal

2.
At that time, I think the connection about the `optitrack` with my PC is not well, and I also want to do the control of the crazyflies group,I try to use the `whoenig/NatNetSDKCrossplatform` #[https://github.com/whoenig/NatNetSDKCrossplatform](url)
But after downloading it, when I went to the `cmake..`step,it seems to prompt me that `boost` version is too low,and I don't know how to do it then.

3.
And in the later the crazyflie always flipped onto the ground and that problem still exists(`No updated pose for motion capture object cf3 [ WARN] `).It's not even as good as it used to be.I found that after using the `hover_swarm.launch`,the led-ring will be green and when we press the `take off`button,the led-ring will be red.
I think it's necessary for us to configure the `NatNetSDKCrossplatform` to control the formation flight of crazyflies.
Can you give me some advice?

whoenig, 2019-06-13:
2.: You can lower the version number here: https://github.com/whoenig/NatNetSDKCrossplatform/blob/master/CMakeLists.txt#L3. It does not need the latest boost version.

Make sure you have a physical ethernet connection between your Optitrack system and your Ubuntu system (i.e., avoid using WiFi).

ygtq99, 2019-06-14:
@whoenig 
Thanks!
I think I can go on then. 
I think after configuring `whoenig/NatNetSDK Crossplatform`and modifying the `hover_swarm.launch`,we can control the crazyflie better.

And this is my hover_swarm.launch,are there some problems? 
### hover_swarm.launch

>  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "FFE7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 3 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_host_name: "169.254.41.1"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "169.254.41.1"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
If all this is done, do I just need to configure the python file to control crazyflies?

whoenig, 2019-06-14:
Your config looks fine. If the example scripts work (e.g., figure8_csv.py), you should be ready to implement your own high-level python scripts.

closed 2020-06-13 by jpreiss