USC-ACTLab/crazyswarm #157: Crazyflies don't take off

BIsichei, 2019-06-05:
Hello, I'm having issues running hover_swarm.launch.

I'm running on Ubuntu 16.04 and using optitrack cameras for motion capture. I've followed the instructions on getting started with [crazyswarm](https://crazyswarm.readthedocs.io/en/latest/usage.html#adjust-configuration-files), and so far I've 

- labelled, assigned IDs to both my test crazyflies and have them running on channel 80
- upgraded their firmwares using cload
- upgraded the crazyradio firmware
- setup pc permissions with `./pcpermissions.sh`
- updated my allCrazyflies.yaml to:
````
crazyflies:
  - id: 1
    channel: 80
    initialPosition: [0.0, -1.0, 0.0]
    type: default
  - id: 2
    channel: 80
    initialPosition: [0.0, -1.25, 0.0]
    type: default
````
- not been able to get mocap_helper to display any readings while using optitrack (motionCapture or libobjecttracker). It was able to display the drones using vrpn, so I decided to go with that. 

- updated the relevant section of hover_swarm.launch to:
````# tracking
      motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      # optitrack_host_name: "<tracking computer ip>"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      vrpn_host_name: "<tracking computer ip>"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
````
- run the chooser gui and been able to check each flie's battery level
- used crazyflie_tools to successfully find each flie at *E701 and *E702

My issue so far is that whenever I try to run hover_swarm, 
- I get a "No Updated pose for motion capture object cf1" error, but rviz displays the current location of each drone. 
- when I hit 'start' on the controller, 'Takeoff!' is displayed on the terminal, and the flies's propellers spin momentarily, then die off. this is what the console looks like after each run
````
 roslaunch crazyswarm hover_swarm.launch 
... logging to /home/ada-1/.ros/log/1de8786e-86f1-11e9-9d7a-c8ff281d5ed9/roslaunch-Unit-1-14729.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Unit-1:33737/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
[... 570 lines cut ...]
[ WARN] [1559771427.928934425]: Don't know ack: Port: 5 Channel: 0 Len: 19
[ WARN] [1559771427.930524286]: Don't know ack: Port: 5 Channel: 0 Len: 19
[ WARN] [1559771427.933642305]: Don't know ack: Port: 5 Channel: 0 Len: 19
[ WARN] [1559771427.935224392]: Don't know ack: Port: 5 Channel: 0 Len: 25
[ WARN] [1559771427.936551005]: Don't know ack: Port: 5 Channel: 0 Len: 14
[ WARN] [1559771427.938006851]: Don't know ack: Port: 5 Channel: 0 Len: 13
[ WARN] [1559771427.939205527]: Don't know ack: Port: 5 Channel: 0 Len: 11
[ WARN] [1559771427.940461180]: Don't know ack: Port: 5 Channel: 0 Len: 13
[ WARN] [1559771427.942366624]: Don't know ack: Port: 5 Channel: 0 Len: 14
[ WARN] [1559771427.948506433]: Don't know ack: Port: 5 Channel: 0 Len: 24
[ WARN] [1559771427.949843782]: Don't know ack: Port: 5 Channel: 0 Len: 15
[ INFO] [1559771427.955585921]: [cf1] reqLogTOC: 0.459253 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find ctrlStat.edist in log toc!
[crazyswarm_server-1] process has died [pid 15759, exit code -6, cmd /home/ada-1/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/ada-1/.ros/log/1de8786e-86f1-11e9-9d7a-c8ff281d5ed9/crazyswarm_server-1.log].
log file: /home/ada-1/.ros/log/1de8786e-86f1-11e9-9d7a-c8ff281d5ed9/crazyswarm_server-1*.log
^C[rviz-4] killing on exit
[crazyswarm_teleop-3] killing on exit
[joy-2] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

````
Is there something i've omitted to do?

whoenig, 2019-06-06:
Looks like you followed the steps correctly. Which firmware did you flash?

For your issues:
1. Could you post the output of `roslaunch crazyswarm mocap_helper.launch` with the vrpn configuration? For the optitrack configuration, did you make sure that you enabled streaming of rigid bodies, labeled markers, and unlabeled markers in Motive? Also double check that Motive shows the correct IP address in the streaming tab. Finally, which version of Motive are you using?
2. I don't recommend flying if you have so many warnings regarding tracking. Nevertheless, the error message you get when you enable logging is related to the log variables configured here https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L19. This variable doesn't exist in your firmware. You could use 'stateEstimate.x' instead.

BIsichei, 2019-06-06:
I used the crazyflie-firmware that comes with the crazyswarm repo. Should I be using the crazyflie2-nrf-firmware instead? 

With the vrpn config, I get 
````
 roslaunch crazyswarm mocap_helper.launch 
... logging to /home/ada-1/.ros/log/41fba474-887c-11e9-9d7a-c8ff281d5ed9/roslaunch-Unit-1-21043.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://Unit-1:35049/

SUMMARY
========

PARAMETERS
 * /mocap_helper/motion_capture_type: vrpn
 * /mocap_helper/object_tracking_type: motionCapture
 * /mocap_helper/optitrack_host_name: 172.25.97.125
 * /mocap_helper/vrpn_host_name: 172.25.97.125
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

ROS_MASTER_URI=http://localhost:11311

process[mocap_helper-1]: started with pid [21060]
check_vrpn_cookie(): VRPN Note: minor version number doesn't match: (prefer 'vrpn: ver. 07.34', got 'vrpn: ver. 07.20  0').  This is not normally a problem.
frame 0:
    points:
    objects:
frame 1:
    points:
    objects:
frame 2:
    points:
    objects:
frame 3:
tracker: cf1
tracker: cf2
    points:
    objects:
frame 4:
    points:
    objects:
      "cf2":
         occluded: 0
         position: [-2.08957, -0.560325, 0.032489]
         rotation: [-0.449568, 0.487306, 0.034267, 0.747829]
      "cf1":
         occluded: 0
         position: [-2.09366, -0.379157, 0.0320606]
         rotation: [-0.349077, -0.270634, 0.226746, -0.868037]
frame 5:
    points:
    objects:
frame 6:
    points:
    objects:
      "cf2":
         occluded: 0
         position: [-2.08957, -0.560287, 0.0324451]
         rotation: [-0.449038, 0.487411, 0.0348656, 0.748051]
      "cf1":
         occluded: 0
         position: [-2.09366, -0.379163, 0.0320689]
         rotation: [-0.349052, -0.270702, 0.226762, -0.868022]
frame 7:
    points:
    objects:
frame 8:
    points:
    objects:
      "cf2":
         occluded: 0
         position: [-2.08957, -0.560324, 0.0324893]
         rotation: [-0.449521, 0.487439, 0.0342297, 0.747772]
      "cf1":
         occluded: 0
         position: [-2.09365, -0.379167, 0.0320535]
         rotation: [-0.349006, -0.2706, 0.226761, -0.868072]
frame 9:
    points:
    objects:
frame 10:
    points:
    objects:
      "cf2":
         occluded: 0
         position: [-2.08956, -0.560318, 0.0324486]
         rotation: [-0.44938, 0.488294, 0.0344136, 0.74729]
      "cf1":
         occluded: 0
         position: [-2.09364, -0.379184, 0.0320531]
         rotation: [-0.34908, -0.270622, 0.226752, -0.868038]
frame 11:
    points:
    objects:    
````
...

I'm using motive 1.9.0, all those options are enabled, and it's publishing to the correct interface. 
[image] 

Whenever I run mocap helper with optitrack, the motive software indicates it's connected to a client, but there are no updates on the client.

````
PARAMETERS
 * /mocap_helper/motion_capture_type: optitrack
 * /mocap_helper/object_tracking_type: motionCapture
 * /mocap_helper/optitrack_host_name: 172.25.97.125
 * /mocap_helper/vrpn_host_name: 172.25.97.125
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    mocap_helper (crazyswarm/mocap_helper)

ROS_MASTER_URI=http://localhost:11311

process[mocap_helper-1]: started with pid [21962]
frame 0:
^C[mocap_helper-1] killing on exit
[mocap_helper-1] escalating to SIGTERM
shutting down processing monitor...
... shutting down processing monitor complete
done
````
I really appreciate your help with this.

whoenig, 2019-06-06:
Firmware: You need to flash both the STM32 and NRF51 firmwares. For both it is best to use the ones in the repo, perhaps even the ones in the prebuilt-folder as those are tested.

VRPN: Your output shows the problem: Every other frame does not contain any tracked rigid bodies. So when you run the hover_swarm launchfile you do see objects tracked in rviz, but every other frame it prints a warning. I have not seen this behavior before. Nevertheless, I don't recommend using the VRPN backend, since it won't allow you to use libobjecttracker (which will be important for larger swarms).

Optitrack: Please try the tools in https://github.com/whoenig/NatNetSDKCrossplatform and see if they are able to connect. Those are mostly unchanged from the official NatNetSDK. I tested with Motive 2.0, so if you do have an option to upgrade to 2.x, this might also help (however, your version of Motive should work in theory, I just was not able to test it.)

lkyfly0765, 2019-06-10:
@whoenig 
Hi whoenig,
How to use the NatNetSDKCrossplatform  as a optitrack user?
How to configure the launch file if I want to use the NatNetSDKCrossplatform?
This is my configure info:

>       # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_host_name: "169.254.41.1"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "169.254.41.1"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False

but when I launch the hover_swarm.launch, there is no error in the terminal window, but the LED ring attached on the crazyflie is solid red, and the light on the crazyradio is off. I think the connection between crazyflie and the crazyradio is not established. And, run the figure 8 file, there is no respond with the crazyflie.

When we choose the vrpn as the motion_capture_type, as run vrpn_client_ros at the same time, we can see that the LED ring is green, and the light on crazyradio is flashing which mean the connection is correctly established. Then I run the figure 8 file, the crazyflie take off and hover at 1 meter above the ground stable, then it moves quickly and flap down to the ground. It can not execute the following trajectory.

I am sure that the firmware is flashed to the latest. Both STM and nRF. So, what may cause this problem?

Thanks a lot for your support!

whoenig, 2019-06-11:
@lkyfly0765 
NatNetSDKCrossplatform is a repository to test your connection to Motive using NatNetSDK. Follow the instructions there on how to use it. This SDK is automatically used when you choose "optitrack" as motion_capture_type.

lkyfly0765, 2019-06-11:
@whoenig 
Thanks for your reply. 
I use the NatNetSDKCrossplatform to test the connection between my laptop and the Motive.  Just as you wrote in the instructions, the ./mcr feedback the correctly which contains the Natnet version and Motive version.  
But as I wrote in my last comment, when I choose optitrack as the mocap_type, the launch file runs correctly, but the LED light is solid red, and the crazyflie don’t respond when I run the python script. 

Also, in the RVIZ window, there is nothing until I launch the vrpn_client _ros sample.launch, is this normal ?

whoenig, 2019-06-11:
`/.mcr` should actually output a lot of data (position of all your drones/markers for every frame). If you only see Natnet version and Motive version, then motive is not streaming data, or there is some issue with the multicast reception. Is there any firewall between the Motive computer and the crazyswarm computer? Did you enable the data streaming in Motive?

What you describe is not normal, but simply caused by the connection issues you are facing.

lkyfly0765, 2019-06-11:
@whoenig 
I have shutdown the firewall both Motive computer and my laptop. And I enable the data streaming on the data streaming panel. But because we use vrpn before, I always turn on the _broadcast vrpn data option_ in that panel, will this cause some problem? Anyway, I will check the firewall and the data streaming settings again later.

By the way, we use the motionCapture as the object tracking type, does the Natnet support it ? Or we can only use Natnet when we choose the libobjecttracker?

I have updated the Motive to the latest version 2.1.1 final, as you said in other issues, the latest version of the crazyswarm support Natnet 3.0, I think the Motive version is not the problem causer.

jpreiss, 2020-06-13:
@lkyfly0765 were you able to resolve this?

lowjunen, 2020-10-21:
I had a similar issue and the fix was to switch the local interface option on motive. The labeling is a little misleading in that the option 'loopback' really means that the optitrack is broadcasting only to the machine the is running motive (i.e itself) whereas there are also setups where the motive machine is separate from the ground station machine. 

I believe this issue might also resolve #155 #152

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss