USC-ACTLab/crazyswarm #158: it seems that crazyflie can't receive pose information quickly

zhongqianyang, 2019-06-09:
Hi,good morning.
we have done a lot of work to try to make the crazyflie fly,and we did it today.
So  we have flash firmware well.
the crazyflie can fly,and it can fly to a certain point well.
but when i start hover_swarm.launch in a terminal,i got output like this
`[ WARN] [1560104331.566231056]: No updated pose for motion capture object cf3
[ WARN] [1560104331.577892009]: No updated pose for motion capture object cf3
[ WARN] [1560104331.601861944]: No updated pose for motion capture object cf3
[ WARN] [1560104331.613605907]: No updated pose for motion capture object cf3
[ WARN] [1560104331.625813694]: No updated pose for motion capture object cf3
[ WARN] [1560104331.637460265]: No updated pose for motion capture object cf3
[ WARN] [1560104331.649289836]: No updated pose for motion capture object cf3
[ WARN] [1560104331.661144378]: No updated pose for motion capture object cf3
`
and then execute figure8_csv.py ,the crazyflie will not Change its direction smoothly，
i think it needs update crazyflies pose quickly, i have read many related issue,but i can't figure it out.so i ask for your help!thanks a lot!

whoenig, 2019-06-09:
Which motion capture system are you using? Are you tracking with the motion capture or using libobjecttracker?
For your mocap, how is your network setup (i.e., make sure that you have a physcal ethernet connection between your Ubuntu PC and the PC that's running your mocap software, and not just connect using Wifi).

luozhexin, 2019-06-10:
Hi,I'm sorry I didn't make it clear.i use optitrack system,and i track with motion capture.we have a physcal connection between two computer.For example,i use a python file to make crazyflie flying a circle,but it will fly in a The line chart because of `[ WARN] [1560104331.566231056]: No updated pose for motion capture object cf3 `i guess.
i run hover_swarm.launch and get crazyflie's pose in rviz directly.
i have check the link quality using crazyflie_tools,the output state the link quality is good.
so,i guess some probable cause
1. is this the /tf  problem?
2.our optitrack's broadcast frequency is 120HZ,does it too quick for the crazyswarm,so it's listmemory flie can't process so much data?
3.our crazyflies' address is //0/99/2M,E7E7E7E701~08,when we make single crazyflie flying,i change allcrazyflies.yaml , i did't add the number because when i do that hover_swarm.launch can't work fine,the crazyswarm_server will died,i don'n know why, so this may represent some deep problem.
thank you for responding!f

zhongqianyang, 2019-06-10:
@whoenig  Hi,I'm so sorry I didn't make it clear.i use optitrack system,and i track with motion capture.we have a physcal connection between two computer.For example,i use a python file to make crazyflie flying a circle,but it will fly in a The line chart because of `[ WARN] [1560104331.566231056]: No updated pose for motion capture object cf3 `i guess.
i run hover_swarm.launch and get crazyflie's pose in rviz directly.
i have check the link quality using crazyflie_tools,the output state the link quality is good.
so,i guess some probable cause
1. is this the /tf  problem?
2.our optitrack's broadcast frequency is 120HZ,does it too quick for the crazyswarm,so it's listmemory flie can't process so much data?
3.our crazyflies' address is //0/99/2M,E7E7E7E701~08,when we make single crazyflie flying,i change allcrazyflies.yaml , i did't add the quantity of crazyflies because when i do that hover_swarm.launch can't work fine,the crazyswarm_server will died.i don'n know why, so this may represent some deep problem.
thank you for responding!

closed 2019-06-11 by zhongqianyang

whoenig, 2019-06-11:
@zhongqianyang How did you fix your problem? Perhaps this will be helpful for somebody who is reading this later.