USC-ACTLab/crazyswarm #159: How to use phase space as the mocap system

zehuilu, 2019-06-18:
Hi guys,

I want to use phase space as the motion capture system. I know I need to enable the phase space in libmotioncapture/CMakeLists.txt. But I don't know what I should put into libmotioncapture/externalDependencies/ and what else I should do after I revise the launch file. Should I put an official SDK in externalDependencies or a ROS package for phase space?

I will appreciate your help very much!

whoenig, 2019-08-12:
It should be the official C++ SDK for phasespace. Note that when we tested it last, we just had some Beta version of the SDK, so things might have changed in the meantime.

zehuilu, 2019-08-12:
@whoenig Thanks for the reply! I will let you know if we have anything updated.

jpreiss, 2020-06-13:
@zehuilu were you able to resolve this?

zehuilu, 2020-06-15:
@jpreiss Hi James, I wasn't able to resolve this problem. I switched to other types of quadrotors and built my controller. Feel free to close this issue :)

jpreiss, 2020-06-15:
Thanks for replying!

closed 2020-06-15 by jpreiss