USC-ACTLab/crazyswarm #161: ./build.sh does not work with gcc-arm-none-eabi

JonathanPlasse, 2019-06-25:
Hello,
I tried to launch `./build.sh`.
There was a build error.
Then I uninstall gcc-arm-none-eabi and install gcc-arm-embedded  instead.
It worked.
Maybe, the documentation should be updated to recommend using gcc-arm-embedded as gcc-arm-none-eabi is obselete.

whoenig, 2019-06-27:
Thanks. Which operating system do you use?

For travis CI we use gcc-arm-embedded: https://github.com/USC-ACTLab/crazyswarm/blob/master/.travis.yml#L89-L92.

Later compiler versions have to be downloaded directly from ARM: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

JonathanPlasse, 2019-06-27:
I use Ubuntu 16.04.
I tried to use gcc-arm-embedded because of `.travis.yml`.

jpreiss, 2020-02-25:
Per https://www.bitcraze.io/docs/crazyflie-firmware/master/build_instructions/ .

closed 2020-02-25 by jpreiss