USC-ACTLab/crazyswarm #162: Unable to log thrust controlling CF from script

cf11111, 2019-06-26:
How can I log the estimated thrust while running the Crazyflie 2.0 from a python script?  I know a log block in the client combines different sensor readings to estimate thrust, but is there a log TOC variable I can use to log the thrust?

whoenig, 2019-06-27:
You can log `motor.m1` to `motor.m4` which is a value between 0 and 65536, where 0 <-> 0g and 65536 <-> 15g, see https://github.com/bitcraze/crazyflie-firmware/blob/252bdb173cac59be859b06c19796eee22a406f90/src/drivers/src/motors.c#L200-L222 and https://github.com/bitcraze/crazyflie-firmware/blob/5d0f8d3fa7817b922b143e442aed9db4e0800d40/src/modules/src/power_distribution_stock.c#L118-L123.

You can setup logging in `hover_swarm.launch`. Note that you can only control around 4 CFs per radio if you enable logging.

cf11111, 2019-06-27:
Thank you so much for the reply.  65536 maps to 15g and not to 60g as it states in the code?

cf11111, 2019-06-27:
Oh sorry I understand now that you are saying for each motor.  Thank you.

closed 2019-07-01 by cf11111