USC-ACTLab/crazyswarm #163: Confusion about log TOC variables

cf11111, 2019-06-27:
Sorry to start a new thread on this, I tried to look it up in other threads but am still confused by the variables in the logTOC.  

1. Are the stateEstimate.roll, pitch, and yaw the rate values (deg/s)?  

2. Also I understand that these stateEstimate values incorporate mocap data into the estimation (I am using estimator 3 fyi) -- is there a way to get position and orientation estimates without using mocap data?

[image]

cf11111, 2019-06-27:
I expected to see variables like acc.x and gyro.x available to be logged, but they do not appear in my TOC.

whoenig, 2019-07-02:
This does not seem to be related to the Crazyswarm. Please ask in the official support forum (https://forum.bitcraze.io/).

closed 2019-07-02 by whoenig