USC-ACTLab/crazyswarm #164: Crazyflie ROS waypoint following

harrafa, 2019-07-17:
I am implementing ROS for waypoint following with aruco tag. 
When I run the controller.cpp at frequency greater than 8Hz, the function transformpose() (to transform world reference points to crazyflie frame) forces the controller node to exit with the error:
 Quaternion malformed, magnitude: 0 should be 1.0
Interestingly, I am able to run the controller at frequencies less than 5Hz.
Could you please take a look at my launch file and controller.cpp for any mistakes I a
[image]
[image]
[image]
[image]

m making. Eagerly awaiting your response.
Thank you.

whoenig, 2019-07-17:
You should not use any version of crazyflie_controller. Instead, use the goTo function which uses the onboard attitude and position controller, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/waypoints.py for an example.

harrafa, 2019-07-17:
Thank you for your timely response. Are you suggesting I install the crayswarm repo instead of crazyflie_ros?

whoenig, 2019-07-17:
You posted this question on the crazyswarm repository, so I assumed that's what you are using. For crazyflie_ros there is an example on how to use the HL mode here: https://github.com/whoenig/crazyflie_ros/blob/master/crazyflie_demo/scripts/test_high_level.py. You can also publish commands to the cmd_position topics instead in order to use the on-board position controller.

harrafa, 2019-07-17:
I am sorry. I will take it off from here.

closed 2019-07-17 by harrafa