USC-ACTLab/crazyswarm #166: Server dies with optitrack rigid body

KYLiu0620, 2019-08-11:
Hi, I use crazyswarm with Optitrack MOCAP. Motive version is 1.10.3.
When I choose optitrack for motion_capture_type and motionCapture for object_tracking_type, the server dies with exit code -11 after random frames as follows:[image]. However, it correctly works with vrpn, and I test figure 8 flight successfully.
I also find if I turn off the streaming of rigid bodies, the server doesn't die with optitrack mocap type. I have cloned the latest code and update the firmware of crazyflies and crazyradio PA. Does anyone have the same problem? Thanks.

Best

whoenig, 2019-08-12:
Hi, We only tested with Motive 2.0. The error indicates a segfault. Could you test if the crossplatform build of the NatNetSDK works for you? You can find instructions and repo here: https://github.com/whoenig/NatNetSDKCrossplatform.

KYLiu0620, 2019-08-12:
Hi,

Thanks for your reply so quickly. I use Ubuntu 16.04 with Boost version 1.58.0, and it is too old to build that. Maybe I need to upgrade and test later.
I have read some previous issues about Optitrack, and I thought this package only worked with Motive 1.10.3. So is it compatible with Motive 2.0 now?
Thanks again!

Best

whoenig, 2019-09-16:
The package works with older Ubuntu as well, but you need to change the CmakeLists.txt to allow using older boost versions.
Yes, newer versions of the package support Motive 2.x.

KYLiu0620, 2019-09-30:
Hi,

Sorry for late reply @whoenig . I just worked with vrpn and motionCapture as object_tracking_type but faced some issues, and thus I decide to use optitrack as motion_capture_type now.
 I have already updated Motive to 2.0.0.4 which is the latest and available for my license, but the error still occurs. I also test NatNetSDKCrossplatform with boost 1.58, and it works very well. Any suggestions? Thanks for your help!

whoenig, 2019-09-30:
Could you post the output of NatNetSDKCrossplatform? I had cases before where the output was garbage (which could lead to a segfault in the crazyswarm_server). For me, resetting the streaming settings in Motive helped. If that doesn't work, I can give you instructions on how to run with a debugger so that we can pinpoint where the crash is triggered.

KYLiu0620, 2019-09-30:
I place 4 markers on 1 CF at the origin, and here is the output of NatNetSDKCrossplatform:
```
Begin Packet
-------
Message ID : 7
Byte count : 190
Frame # : 9878009
Marker Set Count : 1
Model Name: all
Marker Count : 0
Unidentified Marker Count : 0
Rigid Body Count : 0
Skeleton Count : 0
Labeled Marker Count : 4
ID  : [MarkerID: 50025] [ModelID: 0]
pos : [0.05,-0.05,0.01]
size: [0.01]
err:  [0.00]
ID  : [MarkerID: 50251] [ModelID: 0]
pos : [0.01,0.06,0.03]
size: [0.01]
err:  [0.00]
ID  : [MarkerID: 50252] [ModelID: 0]
pos : [0.01,0.01,0.02]
size: [0.02]
err:  [0.00]
ID  : [MarkerID: 50259] [ModelID: 0]
pos : [0.01,-0.03,0.02]
size: [0.02]
err:  [0.00]
Timestamp : 82316.742
Mid-exposure timestamp : 0
Camera data received timestamp : 28182049965
Transmit timestamp : 28182051017
End Packet
-------------
```
On the other hand, I'm trying to change libobjecttracker as object_tracking_type, and I follow the documentation to run mocap_helper.launch to configure marker arrangement. Then I get the output of mocap_helper.launch as follows:
```
frame 307:
    points:
    objects:
      "":
         occluded: 0
         position: [0.0498469, -0.0521146, 0.0107379]
         rotation: [1.16812e+16, 0.0144743, -149504, -812.901]
      "":
         occluded: 0
         position: [0.0102123, -1.01782e-25, -816.904]
         rotation: [1.15594e-14, 1.18525e+32, -6.04814e-23, -2.31558e-35]

```
I place the markers as before, but the output is strange. I didn't define any rigid body, and I also unchecked the streaming option for rigid bodies which is mentioned in the documentation. I'm not sure whether the problems are related to each other. Thanks for your quick reply!

whoenig, 2019-09-30:
If you want to use optitrack for tracking, make sure you actually stream rigid bodies (and not markers). If you want to use libobjecttracker, make sure that you disable the assets in Motive (i.e., you should only see a pointcloud it Motive). Your output looks like you have a rigid body enabled in Motive, but simply do not stream it (a scenario that I did not test).

KYLiu0620, 2019-09-30:
For libobjecttracker case, I further turn off the streaming option 'Solver Replaces Occlusion' which is turned on by default and then get the reasonable output from mocap_helper.launch. I only see that option in Motive 2.0.0.4. Hope this information can help somebody else.
For motionCapture case, I turn on the streaming options 'Solver Replaces Occlusion' and 'Rigid Bodies', and the error remains. The output of NatNetSDKCrossplatform in my last post seems to be normal, so maybe I need further instructions about this. Thanks a lot!

jpreiss, 2020-06-13:
@KYLiu0620 were you ever able to resolve this?

closed 2020-06-20 by KYLiu0620