USC-ACTLab/crazyswarm #167: Cannot Connect to second Crazyflie

timchu93, 2019-08-15:
Hi,

I'm currently working on swarming with the LPS. I followed the steps in the documentation and i would get this error when I try to apply the second crazyflie into the system

```
[crazyswarm_server-1] process has died [pid 5086, exit code -6, cmd /home/timchu/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/timchu/.ros/log/6e51621c-bf35-11e9-9773-94b86df2f120/crazyswarm_server-1.log].
log file: /home/timchu/.ros/log/6e51621c-bf35-11e9-9773-94b86df2f120/crazyswarm_server-1*.log
```

Here are some changes that I made to the file

in **allCrazyflies.yaml**

crazyflies:
  - id: 1
    channel: 100
    initialPosition: [0.6, 0.3, 0.0]
    type: default

  - id: 2
    channel: 110
    initialPosition: [0.6, 0.9, 0.0]
    type: medium

in **hover_swarm.launch**
```

firmware: "bitcraze" # one of "crazyswarm", "bitcraze"
estimator: 2 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
controller: 2 # 1: PID, 2: mellinger
motion_capture_type: "none" # one of none,vicon,optitrack,qualisys,vrpn
object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
vicon_host_name: "vicon"
```
I'm still unsure where the problem is. I appreciate any suggestions

whoenig, 2019-08-17:
Since you use two different channels, I assume you have two Crazyradio PA's plugged in?
If that is not the problem, could you post the full output of the Terminal (where you execute roslaunch)?

timchu93, 2019-08-20:
Hi Whoenig,

I'm actually just using 1 Crazyradio PA. Should I use the same channel but with different address?

here's the full post of the output:

```
SUMMARY
========

PARAMETERS
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/default/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/default/markerConfiguration: 0
 * /crazyflieTypes/large/batteryVoltageWarning: 11.4
 * /crazyflieTypes/large/batteryVoltateCritical: 11.1
 * /crazyflieTypes/large/bigQuad: True
 * /crazyflieTypes/large/dynamicsConfiguration: 0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/i_range_z: 4.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_xy: 30000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kR_z: 40000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_omega_rp: 100
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_xy: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kd_z: 3.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_xy: 1.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/ki_z: 2.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_xy: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kp_z: 8.0
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_xy: 9000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/kw_z: 10000
 * /crazyflieTypes/large/firmwareParams/ctrlMel/mass: 0.38
 * /crazyflieTypes/large/firmwareParams/ctrlMel/massThrust: 5400
 * /crazyflieTypes/large/markerConfiguration: 2
 * /crazyflieTypes/medium/batteryVoltageWarning: 7.6
 * /crazyflieTypes/medium/batteryVoltateCritical: 7.4
 * /crazyflieTypes/medium/bigQuad: True
 * /crazyflieTypes/medium/dynamicsConfiguration: 0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/i_range_z: 2.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_xy: 50000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_xy: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/kd_z: 1.0
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_xy: 0.1
 * /crazyflieTypes/medium/firmwareParams/ctrlMel/ki_z: 0.5
[... 114 lines cut ...]
update ctrlMel/i_range_z to 0.4
update ctrlMel/kR_xy to 70000
update ctrlMel/kR_z to 60000
update ctrlMel/kd_omega_rp to 200
update ctrlMel/kd_xy to 0.2
update ctrlMel/kd_z to 0.4
update ctrlMel/ki_m_z to 500
update ctrlMel/ki_xy to 0.05
update ctrlMel/ki_z to 0.05
update ctrlMel/kp_xy to 0.4
update ctrlMel/kp_z to 1.25
update ctrlMel/kw_xy to 20000
update ctrlMel/kw_z to 12000
update ctrlMel/mass to 0.032
update ctrlMel/massThrust to 132000
[ INFO] [1566302154.475198330]: [cf1] Update parameters
[ INFO] [1566302154.572431191]: CF Console: CONTROLLER: Using Mellinger (2) controller
[ INFO] [1566302154.587893113]: Update params: 0.155312 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  No matching USB Device with devid = 1 found!
[crazyswarm_server-2] process has died [pid 3553, exit code -6, cmd /home/timchu/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/timchu/.ros/log/75b58140-c341-11e9-8e86-94b86df2f120/crazyswarm_server-2.log].
log file: /home/timchu/.ros/log/75b58140-c341-11e9-8e86-94b86df2f120/crazyswarm_server-2*.log
```

Thank you for your response

whoenig, 2019-08-20:
The Crazyswarm requires one Crazyradio (PA) per channel. So if you want to use a single radio, you should use the same channel. In either case, all drones must have unique addresses.

timchu93, 2019-08-21:
Hi whoenig,

I was able to connect to both crazyflies and was able to let them hover. Thank you.

Currently I'm working with the waypoints.py trying to get the 2 cfs to make a square path, however only CF# 1 goes to its designated position and then the python file closes, showing this error
```

[CF# 1 at 2.0 s. [0.0, 0.0, 0.5], CF# 2 at 2.0 s. [0.3, 0.3, 0.75], CF# 1 at 4.0 s. [0.0, 0.6, 0.5], CF# 2 at 4.0 s. [0.3, 0.9, 0.75], CF# 1 at 6.0 s. [0.6, 0.6, 0.5], CF# 2 at 6.0 s. [0.6, 0.9, 0.75], CF# 1 at 8.0 s. [0.6, 0.0, 0.5], CF# 2 at 8.0 s. [0.6, 0.3, 0.75], CF# 1 at 10.0 s. [0.0, 0.0, 0.5], CF# 2 at 10.0 s. [0.3, 0.3, 0.75]]
Traceback (most recent call last):
  File "waypoints1.py", line 71, in <module>
    cf = allcfs.crazyfliesById[waypoint.agent]
KeyError: 2
```
so far I only made changes in the waypoints.csv file not nothing in the python file. Do I need to make any changes to the python file? I understand that this is a different issue, should I close this one and open another discussion for this?

whoenig, 2019-09-16:
Opening a new issue would be best. The error indicates that you do not have a Crazyflie with ID 2?

timchu93, 2019-09-24:
Hi whoenig,

I'll close this issue as it is already resolve and opening a different one

closed 2019-09-24 by timchu93