USC-ACTLab/crazyswarm #169: Waypoints for Second Crazyflie

timchu93, 2019-09-24:
I was able to connect to both crazyflies and was able to let them hover. Thank you.

Currently I'm working with the waypoints.py trying to get the 2 cfs to make a square path, however only CF# 1 goes to its designated position and then the python file closes, showing this error

[CF# 1 at 2.0 s. [0.0, 0.0, 0.5], CF# 2 at 2.0 s. [0.3, 0.3, 0.75], CF# 1 at 4.0 s. [0.0, 0.6, 0.5], CF# 2 at 4.0 s. [0.3, 0.9, 0.75], CF# 1 at 6.0 s. [0.6, 0.6, 0.5], CF# 2 at 6.0 s. [0.6, 0.9, 0.75], CF# 1 at 8.0 s. [0.6, 0.0, 0.5], CF# 2 at 8.0 s. [0.6, 0.3, 0.75], CF# 1 at 10.0 s. [0.0, 0.0, 0.5], CF# 2 at 10.0 s. [0.3, 0.3, 0.75]]
Traceback (most recent call last):
  File "waypoints1.py", line 71, in <module>
    cf = allcfs.crazyfliesById[waypoint.agent]
KeyError: 2

The CF with ID# 2 is powered on and is specified in the allCrazyflies.yaml file and is able to hover through the hover_swarm.launch file

Thanks in advance

whoenig, 2019-09-26:
Is CF2 enabled using chooser.py (or in other words, is it included in crazyflies.yaml)?

whoenig, 2019-10-18:
You should not manually edit crazyflies.yaml. Simply restart chooser.py after your edits in allCrazyflies.yaml and select all CFs that you want to use.

timchu93, 2019-10-20:
Hi whoenig,

Did as you suggested, got the system working. Thank you very much!

closed 2019-10-20 by timchu93