USC-ACTLab/crazyswarm #170: How to get tdoa data in ROS?(LPS)

Supredan, 2019-09-26:
According to the steps in [this issue](https://github.com/USC-ACTLab/crazyswarm/issues/108),I successfully used LPS to fly one CF, the figure8 script and teleop with xbox360 all succeeded.but i have two problems:
1.when i try to fly two CFs,I changed the crazyfile.yaml to
```
crazyflies:
  - id: 2
    channel: 70
    initialPosition: [0, -0.5, 0.0]
    type: default
  - id: 4
    channel: 70
    initialPosition: [0, 0.5, 0.0]
    type: default
```
then i run "roslaunch crazyswarm hover_swarm.launch",Showing the following error:
```
[ INFO] [1569498173.496024911]: [cf2] Update parameters
[ INFO] [1569498173.560001237]: CF Console: CONTROLLER: Using controller 2
[ INFO] [1569498173.573249559]: Update params: 0.105149 s
terminate called after throwing an instance of 'std::runtime_error'
  what():  No matching USB Device with devid = 1 found!
[crazyswarm_server-1] process has died [pid 21858, exit code -6, cmd /home/supdan/bin/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/supdan/.ros/log/62172b06-e043-11e9-8b9b-5cea1d064bc7/crazyswarm_server-1.log].
log file: /home/supdan/.ros/log/62172b06-e043-11e9-8b9b-5cea1d064bc7/crazyswarm_server-1*.log
```
Any suggestions for resolution?
2.The main purpose of my use of crazyswarm is to get tdoa data in real time.I noticed that here[https://github.com/USC-ACTLab/crazyswarm/blob/ab0fe73a05b6e947d5651c411634eb49a3862efc/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L13](https://github.com/USC-ACTLab/crazyswarm/blob/ab0fe73a05b6e947d5651c411634eb49a3862efc/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L13)
Able to generate a rostopic,so i want to change it to "genericLogTopic_log1_Variables:["tdoa3.tdoa"]",Then you should generate a ros node that sends data in real time.
but when i run "rostopic list"in another terminal(Already running hover_swarm.launch),I can't see log1,here is the result:
```
$ rostopic list
/cf4/cmd_full_state
/cf4/cmd_position
/cf4/cmd_stop
/clicked_point
/diagnostics
/initialpose
/joy
/joy/set_feedback
/move_base_simple/goal
/pointCloud
/rosout
/rosout_agg
/tf
/tf_static
/virtual_interactive_object
```
Is my method feasible?If not, is there any suggestion?
Thanks in advance!

whoenig, 2019-09-26:
1. You need one Crazyradio per channel. Perhaps your crazyflies.yaml does not match exactly what you posted here? The error indicates that there is at least one other Crazyflie that the server tries to connect to with a different channel.
2. This is feasible. You will need to change `enable_logging` to True in the launch file. Note that once you enable logging, you will not be able to control a large number of Crazyflies per radio (~2 - 4 should work). If you only need the data for post-processing, it is much better to use the uSD-card deck.

Supredan, 2019-09-28:
Hi whoenig,
thanks for your help.
now i have solve the first problem,which is caused by setting the coordinates in crazyfile.yaml to [0,0,0] instead of [0.0, 0.0 ,0.0].  -_-!
but the second problem still exists,i have done this: 
```
enable_logging: True
```
but when i run hover_swarm.launch , here is the error:
```
terminate called after throwing an instance of 'std::runtime_error'
  what():  Could not find ctrlStat.edist in log toc!
[crazyswarm_server-2] process has died [pid 8112, exit code -6, cmd /home/supdan/bin/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/supdan/.ros/log/5db9d21e-e1c6-11e9-8f14-5cea1d064bc7/crazyswarm_server-2.log].
log file: /home/supdan/.ros/log/5db9d21e-e1c6-11e9-8f14-5cea1d064bc7/crazyswarm_server-2*.log
```
if i change it to False, i can connect to two cfs,and run rostopic list, here is the result:
```
$ rostopic list
/cf3/cmd_full_state
/cf3/cmd_position
/cf3/cmd_stop
/cf4/cmd_full_state
/cf4/cmd_position
/cf4/cmd_stop
/clicked_point
/diagnostics
/initialpose
/joy
/joy/set_feedback
/move_base_simple/goal
/pointCloud
/rosout
/rosout_agg
/tf
/tf_static
/virtual_interactive_object
```
Thank you for your continued patient help！

Supredan, 2019-09-28:
After Upload the precompiled firmware, finally i see the log1:
```
$ rostopic list
/cf3/cmd_full_state
/cf3/cmd_position
/cf3/cmd_stop
/cf3/log1
/clicked_point
/diagnostics
/initialpose
/joy
/joy/set_feedback
/move_base_simple/goal
/pointCloud
/rosout
/rosout_agg
/tf
/tf_static
/virtual_interactive_object
```
then i will try to get the tdoa data ,thanks!

Supredan, 2019-09-28:
All right,Things are not so smooth.
I change hover_swarm.launch to this:
```
genericLogTopics: ["log1"]
genericLogTopicFrequencies: [10]
genericLogTopic_log1_Variables: ["tdoa3.logId", "tdoa3.logOthrId", "tdoa3.tdoa"]
```
which is according to here:
[https://github.com/bitcraze/crazyflie-firmware/blob/13aa1372bc3ba5b14922fc2f9d8999270882021a/src/deck/drivers/src/lpsTdoa3Tag.c#L334](https://github.com/bitcraze/crazyflie-firmware/blob/13aa1372bc3ba5b14922fc2f9d8999270882021a/src/deck/drivers/src/lpsTdoa3Tag.c#L334)
but when i run "rostopic echo /cf3/log1",the output seems not right:
```
---
header: 
  seq: 363
  stamp: 
    secs: 64
    nsecs: 788000000
  frame_id: ''
values: [0.0]
---
header: 
  seq: 364
  stamp: 
    secs: 64
    nsecs: 808000000
  frame_id: ''
values: [0.0]
---
header: 
  seq: 365
  stamp: 
    secs: 64
    nsecs: 848000000
  frame_id: ''
values: [0.0]
---
header: 
  seq: 366
  stamp: 
    secs: 64
    nsecs: 868000000
  frame_id: ''
values: [0.0]
---
header: 
  seq: 367
  stamp: 
    secs: 64
    nsecs: 888000000
  frame_id: ''
values: [0.0]
...............
```
Is there anything wrong with my steps? @whoenig 
Thanks in advance!

whoenig, 2019-09-30:
You can only read or set the current value of a parameter. For continuous streaming of data, the variable of interest needs to be a logging variable in the firmware. 

Your steps to look at the values for "tdoa3.tdoa" are correct. What do you want to do with the other two (i.e., the parameters)? Do you simply want to set them to a specific value at the beginning?

Supredan, 2019-09-30:
hi,thanks for your reply! 
From my understanding,the tdoa3.tdoa  data represents the time difference between two anchor reaching cf, But I also need to know which two anchors this time difference comes from, so that I can calculate the position of cf, right?Can only tdoa3.tdoa calculate the position of cf?thanks!

发自我的iPhone

whoenig, 2019-09-30:
The current values of the parameters are actually stored in the ROS parameter server and you can use `rosparam` and others to look at the value. You can also change parameters - easiest by using the setParam function in your high-level python scripts.

Supredan, 2019-10-05:
hi, @whoenig 
sorry to bother you again -_-!
Because I want to record a rosbag, so I chose to use tdoa2 (which all can be published with rostopic).i write this in launch file:
```    
      genericLogTopics: ["log1", "log2", "log3"]
      genericLogTopicFrequencies: [100, 100, 100]
      genericLogTopic_log1_Variables: ["stateEstimate.x","stateEstimate.y","stateEstimate.z"]
      genericLogTopic_log2_Variables: ["tdoa.d0-1", "tdoa.d1-2", "tdoa.d2-3", "tdoa.d3-4"]
      genericLogTopic_log3_Variables: ["tdoa.d4-5", "tdoa.d5-6", "tdoa.d6-7", "tdoa.d7-0"]
```
then i use
"$ rosbag record -O cf8_2 /cf3/log1 /cf3/log2 /cf3/log3"
here is the rosbag (it record the data of running figure8_csv.py):[cf8_2.bag.zip](https://github.com/USC-ACTLab/crazyswarm/files/3692857/cf8_2.bag.zip)
but It's not quite right,when i use cfclient to log the same data,the "tdoa.d3-4" represent the difference in distance to anchor 3 VS anchor 4, in meters.it looks like this:
```
Timestamp,tdoa.d3-4,tdoa.dist3-4
1101814,-0.20174585282802582,33557
1101914,0.07037646323442459,33513
1102014,-0.17359526455402374,33556
1102114,-0.21582114696502686,33548
1102214,-0.004691763781011105,33499
1102314,-0.1126023381948471,33557
1102414,-0.07037646323442459,33512
1102514,0.004691763781011105,33489
1102614,0.00938352756202221,33538
1102714,0.00938352756202221,33503
```
when i put the cf in the middle of anchor 3 and 4,Log data as shown above,"tdoa.d3-4" approximately equal to 0.but in rosbag or when i run "rostopic echo /cf3/log2" to get the data, it looks like this:
```
$ rostopic echo /cf3/log2
header: 
  seq: 247
  stamp: 
    secs: 51
    nsecs: 956000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
header: 
  seq: 248
  stamp: 
    secs: 52
    nsecs:  56000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
header: 
  seq: 249
  stamp: 
    secs: 52
    nsecs: 156000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
header: 
  seq: 250
  stamp: 
    secs: 52
    nsecs: 256000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
header: 
  seq: 251
  stamp: 
    secs: 52
    nsecs: 356000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
header: 
  seq: 252
  stamp: 
    secs: 52
    nsecs: 456000000
  frame_id: ''
values: [9327.0, 9976.0, 0.0, 0.0]
---
```
and this :
```
$ rostopic echo /cf3/log3
header: 
  seq: 288
  stamp: 
    secs: 56
    nsecs:  63000000
  frame_id: ''
values: [0.0, 0.0, 71.0, 6474.0]
---
header: 
  seq: 289
  stamp: 
    secs: 56
    nsecs: 163000000
  frame_id: ''
values: [0.0, 0.0, 71.0, 6474.0]
---
header: 
  seq: 290
  stamp: 
    secs: 56
    nsecs: 263000000
  frame_id: ''
values: [0.0, 0.0, 71.0, 6474.0]
---
header: 
  seq: 291
  stamp: 
    secs: 56
    nsecs: 363000000
  frame_id: ''
values: [0.0, 0.0, 71.0, 6474.0]
---
header: 
  seq: 292
  stamp: 
    secs: 56
    nsecs: 463000000
  frame_id: ''
values: [0.0, 0.0, 71.0, 6474.0]
---
```
when the cf take off, every "0.0" will turn to a value more than 50,000,and the [71.0, 6474.0]  [9327.0, 9976.0] almost still the same.
Do you have any advice?
thanks again!

jpreiss, 2020-06-13:
closing as this seems mostly related to the logging system rather than using the LPS with crazyswarm.

closed 2020-06-13 by jpreiss

easy98, 2020-12-05:
Hi,

Is there any way to log the crazyflie's position directly?

whoenig, 2020-12-21:
(This issue is closed - please open a new issue.)