USC-ACTLab/crazyswarm #171: ./build.sh error

Allenhanbo, 2019-09-30:
After I Installed the dependencies and cloned the repository into my workspace, then I run the ./build.sh, I meet some error like this:

./build.sh
arm-none-eabi-size cf2_nrf.elf
   text	   data	    bss	    dec	    hex	filename
  27320	    452	   2656	  30428	   76dc	cf2_nrf.elf
S110 Activated
BLE  Disabled
Built for platform cf2
  CLEAN_VERSION
  CC    usddeck.o
src/deck/drivers/src/usddeck.c: In function 'f_gets_without_comments':
src/deck/drivers/src/usddeck.c:239:5: error: array subscript has type 'char' [-Werror=char-subscripts]
     if (isspace(c)) {
     ^
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'usddeck.o' failed
make[1]: *** [usddeck.o] Error 1
Makefile:338: recipe for target 'build' failed
make: *** [build] Error 2

whoenig, 2019-09-30:
Your compiler for building the firmware is too old. You can find a newer compiler at https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads. (Make sure to uninstall the previous compiler version before installing the new one.)

Allenhanbo, 2019-09-30:
Hi, thanks for replying me. Could you provide the command to uninstall the previous complier version? I searched for a while, but I didn't find a right to uninstall it.

whoenig, 2019-09-30:
Depends on how you installed it. `sudo apt-get remove gcc-arm-none-eabi` or `sudo apt-get remove gcc-arm-embedded` on Ubuntu should do the trick.

Allenhanbo, 2019-09-30:
Thank you so much! I've already delete the old one. I download the newest version pack from the website you provide. After I unpacked the download, I can not open the arm-none-eabi-gcc file like this:
hc@hc-Lenovo:~/gcc-arm-none-eabi-8-2019-q3-update/bin$ arm-none-eabi-gcc
arm-none-eabi-gcc: fatal error: no input files
compilation terminated.

whoenig, 2019-09-30:
That's normal. You need to follow the readme.txt in the archive for installation instructions. Once successfully installed, ./build.sh should run successfully.

Allenhanbo, 2019-09-30:
I follow the readme.txt to add the path in bashrc file, but still can't install it.

Allenhanbo, 2019-09-30:
I install successfully. I run ./build.sh again, there's new error:
hc@hc-Lenovo:~/crazyswarm$ ./build.sh
arm-none-eabi-size cf2_nrf.elf
   text	   data	    bss	    dec	    hex	filename
  27320	    452	   2656	  30428	   76dc	cf2_nrf.elf
S110 Activated
BLE  Disabled
Built for platform cf2
  CLEAN_VERSION
  CC    list.o
  CC    tasks.o
  CC    queue.o
  CC    timers.o
  CC    heap_4.o
  CC    port.o
  CC    stm32f4xx_adc.o
  CC    stm32f4xx_dbgmcu.o
  CC    stm32f4xx_dma.o
  CC    stm32f4xx_exti.o
  CC    stm32f4xx_flash.o
  CC    stm32f4xx_gpio.o
  CC    stm32f4xx_i2c.o
  CC    stm32f4xx_iwdg.o
  CC    stm32f4xx_rcc.o
  CC    stm32f4xx_spi.o
  CC    stm32f4xx_tim.o
  CC    stm32f4xx_usart.o
  CC    stm32f4xx_misc.o
  CC    stm32f4xx_syscfg.o
  CC    usb_core.o
src/lib/STM32_USB_OTG_Driver/src/usb_core.c: In function 'USB_OTG_WritePacket':
src/lib/STM32_USB_OTG_Driver/src/usb_core.c:183:7: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Werror=attributes]
       USB_OTG_WRITE_REG32( fifo, *((__packed uint32_t *)src) );
       ^~~~~~~~~~~~~~~~~~~
src/lib/STM32_USB_OTG_Driver/src/usb_core.c: In function 'USB_OTG_ReadPacket':
src/lib/STM32_USB_OTG_Driver/src/usb_core.c:208:5: error: 'packed' attribute ignored for type 'uint32_t *' {aka 'long unsigned int *'} [-Werror=attributes]
     *(__packed uint32_t *)dest = USB_OTG_READ_REG32(fifo);
     ^
cc1: all warnings being treated as errors
tools/make/targets.mk:26: recipe for target 'usb_core.o' failed
make[1]: *** [usb_core.o] Error 1
Makefile:338: recipe for target 'build' failed
make: *** [build] Error 2

Allenhanbo, 2019-09-30:
Thanks I've already build successfully.

closed 2019-10-04 by whoenig