USC-ACTLab/crazyswarm #172: random timeout Error on upload_trajectory

sadransh, 2019-10-16:
I randomly get such error when trying to run demo with multiple crazyflies. Some times with trying to re run the command (python figure8_csv.py) drones start flying.
 Do you have guess what may cause this happen? and how can I solve it?  the timeout happens sometimes on cf4 and randomly changes to cf3.

[image] 

**OS :** Ubuntu 16.04 
**System :** 
- crazyflies and radio are updated to latest master commit.

- working on VRPN based system the cameras are Motion Analysis. 

**Warnings recieved :**

- I get warning "No updated pose for motion capture object %s". I think the reason is different data structure which Motion Analysis software is sending. 

- sometimes get high latency warning (very rare)

whoenig, 2019-10-17:
Can you please try changing https://github.com/whoenig/crazyflie_cpp/blob/master/src/Crazyflie.cpp#L1268 to `if (false) {`? Then recompile using `catkin_make` in the `ros_ws` folder. I have seen this kind of behavior before too, and I was able to fix it that way. I think it is related to an issue on the firmware side where sometimes packets still get dropped.

The warnings are normal, as long as they don't pop-up constantly.

sadransh, 2019-10-17:
Haven't seen the error yet. ( with 8 tests ) 
I'll close the issue for now. 
Thanks a lot.

I ll be happy if you show me any possible problems or developements needed on this project that I can work on it.

closed 2019-10-17 by sadransh

whoenig, 2019-10-18:
Thanks for the feedback. PRs are always welcome. Just start using it and I am sure you will run into something that can be improved. Thanks!