USC-ACTLab/crazyswarm #173: hover_swarm.launch failed

JIMMMY0, 2019-10-18:
**I have already gone through the “Flying Multiple UAVs Using ROS”, Springer Book on Robot Operating System (ROS), 2017. and succeeded trying all the mentioned examples in that pdf. However, when I tried to launch the hover_swarm.launch  example, I encountered the following error:**

```bash
NODES
  /
    crazyswarm_server (crazyswarm/crazyswarm_server)
    crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
    joy (joy/joy_node)
    rviz (rviz/rviz)

ROS_MASTER_URI=http://192.168.50.250:11311

process[crazyswarm_server-1]: started with pid [5244]
process[joy-2]: started with pid [5245]
process[crazyswarm_teleop-3]: started with pid [5250]
process[rviz-4]: started with pid [5264]
[ INFO] [1571402262.035176274]: Wait for services...
[ INFO] [1571402262.052925946]: Manager ready.
[ERROR] [1571402262.114015802]: Couldn't open joystick force feedback!
[ INFO] [1571402262.114136138]: Opened joystick: /dev/input/js0. deadzone_: 0.050000.
ch: 1
[FATAL] [1571402262.180151112]: ASSERTION FAILED
	file = /home/jimmmy/crazyswarm/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp
	line = 1094
	cond = pos[j].getType() == XmlRpc::XmlRpcValue::TypeDouble

[crazyswarm_server-1] process has died [pid 5244, exit code -5, cmd /home/jimmmy/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/jimmmy/.ros/log/a2a3df30-f19a-11e9-9341-10feed2090c5/crazyswarm_server-1.log].
log file: /home/jimmmy/.ros/log/a2a3df30-f19a-11e9-9341-10feed2090c5/crazyswarm_server-1*.log
[rviz-4] process has finished cleanly
log file: /home/jimmmy/.ros/log/a2a3df30-f19a-11e9-9341-10feed2090c5/rviz-4*.log
```

**Tried to figure out for several days but still cannot solve it.
Here is my launch file:**

```xml
<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "FFE7E7E7E7"
      world_frame: "/world"
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      # genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      # genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      # genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      # genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      # genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 3 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "vicon" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "DESKTOP-SHNE1DP"
      vicon_host_name: "192.168.50.3:801"
      # optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_x" args="/cf2/log1/values[0]"/> -->
  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_roll" args="/cf1/log1/values[2] /cf1/log1/values[3]"/> -->

</launch>
```

**I didn't change the crazyflieTypes.yaml.
My crazyflie.yaml is** 
```yaml
crazyflies:
- channel: 80
  id: 1
  initialPosition: [0, 0.5, 0]
  type: default
```
**Thanks for the help!**

whoenig, 2019-10-29:
You might need to change the initialPosition to use double values, i.e. "[0.0, 0.5, 0.0]".

lkyfly0765, 2019-11-08:
@whoenig 
Hi whoenig, I have the same error with @JIMMMY0 ，

> [ERROR] [1573218738.212674883]: Couldn't open joystick force feedback!

I just upgrade my crazyswarm package to the latest version, and I met this new error. Could you please help me to find why there is a force feedback error? I never met this before.
[ERROR] [1573218738.212674883]: Couldn't open joystick force feedback!

Thanks!

whoenig, 2019-11-11:
I think this is a warning and not an error - I get that as well. It might be related to a newer version of the joy package.

lkyfly0765, 2019-11-18:
Thank you, whoenig!
However, I still have other issues. Could you please help me to solve it in #122 ?

JIMMMY0, 2019-12-08:
Thanks for your time @whoenig on this tiny problem that I ignored. Others problems related to the launch have also been solved. Other debug methods  include checking the mapping of the keys of the xbox controller, parameter setting "enHighLevel", try adding an LED ring for checking the connection, make sure the actual positions are close enough to the set initial position and make sure the markers configuration are unified.

closed 2019-12-08 by JIMMMY0