USC-ACTLab/crazyswarm #174: Battery command unresponsive using latest firmware

williamdwarke, 2019-10-28:
Hello -

Upon uploading the latest stm32 and nrf51 firmware to my lab's crazyflies, the crazyflie_tools battery function (as used by chooser.py) has ceased to function. For example, 

`rosrun crazyflie_tools version --uri radio://0/80/2M/E7E7E7E701`

works, but 

`rosrun crazyflie_tools battery --uri radio://0/80/2M/E7E7E7E701`

times out every time. So far it's not a debilitating issue, but it would be nice to have it back for convenience. Thanks for your help!

whoenig, 2019-10-28:
I noticed that as well. A workaround for now is to go back to a version before https://github.com/bitcraze/crazyflie2-nrf-firmware/commit/539ac3ca5785773a4a213ed8dcc05e849ed7d07d for the NRF51 firmware. It should be fine to use the latest STM32 firmware.

williamdwarke, 2019-10-28:
Thanks for the quick response! I just tried rolling back to 596410f and now nothing works so it must have something to do with the STM32 firmware also. Any idea of how far I can roll back the STM32 firmware without affecting compatibility with the lastest version of crazyswarm? Otherwise I can just leave it be for now until Bitcraze gets it fixed on their end.

whoenig, 2019-10-29:
I tested https://github.com/bitcraze/crazyflie-firmware/tree/febc78015a9b8a3908114f21a4d37f4f4a39a4fd for the STM firmware. If "nothing works" you might have had a bad flash. Make sure you either use `crazyflie_tools flash` or the `make flash` with the JTAG adapter (both verify the data they wrote). I have seen issues with `make cload` where the flashed image did not actually match what it should be.

williamdwarke, 2019-10-29:
Perfect - everything is working now with the nrf51 firmware commit you updated crazyswarm to reference. I also did a little bit of troubleshooting on my own that may be able to help you to maintain compatibility between crazyflie_tools and the latest firmware. It seems that the vbat bootloader request data is now being returned in the initial ack from the nrf51 instead of in a subsequent ping response. So, in crazyflie_cpp/src/Crazyflie.cpp, within `void Crazyflie::handleRequests`, the first `handleBatchAck(ack, crtpMode)`, which is just ensuring an ack for the request, is throwing out the pertinent data, causing the following wait loop to time out. This happens in the following 2 lines of the  do...while loop:
```
sendPacketInternal(request.request.data(), request.request.size(), ack, useSafeLink);
handleBatchAck(ack, crtpMode); <-- vbat is contained in ack.data[0] but ignored
```
I found a couple ways of getting the data out of it; one is a bit clunky, but the easiest way is to just send the request packet beforehand via `sendPacketOrTimeout(req)` to make sure one response gets picked up properly by the second `handleBatchAck(ack, crtpMode)` and it works pretty consistently. My vbat() function looks like this now:
```
float Crazyflie::vbat()
{
  if (m_radio) {
    crtpNrf51GetVBatRequest req;
    sendPacketOrTimeout(req);
    startBatchRequest();
    addRequest(req, 2);
    handleRequests(true);
    return getRequestResult<crtpNrf51GetVBatResponse>(0)->vbat;
  } else {
    return nan("");
  }
}
```
Anyways, hope this helps! I'd be interested in hearing your take on this issue.

whoenig, 2019-11-12:
Thanks for the investigation! It looks like safelink was disabled in the latest NRF firmware. I created an issue here: https://github.com/whoenig/crazyflie_cpp/issues/10 and should have a fix soon.

jpreiss, 2021-01-30:
closing as resolved.

closed 2021-01-30 by jpreiss