USC-ACTLab/crazyswarm #175: Problem controlling the second crazyflie

hugokohli, 2019-10-30:
Hi, 
I'm using 2 Crazflies 2.1 and I'm trying to make them fly together synchronously. 
For now my goal is very simple :
Take off, wait 2 sec, and land. 
I'm able to successfully control one drone alone (static flight). 

Once the two drones are connected (using 2 radios), I'm able to successfully control the first drone, but for the second one, it takes off, and goes in a random direction without staying static. 

I followed the instructions given in https://crazyswarm.readthedocs.io/en/latest/. 
The address of the first drone is "OxE7E7E701", using channel 80 of the first radio, and the second one is "OxE7E7E702", using channel 90 of the second radio.
It is not a hardware problem because each drone alone flies very well, and furthermore, I tried to exchange the addresses and it is always the drone with address "OxE7E7E702" which has problems...

I'm using "optitrack" as motion capture. 
The position of the rigid bodies is sent by motive, the first drone being named "cf1" with ID 1, and second drone being named "cf2" with ID2. The transmission seems to work as I can see the position of the two drones in RVIZ. Again, I know that it is not a problem of marker placement as each drone flies fine when used alone.

I used the hover_swarm.launch launch file (crazyswarm repository cloned the 28.10.19), that I modified a bit according to my needs (see below). The control of the node is done in the node "droneControl", which is launched automatically in the hover_swarm.launch. 

In the control node, I use the ROS services (/takeoff, /land). I choose which drone I want to fly by changing the string int the list DRONE_RIGIDBODY ("cf1", "cf2" or both). To be really clear : drone1 flies well when "cf1" is put, drone2 flies randomly when "cf2" is put.

I would really appreciate any advice, what am I doing wrong ? My code (hover_swarm.launch and droneControlNode.py) is linked just bellow. Thanks in advance !

hover_swarm.launch

´´´
<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js1" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      firmware: "crazyswarm" # one of "crazyswarm", "bitcraze"
      broadcast_address: "E7E7E7E701"
      world_frame: "/world"
      # genericLogTopics: ["log1"]
      # genericLogTopicFrequencies: [10]
      # Log the multi ranger deck distances on topic /cf1/log1:
      genericLogTopic_log1_Variables: ["range.front", "range.back", "range.up", "range.zrange", "range.left", "range.right"]
      # genericLogTopic_log1_Variables: ["ekfprof.usec_setup", "ekfprof.usec_innov", "ekfprof.usec_gain", "ekfprof.usec_corr", "ekfprof.usec_cov"]
      # genericLogTopic_log1_Variables: ["profiling.usec_ekf", "profiling.usec_traj", "profiling.usec_ctrl", "profiling.usec_idle"]
      # genericLogTopic_log1_Variables: ["stabilizer.x", "ctrltarget.x", "vicon.x", "stabilizer.z", "ctrltarget.z", "vicon.z"]
      # genericLogTopic_log1_Variables: ["stateEstimate.roll", "ctrlMel.rolld"]
      # genericLogTopic_log1_Variables: ["ctrlStat.edist"]
      # genericLogTopic_log1_Variables: ["stateEstimate.yaw", "ctrltarget.yaw", "stateEstimate.x", "ctrltarget.x"] #, "ctrltarget.x", "stateEstimate.roll", "ctrlMel.rolld"]
      firmwareParams: # for all (independent of type)
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman, 3: kalmanUSC (only crazyswarm firmware)
          controller: 1 # 1: PID, 2: mellinger
        #ring:
        #  effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
        #  solidBlue: 255 # if set to solid color
        #  solidGreen: 0 # if set to solid color
        #  solidRed: 0 # if set to solid color
        #  headlightEnable: 0
        #ekf:
        #  ext_var_xy: 1.5e-7 # 1e-7 # 1.5e-7
        #  ext_var_vel: 2e-4 # 2e-4
        #  ext_var_q: 4.5e-3 # 2e-3 # 4.5e-3
        #  gyro_var_xy: 0.2e-2 # 0.2e-4
        #  gyro_var_z: 0.2e-2 # 0.2e-4
        #  acc_var_xyz: 2.4e-3 # 2.4e-3
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      # vicon_host_name: "vicon"
      optitrack_host_name: "192.168.1.249"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      vrpn_host_name: "192.168.1.249"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: True
    </rosparam>
  </node>

  

  

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  
  

  

<node name="droneControlNode" pkg="crazyswarm" type="droneControlNode.py" output="screen">
</launch>

´´´

droneControlNode.py :
´´´
#!/usr/bin/env python

import numpy as np
# ROS
import rospy
from std_msgs.msg import String, Duration, Header
from geometry_msgs.msg import Point, Quaternion, PoseStamped, Pose
import tf.transformations

# Python
import time
from pynput import mouse
from math import sqrt
import threading

# Crazyflie
from crazyflie_driver.srv import *
from crazyflie_driver.msg import Position as PositionMsg

from pycrazyswarm import *
import uav_trajectory

DRONE_RIGIDBODY = ["cf1"]
FLYING = 0
LANDED = 1
TAKINGOFF = 2
LANDING = 3

TAKE_OFF_HEIGHT = 0.5# m
TAKE_OFF_DURATION = 3.0 # s

LAND_HEIGHT = 0.06 # m
LAND_DURATION = 3.0 # s

COMMAND_SEND_RATE = 20 # Hz
COMMAND_UPDATE_RATE = 10 # Hz

MOCAP_INIT_TIMEOUT = 10 # s

droneState = LANDED
targetPositionInitialized = False

def takeoffandland():
    global droneState
    if (droneState==LANDED) :
        droneState = TAKINGOFF
        time.sleep(TAKE_OFF_DURATION)
        droneState = FLYING
        time.sleep(0.0)
        droneState = LANDING
        time.sleep(LAND_DURATION)
        droneState = LANDED

def updateState(takeoffCommand, landCommand):

    global droneState

    takeoffDuration = rospy.Duration.from_sec(TAKE_OFF_DURATION)
    landDuration = rospy.Duration.from_sec(LAND_DURATION)

    rate = rospy.Rate(COMMAND_SEND_RATE)
    while(droneState != LANDED):      
        if (droneState == TAKINGOFF) :
            takeoffCommand(groupMask=0, height=TAKE_OFF_HEIGHT, duration=takeoffDuration)
        elif (droneState == LANDING) :
            landCommand(groupMask=0, height=LAND_HEIGHT, duration=landDuration)
        rate.sleep()

def controlDrone():
# Create node
    rospy.init_node('droneControlNode')

    rospy.loginfo("Initializing drone control node.")
    
    takeoff = {}
    #goTo = {}
    land = {}

    for droneRigidBody in DRONE_RIGIDBODY :
        rospy.wait_for_service(droneRigidBody + '/takeoff')
        #rospy.wait_for_service(droneRigidBody + '/go_to')
        rospy.wait_for_service(droneRigidBody +'/land')

        # Acquire reference to the high-level command services
        takeoff[droneRigidBody] = rospy.ServiceProxy(droneRigidBody+'/takeoff', Takeoff)
        #goTo[droneRigidBody] = rospy.ServiceProxy(droneRigidBody+'/go_to', GoTo)
        land[droneRigidBody] = rospy.ServiceProxy(droneRigidBody+'/land', Land)
        rospy.loginfo(droneRigidBody)

    threading.Thread(target = takeoffandland).start()
    for droneRigidBody in DRONE_RIGIDBODY :
    	threading.Thread(target=updateState, args=(takeoff[droneRigidBody], land[droneRigidBody])).start()
    
    rate = rospy.Rate(COMMAND_UPDATE_RATE)
    while not rospy.is_shutdown():
        rate.sleep()

if __name__ == '__main__':
    try:
        controlDrone()
    except rospy.ROSInterruptException:
        pass
´´´

whoenig, 2019-10-31:
Your launch file looks fine (although I recommend using the mellinger controller). For your script, can you try https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/niceHover.py? Your control node is overly complicated and you do execute takeoff/land in a loop, which might cause some strange side effects.

Which firmware are you using?

hugokohli, 2019-11-04:
Hi @whoenig ,
As you suggested, I tried with the script niceHover.py and the same behavior occurred for the second drone. I have to say that the command "allcfs.takeoff()" didn't work, i had to put it in a loop :
```
for cf in allcfs.crazyflies:
    cf.takeoff(targetHeight=Z, duration=1.0+Z)
```
I am using the latest firmware, here is the output of the console of cfclient when both crazyflies 2.1 are connected: 

```
SYS: Crazyflie 2.1 is up and running!
SYS: Build NA:NA (NA) MODIFIED
SYS: I am 0x20303146354B501100190040 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
posextbrinup. initialized: 2
DECK_DRIVERS: Found 16 drivers
DECK_INFO: Found 0 deck memory.
DECK_CORE: 0 deck enumerated
IMU: BMI088 Gyro I2C connection [OK].
IMU: BMI088 Accel I2C connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using estimator 1
CONTROLLER: Using controller 1
EEPROM: I2C connection [OK].
SYS: Free heap: 14856 bytes
---
SYS: Crazyflie 2.1 is up and running!
SYS: Build NA:NA (NA) MODIFIED
SYS: I am 0x20373048374D5019003D0021 and I have 1024KB of flash!
CFGBLK: v1, verification [OK]
posextbrinup. initialized: 1
DECK_DRIVERS: Found 16 drivers
DECK_INFO: Found 0 deck memory.
DECK_CORE: 0 deck enumerated
IMU: BMI088 Gyro I2C connection [OK].
IMU: BMI088 Accel I2C connection [OK]
IMU: BMP388 I2C connection [OK]
ESTIMATOR: Using estimator 1
CONTROLLER: Using controller 1
EEPROM: I2C connection [OK].
SYS: Free heap: 14856 bytes
```

whoenig, 2019-11-12:
If allcfs.takeoff() doesn't work, perhaps your NRF firmware is not up-to-date (broadcast support is required). Did you also update your NRF firmware?

jpreiss, 2020-06-13:
@hukohli were you ever able to resolve this?

jpreiss, 2021-01-30:
closing due to inactivity.

closed 2021-01-30 by jpreiss