USC-ACTLab/crazyswarm #176: Help tuning mellinger controller for bigger battery

williamdwarke, 2019-11-19:
Good afternoon - 

This isn't so much an issue but a request for help (or documentation) related to tuning the mellinger controller for a substantially bigger battery. The battery weighs 12.6 grams (as opposed to the original 7.1), and I know the crazyflie can carry it just fine; the controller just goes completely unstable when I try to do anything but sit still or make very slow movements. I've tried increasing the "mass" parameter in crazyflieTypes to 0.0375 to reflect the mass increase but it's only making things worse and the original paper by mellinger isn't very insightful in this respect. Honestly if you could just let me know which parameters to adjust and in which direction that would be a huge help since I can do all the fine tuning myself.

Regards,
William

williamdwarke, 2019-11-19:
I just wanted to add that it is just a 500mAh battery, and I have heard of people running 750s on the crazyflies before for extreme flight times, so I'm pretty confident it should work with a bit of tuning.

whoenig, 2019-11-20:
I agree that it should work, but it also depends on the battery. For example, we found that the default battery can produce about 45g max thrust, while the ones from Hobbyking achieve up to 55g. So depending you the decks, you might be very close to motor saturation, which has exactly the effect you are seeing. Measuring this exactly requires a load cell, but you can also collect some (pwm, vbat) data to get a prediction (battery voltage drops significantly during high motor speeds for worse batteries).

For tuning:

- Make sure the battery is balanced, i.e., the cog is still in the middle of the quad
- Reduce the position gains (kp_xy, kd_xy, kp_z, kd_z), e.g., half them to focus on attitude gains better
- Without data it is hard to tell, but you can try reducing kR_xy and increasing kw_xy.

If you have the uSD deck, it is best to log the data on the card, and then analyze it after each tuning step to figure out in which direction to go.

williamdwarke, 2019-11-20:
Thanks for the advice! Which batteries from Hobbyking would you recommend? I've been on the hunt for some better performing batteries.

whoenig, 2019-11-20:
We use https://hobbyking.com/en_us/turnigy-nano-tech-300mah-1s-20-40c-lipo-pack-losi-mini-compatible.html. Not as high capacity as what you are looking for, but holds up very well so far (and has the identical mechanical properties as the original batteries).

williamdwarke, 2019-11-20:
Perfect, thanks a ton! You can go ahead and mark this thread as closed; I'll give tuning a shot but I think you're right and I'm just saturating the motors and need to go for a lighter battery.

closed 2019-11-20 by whoenig