USC-ACTLab/crazyswarm #177: Single marker configuration stops working when its ID is changed to '4'

zeroos, 2019-11-21:
Hey!

We've run into a rather strange issue yesterday. We wanted to switch to a single marker configuration on our crazyflies. We added a new marker configuration with just a single marker and the crazyswarm server was failing a couple of seconds after starting. 

We were able to reproduce the behavior using the default crazyswarm setup with the following markerConfiguraitons in the crazyflieTypes.yaml:

`
markerConfigurations:
  "0":  # for standard Crazyflie
    numPoints: 4
    offset: [0.0, -0.01, -0.04]
    points:
      "0": [0.0177184,0.0139654,0.0557585]
      "1": [-0.0262914,0.0509139,0.0402475]
      "2": [-0.0328889,-0.02757,0.0390601]
      "3": [0.0431307,-0.0331216,0.0388839]
  "1": # medium frame
    numPoints: 4
    offset: [0.0, 0.0, -0.03]
    points:
      "0": [-0.00896228,-0.000716753,0.0716129]
      "1": [-0.0156318,0.0997402,0.0508162]
      "2": [0.0461693,-0.0881012,0.0380672]
      "3": [-0.0789959,-0.0269793,0.0461144]
  "2": # big frame
    numPoints: 4
    offset: [0.0, 0.0, -0.06]
    points:
      "0": [0.0558163,-0.00196302,0.0945539]
      "1": [-0.0113941,0.00945842,0.0984811]
      "2": [-0.0306277,0.0514879,0.0520456]
      "3": [0.0535816,-0.0400775,0.0432799]
  "3":  # for standard Crazyflie with a single marker
    numPoints: 1
    offset: [0.0, -0.01, -0.04]
    points:
      "0": [0.0177184,0.0139654,0.0557585]
  "4":  # for standard Crazyflie with a single marker
    numPoints: 1
    offset: [0.0, -0.01, -0.04]
    points:
      "0": [0.0177184,0.0139654,0.0557585]
`
(setup number 3 is copied under number 4).

We were trying to fly one drone. If the markerSetup on this drone was set to 3 everything was fine. As soon as we changed it to 4 it started failing.

Are you able to reproduce this problem? Is there any assistance we can give in order to help? We were not able to debug it, because we program our experiments mainly in Python and we don't really know how the server works.

whoenig, 2019-11-21:
Did you increase https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/crazyflieTypes.yaml#L102 to 5?

I plan to change this from arrays to name-based lookup in the future, which will also improve the error handling/messages.

zeroos, 2019-11-22:
Thank you, that was the problem! I guess we are too much used to Python and not having to define list lengths. ;) Sorry for trouble!

I'd use the occasion to put some self promotion though: we are opening a new drone hall at the University of Klagenfurt soon (10x7x11m, yes, height is 11m) and we are using Crazyswarm to demonstrate our research during the opening. You can take a look at the video from tests: 
https://youtu.be/Db6aiSa4soU . Your project really helped us, so thanks a lot for your work!

Best wishes,
Michał

closed 2019-11-22 by zeroos

whoenig, 2019-11-22:
Thanks for sharing - that looks amazing! Feel free to reach out here or via email if you have any other questions (or feedback).
Best,
Wolfgang