USC-ACTLab/crazyswarm #178: Crazyflie Couldn't take off

YimingWan, 2019-11-24:
Hi Wolfgang,
I just faced a very tricky issue, I tried to run niceHover and figure8 python scripts with my crazyflie, but crazyflie couldn't take off. Vicon system worked, rviz worked(crazyswarm hover_swarm.launch), then after I hit enter to run the sripts, first second crazyflie's motors had reactions, all propellers spun like 1 second the stopped. 
In rviz it shown like the photo attached there. I have 7 crazyflies, there are 3 of them have this issue, I tried to update the firmware but after update they still have this issue, so the only way I have right now is asking you for help see if you faced this kind of problem before. Looking forward to your reply.

Best,
Yiming
[image]

whoenig, 2019-11-25:
Hi Yiming,
Does the problem reproduce consistently with any of the 3 affected CFs, even if you try flying them alone?

I have seen similar issues before, when:
1) There is an issue with the IMU/IMU calibration. When you connect, make sure that the console prints "Ready to fly.", for each Crazyflie. Otherwise the firmware might still try to figure out the bias of the IMU. -OR-
2.) For some reason one of the safety features gets enabled at runtime. If you run the latest official firmware, try disabling the SITAW framework (comment out https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/interface/sitaw.h#L39 and recompile).

YimingWan, 2019-11-25:
Hi Wolfgang,
Yes, this problem reproduces consistently, even I try to fly them individually the issue still there.
I will try to solve this by your guides and give you feedback!
Thanks a lot and wish you have a good night!

Best,
Yiming

YimingWan, 2019-11-25:
Hi Wolfgang,
I figured out the issue, like you said it is the latest official firmware's issue, I followed your guide then disabled the SITAW framework then crazyflie can hover.
Again, thank you so much for your help much appreciated!

Best,
Yiming

closed 2019-11-25 by YimingWan

whoenig, 2019-11-25:
I don't recommend keeping this disabled permanently (this is an important safety feature especially for larger swarms; it also protects motor damages that might happen if you crash and don't turn of the motors). Perhaps you can figure out which of the sitaw conditions are triggered and why.

YimingWan, 2019-11-25:
I got it, I will take a look at it! Thanks for your suggestion!