USC-ACTLab/crazyswarm #182: Errors while uploading trajectories

williamdwarke, 2019-12-03:
This is an issue that's been plaguing me for a while now and I have tried many workarounds but nothing seems to completely fix it. When I try to upload trajectories to the crazyflies, I get the following error:
```
[ INFO] [1575360607.566731651]: [cf7] Upload trajectory
[ INFO] [1575360607.604054796]: [cf7] ESTIMATOR: Using Kalman (2) esSYS: Assert failed src/modules/src/crtp.c:186
```
Sometimes I also see the error:
`esOW: Cmd 0x21 timeout`
on the second line. I suspect this may have something to do with the latest firmware, because I never saw it before and now it even happens when I try to run **python figure8_csv.py**. I have tried moving the trajectory upload command to many different parts of the code (before takeoff, during hover, etc.) but it still happens periodically, causing crazyflies to crash or preventing them from even taking off.

whoenig, 2019-12-03:
I have not seen that issue recently, but it is related to C++ sending radio packets "too fast", so some of the queues in the firmware overflow and packets have to be dropped on the firmware side. The easiest "fix" is to comment out the assert https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/crtp.c#L186. This won't affect correctness, as crazyflie_cpp is resending packets that are not acknowledged/dropped on the firmware side.

I am not sure about the other error, but I have also seen that in my experiments in newer firmwares. It does not seem to affect the execution. Perhaps you can find out more about that in the official forum/issue tracker for the firmware.

jpreiss, 2021-01-30:
@williamdwarke were you able to resolve this?

jpreiss, 2021-08-04:
closing due to inactivity.

closed 2021-08-04 by jpreiss