USC-ACTLab/crazyswarm #183: synchronized logging problem

sadransh, 2019-12-20:
Hi, I want to log data from one crazyflie. I used `"rosbag record -a"` to get /tf and other data: I used these settings : 
`genericLogTopics: ["log1","log2"]
 genericLogTopicFrequencies: [10,10]
 genericLogTopic_log1_Variables: [ "gyro.x","gyro.y", "gyro.z"]
 genericLogTopic_log2_Variables: [ "acc.x","acc.y", "acc.z"]`

and the result of rosbag info is like this:
```     tf2_msgs/TFMessage              [94810edda583a504dfda3829e70d7eec]
topics:
     /cf1/log1       230 msgs    : crazyflie_driver/GenericLogData
             /cf1/log2      2156 msgs    : crazyflie_driver/GenericLogData
             /diagnostics     23 msgs    : diagnostic_msgs/DiagnosticArray
             /joy            164 msgs    : sensor_msgs/Joy                
             /rosout          15 msgs    : rosgraph_msgs/Log               (5 connections)
             /rosout_agg       2 msgs    : rosgraph_msgs/Log              
             /tf            1807 msgs    : tf2_msgs/TFMessage'
```
 the collected data are not synchronized. I need to log with somehow high data rate.
I need to know the position of the crazyflie at the moment of collecting data. Is there any way to do so? will using usdcard help to have all the mentioned data? 
I want to log some data related to the motor too. like PWM or thrust for each motor and ..? Do you have any suggestion to log such data?

whoenig, 2019-12-20:
The values within one log topic are synchronized, but values across different topics are not. You can create your own custom log configuration (as you did for log1 and log2) above and combine the gyro and acc readings. Unfortunately, you can only log ~26 Bytes per log configuration, so about 6 float variables. Finally, logging over the radio is limited (in the firmware) to 100 Hz.

The best choice is likely the uSD card. I have successfully used it to log >20 variables at high rates (you can go up to 1 khz if you only log a few variables). All variables on the uSD card are synchronized (default timestamp in ms, but its also possible to get a us timer). You can download the latest log file from the uSD card over the radio using the crazyflie_tools package.

sadransh, 2019-12-20:
Great answer! Thank you very much.
How about /tf data and motor are they accessible from firmware to log them into the uSD card?

whoenig, 2019-12-20:
/tf: It depends on how you localize. I assume that you use a mocap or lighthouse? In this case, the logging variables are defined here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/modules/src/stabilizer.c#L636-L677 (the *Z version is compressed to minimize file-size/overhead).

Motor: The raw PWM values have logging variables defined here: https://github.com/bitcraze/crazyflie-firmware/blob/master/src/drivers/src/motors.c#L323-L328.

The uSD card supports all logging variables.

jpreiss, 2020-06-13:
@sadransh any updates to share with future readers of this issue?

easy98, 2021-01-27:
Hi, 

I also have the same problem. But since now I have already attached the flow-deck and the lps-deck. So it seems that I cannot add uSD-card deck at the same time. So is there other solutions to solve this problem and get synchronized logging variables? Thanks.

whoenig, 2021-01-27:
Bitcraze now sells longer pin headers for decks, which should allow you to attach 3 decks. Perhaps that suffices? Otherwise you can use the logging over the Crazyradio (just a few variables can be logged simultaneously), or physically connect a J-Link on the debug port (to collect data without flying).

closed 2021-02-20 by jpreiss