USC-ACTLab/crazyswarm #186: Naming of cmdVel

jpreiss, 2020-01-14:
If I understand correctly, the [actual behavior of `cmdVel`](https://github.com/USC-ACTLab/crazyswarm/commit/ca1094652d7b2d24e2d3fc635b8be595d7cc58c3#diff-59d91c51e0369252f72df313bab9d0eaR440) is to send a control input of (absolute roll angle, absolute pitch angle, yaw angular velocity, thrust), corresponding to the typical "easy mode" (non-acrobatic) manual control inputs from an RC joystick transmitter.

In my opinion, the name `cmdVel` clashes with the behavior. I would expect from the name `cmdVel` that the setpoint is a 3D velocity vector and an absolute yaw or yaw velocity. That is the exact input we need for the desired new single-integrator abstraction. The firmware's `commander_generic` [already supports this kind of input](https://github.com/bitcraze/crazyflie-firmware/blob/2e5d259f68b32ef932d1e6077b853c5401e8da43/src/modules/src/crtp_commander_generic.c#L98), although it appears that we don't handle it in the Mellinger controller as of now, and it doesn't look like `crazyflie_cpp` is capable of sending the packet.

I guess we could either:

1. Call the single-integrator input something like `cmdVelWorld` and keep `cmdVel`
2. Rename `cmdVel` to e.g. `cmdAttitudeThrust` and use `cmdVel` for the single-integrator input
3. Rename `cmdVel` for correctness, but call the single-integrator input something else, so current scripts that use `cmdVel` break loudly with a "method does not exist" exception instead of silently changing behavior.

What do you think?

whoenig, 2020-01-20:
This name was chosen mostly for historical reasons: When I started i looked at the hector_quadcopter package and that's the name they used. I agree that this name is most unfortunate. I don't like changing the behavior silently, so that leaves 1. or 3. Another option (4.) would be to keep cmdVel, but print some kind of "deprecated" warning (and find two new names, e.g., cmdAttitudeThrust, and cmdVelWorld).

jpreiss, 2020-01-20:
4. is a good idea. We can use the built-in `warnings` module and `DeprecationWarning` and then remove it in a year or so (idk if our concept of "versions" is meaningful enough to schedule a version where it will be deleted...)

jpreiss, 2020-01-20:
Would you also want to do the same thing for `crazyflie_ros`? I guess there, the public API is the topic name `cmd_vel`.

hanzheteng, 2020-01-21:
I haven't looked into the history of this name and the meaning behind it. However, one thing came to my mind, from a user's perspective, is the compatibility with the Bitcraze official python library [crazyflie-lib-python](https://github.com/bitcraze/crazyflie-lib-python). In other words, it would be great if we can expect the same behavior when sending the same velocity commands from crazyflie-lib-python and from crazyswarm.

jpreiss, 2020-01-23:
@hanzheteng good point. In crazyflie-lib-python, it is `velocity_world`.

For future reference the commander API is here: https://github.com/bitcraze/crazyflie-lib-python/blob/master/cflib/crazyflie/commander.py

jpreiss, 2020-02-25:
Using `cmdVelocityWorld`. Implemented in #197.

closed 2020-02-25 by jpreiss