USC-ACTLab/crazyswarm #194: Crazyflie takes off but drifts away

StevenLandgraf, 2020-01-31:
We use:
- Motive 2.0.2 Final (Solver replaces occlusion OFF) -> https://github.com/USC-ACTLab/crazyswarm/issues/166
- most recent crazyswarm packages
- newest firmware on crazyflies

Problem: 
The crazyflie takes off with hover_swarm.launch, after two seconds it drifts away and crashes.

hover_swarm.launch:

```<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 15 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          pNAcc_xy: 1.0
          pNAcc_z: 2.0
      # tracking
      motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      optitrack_host_name: "172.22.147.75"
      optitrack_local_ip: "172.22.147.74" # only needed if optitrack is selected
      # optitrack_server_ip: "172.22.147.75" # only needed if optitrack is selected
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_x" args="/cf2/log1/values[0]"/> -->
  <!-- <node pkg="rqt_plot" type="rqt_plot" name="rqt_plot_roll" args="/cf1/log1/values[2] /cf1/log1/values[3]"/> -->

</launch>```

whoenig, 2020-02-04:
That sounds like a tracking, or communication problem. Make sure you update the NRF51 firmware as well. If you have the LED ring make sure that it turns green. Otherwise, please add logging for "stateEstimate.x" (and .y, .z) and change "enable_logging" in the launch file to True. Look at the logging variables (e.g., using rostopic echo) and make sure that the state estimate updates as you manually move the Crazyflie around in your motion capture space.

Hope that helps and sorry for the delay!

StevenLandgraf, 2020-02-05:
Thanks for your answer. 
The solution was in fact a problem with the pose of our Crazyflies:
The Crazyflies need to point towards the x-axis of our MoCap-System when creating a RigidBody and we were confused about which direction that is. We figured it out by manually moving our Crazyflie around, like you also suggested.

closed 2020-02-05 by StevenLandgraf