USC-ACTLab/crazyswarm #195: Build failed - git/ cmake errors

bttner, 2020-02-12:
When using ROS Melodic, Ubuntu 18.04.4 LTS and running the commands
```
$ sudo apt install git swig libpython-dev python-numpy python-yaml python-matplotlib gcc-arm-none-eabi libpcl-dev libusb-1.0-0-dev sdcc ros-melodic-vrpn-client-ros
$ git clone https://github.com/USC-ACTLab/crazyswarm.git
$ cd crazyswarm
$ ./build.sh
```
I get the error message 

> Fetched in submodule path 'externalDependencies/qualisys_cpp_sdk', but it did not contain 9562108240a74c930c41cc05e993873cb081a76e. Direct fetching of that commit failed. 

Full terminal output: [fetching_fail.txt](https://github.com/USC-ACTLab/crazyswarm/files/4190794/fetching_fail.txt) 

I tried to solve this problem by manually cloning the repository, e.g., 
```
$ cd ros_ws/src/externalDependencies/libmotioncapture/
$ rm -rf externalDependencies/qualisys_cpp_sdk
$ git clone https://github.com/qualisys/qualisys_cpp_sdk.git externalDependencies/qualisys_cpp_sdk
$ cd ~/crazyswarm/
```
but then cmake failed eventually. Full terminal output: [cmake_fail.txt](https://github.com/USC-ACTLab/crazyswarm/files/4190822/cmake_fail.txt), [CMakeOutput.log](https://github.com/USC-ACTLab/crazyswarm/files/4191017/CMakeOutput.log), [CMakeError.log](https://github.com/USC-ACTLab/crazyswarm/files/4191019/CMakeError.log)

Do I get these errors simply because I am not following your recommendations of using ROS Kinetic and Ubuntu 16.04? Otherwise, do you may know a way of solving this problem?

barbosafs, 2020-02-12:
Hello there. I also got the same error as above. I do also run Melodic on 18.04.
What I understood from a quick search on the internet is that such a submodule failure might be related to not having permission to clone such repo. See [here](https://stackoverflow.com/questions/42417294/why-does-git-fail-to-fetch-specific-valid-submodule-for-a-given-commit-and-how-t).

Following a suggestion from that link, I modified line 17 of Crazyswarm's build.sh to 
`git submodule update --force --recursive --init --remote`
I've uploaded a [build.txt](https://github.com/USC-ACTLab/crazyswarm/files/4191474/build.txt) here too, for clarity.

It seems to build normally now. 
I don't have Qualisys to check if its working as expected.

Perhaps it would be nice if there was a way of selecting which motion capture system one wants to install since the beginning, instead of having to install all of them every time.

Cheers,
Fernando

bttner, 2020-02-12:
Thanks. That fixed the problem.

duembgen, 2020-02-20:
I had the same problem, same system. Thank you for the fix.

jpreiss, 2020-05-29:
PRs for single-mocap builds would be welcome, but I'm closing this issue due to low priority.

closed 2020-05-29 by jpreiss