USC-ACTLab/crazyswarm #204: Simulation mode with python3

Rucy-kim, 2020-04-21:
Hi, guys.

I am using python3 in my project now, so I want to run simulation in python3. 

After installing and building the package, I ran 'python figure8_csv.py --sim'. 
However, I get the following error message:

```
Traceback (most recent call last):
  File "figure8_csv.py", line 5, in <module>
    from pycrazyswarm import *
  File "/home/drone/work/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/piecewise.py", line 1, in <module>
    from .cfsim import cffirmware as firm
  File "/home/drone/work/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 28, in <module>
    _cffirmware = swig_import_helper()
  File "/home/drone/work/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/cffirmware.py", line 24, in swig_import_helper
    _mod = imp.load_module('_cffirmware', fp, pathname, description)
  File "/home/drone/.pyenv/versions/3.5.2/lib/python3.5/imp.py", line 242, in load_module
    return load_dynamic(name, filename, file)
  File "/home/drone/.pyenv/versions/3.5.2/lib/python3.5/imp.py", line 342, in load_dynamic
    return _load(spec)
ImportError: /home/drone/work/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/_cffirmware.so: undefined symbol: PyInstance_Type
```

What can I do to fix this?

Thanks for the help.

whoenig, 2020-04-23:
We don't support Python3, since ROS Melodic still depends on Python2 (at least in the default build). That being said, the issue probably stems from the fact that the build script builds the simulation module for Python2. You can try adjusting that in https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/pycrazyswarm/cfsim/Makefile#L5-L9.

jpreiss, 2020-04-23:
+1 you should be able to fix it by running `setup.py` under `python3` and passing the correct command line args to `swig` to generate python 3 code. Keep in mind this will only work for simulation.

Rucy-kim, 2020-04-23:
I think it would be better to use python2.
thank you for the reply.

I will close the issue.

closed 2020-04-23 by Rucy-kim