USC-ACTLab/crazyswarm #205: Use lighthouse positioning system

Rucy-kim, 2020-04-24:
Hi

I am trying to test crazyswarm using lighthouse positioning system.

But i don't know how to set the position of base station

I proceeded with the contents of this link and succeeded in acquiring position with the latest firmware of bitcraze.
[https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/functional-areas/lighthouse_overview.md](https://github.com/bitcraze/crazyflie-firmware/blob/master/docs/functional-areas/lighthouse_overview.md)

Do you have to do the same in crazyswarm to use the lighthouse positioning system?

Thank you in advance for any help you can provide!

whoenig, 2020-04-25:
The same approach should work. Simply use the modified firmware, and make sure that you disable the motion capture (set https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L36 to "none").

Rucy-kim, 2020-04-27:
That sounds great! Thank you for your reply.

closed 2020-04-27 by Rucy-kim