USC-ACTLab/crazyswarm #206: Several problems with different cmd types

KYLiu0620, 2020-04-27:
Hi,

I have used cmd_position for the flight successfully before, and now I want to try other method but bump into some problems with new implementation.

1. The output of my controller is  linear acceleration of each agent in world frame. Which low-level command is better to perform such kind of application?

2. If I want to use **cmdFullstate** for the above controller, how to obtain the command of angular velocity (omega) with real-time linear acceleration?  What material can I learn for this part? I cannot understand with the source code in uav_trajectory.py.

3. I have tried testCmdVelocityWorld.py with hover_swarm.launch in crazyswarm, but the crazyflie only took off and hovered instead of performing the circle as the simulation. I checked the source code of crazyswarm_server.cpp, and it doesn't subscribe to the topic 'cmd_velocity_world'. I tried to add that by myself, but it randomly fell down. Does crazyswarm_server.cpp support **cmd_velocity_world** now?

4. In the documentation, yaw rate of **cmd_vel** is positive for turning CCW. However, it turned CW with yaw rate = 0 and CCW with yaw rate = -90 when I tried.

I tried all the aforementioned problems with the newest commit of this package. Thanks for all the help and suggestions.

jpreiss, 2020-05-07:
1. There is no command that takes an acceleration vector directly. The roll and pitch angles and thrust magnitude in `cmd_vel` are essentially a thrust vector in polar coordinates (thrust = acceleration + gravity compensation). Unfortunately, the thrust of `cmd_vel` is not in a physically meaningful unit, so it may require some tuning. Another option is to integrate your controller's accelerations to get desired position and velocity, and then send `cmdFullState`. If doing the latter, I would suggest reducing the PID position gains.

2. If you are tracking a trajectory, angular velocity is related to the third derivative of position. The math is explained in the paper "Minimum Snap Trajectory Generation and Control for Quadrotors" by Mellinger and Kumar. If you don't have an analytical third derivative, or don't want to do the math, just set `omega` to zero. The trajectory tracking accuracy will degrade a little, but it will still work.

3. See my comment on #209.

4. Do you really mean "it turned CW with yaw rate = 0" or was it a typo for "yaw rate = 90"? Either way, I never used `cmd_vel` so I'm not sure.  @whoenig might know. This behavior could occur if you put on your "A" propellers backwards.

KYLiu0620, 2020-05-07:
> 4. Do you really mean "it turned CW with yaw rate = 0" or was it a typo for "yaw rate = 90"? Either way, I never used cmd_vel so I'm not sure. @whoenig might know. This behavior could occur if you put on your "A" propellers backwards.

That's not a typo. I know that sounds weird, but I have checked the command of yaw rate. I didn't find any discussion about the issue about cmd_vel, so I just want to ask your experience. Thanks a lot for your reply!

whoenig, 2020-05-25:
Sorry for the delay! I haven't used `cmd_vel` with the crazyswarm, so it is possible that there is a bug. If you use plain crazyflie_ros and try to follow the tutorial for manual, joystick-based control it will use `cmd_vel` as mode. Here, the quadrotor does not yaw unless the appropriate joystick is moved.

Your use-case sounds like you have a double-integrator model. The easiest for now is to use the cmdFullState, where you do the integration of position and velocity on the host side. As James pointed out, for practical purposes you can set omega to zero.

ucaszzw, 2020-06-03:
@KYLiu0620 @JIMMMY0 Hello, I have the same problem in "CMD"_ velocity_ World ".Would like to ask you if you have solved it?

KYLiu0620, 2020-06-03:
Hi guys,
@whoenig @jpreiss Thanks a lot for your suggestions.
@ucaszzw I haven't tried. I'm working on cmdFullState now.

ucaszzw, 2020-06-03:
@KYLiu0620 @whoenig @jpreiss 
Hello, thank you for your reply. I don't know much about cmdfullstate.

I have two questions:
1. Can you use the cmdfullstate command to control the UAV's speed?
2. If I want to know about the usage of cmdfullstate, which files should I refer to? Thank you very much!

Question details:
What I want is to control the speed of UAV. For cmdfullstate, it has many parameters. I made a simple attempt, in which ACC, yaw, Omega and other parameters can be set to 0. However, POS cannot be set to 0. When I set POS and vel at the same time, where POS = CF1. Position() + DT * vel; vel = vel. I have only carried out simulation test, its effect is similar to position control, and I can't control the speed. I want to ask, where did I make a mistake?

jpreiss, 2020-06-03:
To approximate velocity control with `cmdFullState`, you should integrate your velocity commands over time to get a position command. Commanding `.pos = cf1.position() + dt * vel` could create a positive feedback loop, because you are essentially accepting the disturbance accumulated so far instead of trying to reject it. Something like

	rate = 30.0 # Hz
	posCmd = cf.initialPosition
	while True:
		velCmd = # whatever is determining the goal velocity
		posCmd += velCmd / rate
		cf.cmdFullState(posCmd, velCmd, ...)
		timeHelper.sleepForRate(rate)

ucaszzw, 2020-06-04:
Thank you for your reply, I will use the position information to make feedback in velocity control. I will test it as soon as possible.

ucaszzw, 2020-06-19:
> 1. The output of my controller is  linear acceleration of each agent in world frame. Which low-level command is better to perform such kind of application?
For your first question, I am facing the same situation.

I'd like to ask you how did you solve it? I'm not familiar with crazyswarm enough. I would appreciate it if you could attach your solution code.

ucaszzw, 2020-06-19:
@jpreiss @whoenig 
I have never been able to understand the meaning of this command "`cmdFullState(pos, vel, acc, yaw, omega)`"; For` cmdVelocityWorld(vel, yawRate)`, it is easy to understand that The onboard controller will try to achieve this velocit; For `cmdPosition(pos , Yaw = 0)`, it is easy to understand that the onboard controller will reach the desired position.

1. What function can the `cmdFullState (pos, vel, acc, yaw, omega)` command achieve?
If you look at the command itself, the onboard controller will reach a specific position, maintain a specific speed and acceleration... This is obviously a contradiction.
2. If I want crazyflie to reach the desired speed or the expected linear acceleration in the world coordinate system. How should I use `cmdFullState (pos, vel, acc, yaw, omega)`?

KYLiu0620, 2020-06-20:
@ucaszzw 
> 2. If I want crazyflie to reach the desired speed or the expected linear acceleration in the world coordinate system. How should I use `cmdFullState (pos, vel, acc, yaw, omega)`?

As far as I know, you can send desired velocity as vel and integrate it as pos command. The other arguments can be set to zero.

closed 2021-02-20 by jpreiss