USC-ACTLab/crazyswarm #207: Mellinger controller on Python

eduardo-del, 2020-05-04:
Hi guys 

I have a problem when i try to run some example from cflib, when i try to run the position_commander_demo.py code my crazyflie just conneting with the crazyradio but the crazyflie dont run the task, just connected and after a little time it disconected, like if the crazyflie was finished his task.

On the firmware im using the mellinger controller.
In the moment just want to use the mellinger controller on Python, maybe afteer i will to use the mellinger controller on ROS.

Can someone help me?

thank you!!

jpreiss, 2020-05-04:
Are you using any Crazyswarm functionality? `position_commander_demo.py` is a file in https://github.com/bitcraze/crazyflie-lib-python. This is part of the official Bitcraze python library, not the Crazyswarm project.

eduardo-del, 2020-05-04:
No, i just using the cflib on Python
Do you think that is necessary to use Crazyswarm for this task?

whoenig, 2020-05-04:
Crazyswarm is not necessary. For cflib, please use the [official support forum](https://forum.bitcraze.io/) - this issue tracker is only for issues related to current Crazyswarm users.

eduardo-del, 2020-05-05:
Hi @whoenig  and @jpreiss  thanks for answers.
Today i used Crazyswarm with Optirack and  i followed this tutorial   [https://crazyswarm.readthedocs.io/en/latest/usage.html](url), when i try to proof the code  figure8_csv.py  with `$roslaunch crazyswarm hover_swarm.launch` i have some problem, it's the next 
`eduardo@eduardo-Aspire-V3-574:~/crazyswarm$ roslaunch crazyswarm hover_swarm.launch
... logging to /home/eduardo/.ros/log/4c64d070-8e2f-11ea-9ea0-a8a7952f836d/roslaunch-eduardo-Aspire-V3-574-5810.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

RLException: error loading <rosparam> tag: 
	Error within YAML block:
	while parsing a block mapping
  in "<unicode string>", line 2, column 7:
          world_frame: "/world"
          ^
expected <block end>, but found '<block mapping start>'
  in "<unicode string>", line 32, column 8:
           optitrack_host_name: "192.168.2.10"
           ^

YAML is:

      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          pNAcc_xy: 1.0
          pNAcc_z: 2.0
      # tracking
      motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      #vicon_host_name: "vicon"
       optitrack_host_name: "192.168.2.10"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
       vrpn_host_name: "optitrack"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    
XML is <rosparam>
      world_frame: &quot;/world&quot;
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: [&quot;log1&quot;]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: [&quot;stateEstimate.x&quot;, &quot;ctrltarget.x&quot;]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
[... 59 lines cut ...]
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          pNAcc_xy: 1.0
          pNAcc_z: 2.0
      # tracking
      motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      #vicon_host_name: "vicon"
       optitrack_host_name: "192.168.2.10"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
       vrpn_host_name: "optitrack"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False` 

Sorry for insisting but when i'm at  laboratory i use optitrack with crazyswarm and when i'm at house i just can use cflib and i  working with both.

I hope you can help me with my crazyswarm problem, thank you

whoenig, 2020-05-05:
(Re-opening). It looks to me like a whitespace problem. When you edit hover_swarm.launch (the yaml portion), make sure you enable that you can see white spaces in your editor and have consistent spaces - this is a similar issue that you would have when you work with Python source code files and mix tabs and spaces.

eduardo-del, 2020-05-05:
Okay @whoenig  thank you, i will check the file and then tell you if my problem was resolved.

eduardo-del, 2020-05-09:
Hi @whoenig  i checked my file hover_swarm.launch and i  has the same problem
i check that all the code is like in the example from [https://crazyswarm.readthedocs.io/en/latest/usage.html](url) and my file is this: 

`<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          pNAcc_xy: 1.0
          pNAcc_z: 2.0
      # tracking
      motion_capture_type: "vrpn" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      #vicon_host_name: "vicon"
       optitrack_host_name: "192.168.2.10"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
       vrpn_host_name: "optitrack"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  
  

</launch>`

Dou you know what is my error?

Thank you!!

eduardo-del, 2020-05-09:
[image]
[image]

whoenig, 2020-05-25:
I can't see the whitespace from your pictures. Could you either upload the file, or enable visible whitespace in your editor (looks like you use gedit; a discussion on how to enable that feature is here: https://stackoverflow.com/questions/17162851/option-to-display-control-characters-in-gedit/17162916)?

eduardo-del, 2020-05-25:
HI @whoenig thnak you for the answer, sure my file is here 

[https://github.com/eduardo-del/eduardo-del-Crazyswarm](url)

eduardo-del, 2020-05-25:
can you see the file?, if i select the link i can't open the file but in my folder i can see it.

closed 2021-02-20 by jpreiss