USC-ACTLab/crazyswarm #209: Use **cmdVelocityWorld(self, vel, yawRate)**  in experiment

JIMMMY0, 2020-05-07:
Hi, we are successful to run the simulation "**cmdVelocityCircle.py**". However, when we try to control a crazyflie using this script, the crazyflie has a weird behavior that is does not follow the velocity commands.

Similar problem is raised in the 3rd point in this [issue](https://github.com/USC-ACTLab/crazyswarm/issues/206).  We suspect that the class **CrazyflieServer()** in pycrazyswarm does not support velocity control yet.

I tried to modify the source codes but it's kinda difficult for me to implement successfully. Therefore, we sincerely ask for some suggestions on:
- how to use **cmdVelocityWorld** in the ROS in VICON environment ?
- How to modify the source codes to make "cmdVelocityCircle.py" implementable in experiment ?

Really appreciate if you could help me. @jpreiss @whoenig

jpreiss, 2020-05-07:
It looks like we fixed some other things related to `cmdVelocityWorld` in `crazyflie_ros` and forgot to update the submodule reference in crazyswarm. Try pulling the current master branch in `ros_ws/src/crazyflie_ros`.

JIMMMY0, 2020-05-07:
Thanks for your suggestion. I have `git pull` `crazyflie_ros` and `git submodule update` `crazyflie_cpp` and `crazyflie_driver`. And the ROS packages and crazyflie_cpp are all built successfully. However, if we use the crazyswarm, the crazyflie still cannot follow the velocity commands at all.

One more thing, if we use crazyflie_ros stack, by using `rostopic list` we can see the subscribed topic `cmdVelocityWorld`. However, if we use the crazyswarm, the rostopic list is as follows:
[image]

Really appreciate if this could be solved.

jpreiss, 2020-05-09:
what is the output of `git branch` inside the `crazyflie_ros` directory?

JIMMMY0, 2020-05-11:
Hi, it is the *master branch. Do I need to check out other versions ?

jpreiss, 2020-05-26:
I'm sorry you are having trouble with `cmdVelocityWorld`. Unfortunately, my university campus is closed due to COVID-19, so I can't enter the motion capture facility for hardware testing.

For now, is there any way you can work with `cmdFullState` or `cmdPosition` instead?

Also, to clarify: 

- When you run `cmdVelocityWorld.py`, the takeoff works correctly, right? It only flips over after you start sending velocity commands?
- Can you run other scripts like `cmdFullState.py` or `figure8_csv.py` successfully?

ucaszzw, 2020-06-02:
> Hi, we are successful to run the simulation "**cmdVelocityCircle.py**". However, when we try to control a crazyflie using this script, the crazyflie has a weird behavior that is does not follow the [...]

Hello Li, I have the same problem with you when I use crazyswarm for testing. I can use the cmdvelocityworld command in the simulation phase, but this command is invalid when flying crazyflie. Do you have a solution now?

JIMMMY0, 2020-06-04:
@jpreiss Thanks for your reply. We can use `cmdVelocityWorld` to control the quadrotor now by creating a publisher publishing the velocity command and reset the controller as `cf.setParam("stabilizer/controller", 1)` using PID controller instead of Mellinger controller. This solution works in `crazyflie_ros` stack. 

To address the clarifications: 
> When you run cmdVelocityWorld.py, the takeoff works correctly, right? It only flips over after you start sending velocity commands?

I think you mean the python script `cmdVelocityCircle.py` ? This works only in simulation. If in the experiment, it can take-off properly but cannot execute the `goCircle` function as expected, and it flips over. Yes, it only flips over when we send velocity commands.

> Can you run other scripts like cmdFullState.py or figure8_csv.py successfully?

I haven't tried cmdFullState.py. However, for the other high-level commands such as takeoff, goGo, trajectory tracking such as figure8 all works fine. 

@ucaszzw Yes, in the simulation everything is fine but not in the real experiment. We suspect that the crazyflie server in Crazyswarm does not subscribe the velocity commands. Try the solution as I   mentioned above. It may work.

jpreiss, 2020-06-04:
good catch, now that you mention I'm not sure if we ever tested cmdVelocityWorld with the Mellinger controller. I used the PID controller because I knew it had already been tuned by Bitcraze to handle velocity control modes for the flow deck, etc.

jpreiss, 2020-06-04:
16e3aca added note about PID controller to `cmdVelocityWorld` docstring.

ucaszzw, 2020-06-04:
Thank you very much for your help.I will try as soon as possible.

ucaszzw, 2020-06-10:
> [16e3aca](https://github.com/USC-ACTLab/crazyswarm/commit/16e3aca51b6c6a1e98218745cbec8c50da06fd0b) added note about PID controller to `cmdVelocityWorld` docstring.

Hello, jpreiss. I modified crazyswarm\ros_ ws\src\crazyswarm\launch\hover_ swarm.launch`, and changed the 'firmwareParams.stabilizer.controller ` to 1
I tested the script under OpticTrack and found that the modification had no effect. The mav can still take off and land, but it can't control the velocity.

swarm5, 2020-07-21:
@jpreiss @ucaszzw @whoenig Have you successfully tested `cmdVelocityWorld` on the PID controller platform? According to ucaszzw, switching to PID controller seems to have no effect?

swarm5, 2020-07-21:
@jpreiss @whoenig Hello, I am a researcher specializing in UAV distributed formation theory. Our research group has proposed a series of control algorithms for UAV formation, but it lacks a platform for experimentation. I saw you show the application of Distributed Control Law at IROS 2019. I want to know where I can quickly find these test scripts so that I can quickly reproduce your experiments (We are not familiar with the use of crazyswarm programs)and test our latest algorithms on this basis.

whoenig, 2020-07-21:
I have not personally tested `cmdVelocityWorld` with any controller.

For the control law at IROS 2019, are you referring to [these slides](https://drive.google.com/file/d/15favAyrLLpC_O6nrAp-eIbZijFUMLgwV/view)? In that case, this was a hybrid solution: firmware changes + high-level script, so it will require some getting-used time to the platform. Depending on your control law formulation, it might actually be sufficient to have high-level scripts only. Perhaps you can describe a bit more your use-case and we can find the most similar example(s).

jpreiss, 2020-07-21:
I have no information about `cmdVelocityWorld` beyond the older discussion in this thread. I used it successfully, but it's possible that I failed to commit something, etc.

jpreiss, 2020-07-21:
See also #206. Unfortunately, we implemented `cmdVelocityWorld` right before COVID-19 so I didn't get a chance to test as thoroughly as I wanted.

swarm5, 2020-07-21:
@whoenig @jpreiss thank you all. Then I will explore it myself. Yes, I have referenced those slides.
My control algorithm is:
1. Calculate the velocity of the machine based on other uavs, and use velocity control commands.
2. In a difficult place, I want to test an offensive and defensive confrontation scenario. I need crazyswarm to provide the drone with the real-time coordinate position of the attacker (the attacker can be a T-shaped calibration stick, or other aircraft, using the handle control)
3. I need to record the flight log, including time, speed, location

MauroPfister, 2020-09-15:
@JIMMMY0 @swarm5 @jpreiss I think the `cmdVelocityWorld` command does not work because there is a missing bit of code in `crazyswarm_server.cpp`. I copied over some bits from `crazyflie_server.cpp` and the commands seem to work fine (only with PID). Since I am not sure if I did everything correctly I just pasted the diff of my changes below. If desired, I can also make a PR.

```diff
diff --git a/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp b/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp
index 5ba998b..d5522ea 100644
--- a/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp
+++ b/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp
@@ -17,6 +17,7 @@
 #include "crazyflie_driver/SetGroupMask.h"
 #include "crazyflie_driver/FullState.h"
 #include "crazyflie_driver/Position.h"
+#include "crazyflie_driver/VelocityWorld.h"
 #include "std_srvs/Empty.h"
 #include <std_msgs/Empty.h>
 #include "geometry_msgs/Twist.h"
@@ -210,6 +211,7 @@ public:
     m_subscribeCmdVel = n.subscribe(tf_prefix + "/cmd_vel", 1, &CrazyflieROS::cmdVelChanged, this);
     m_subscribeCmdPosition = n.subscribe(tf_prefix + "/cmd_position", 1, &CrazyflieROS::cmdPositionSetpoint, this);
     m_subscribeCmdFullState = n.subscribe(tf_prefix + "/cmd_full_state", 1, &CrazyflieROS::cmdFullStateSetpoint, this);
+    m_subscribeCmdVelocityWorld = n.subscribe(tf_prefix + "/cmd_velocity_world", 1, &CrazyflieROS::cmdVelocityWorldSetpoint, this);
     m_subscribeCmdStop = n.subscribe(m_tf_prefix + "/cmd_stop", 1, &CrazyflieROS::cmdStop, this);
 
     if (m_enableLogging) {
@@ -500,6 +502,23 @@ public:
     // }
   }
 
+  void cmdVelocityWorldSetpoint(
+    const crazyflie_driver::VelocityWorld::ConstPtr& msg)
+  {
+    // ROS_INFO("got a velocity world setpoint");
+    // if (!m_isEmergency) {
+      float x = msg->vel.x;
+      float y = msg->vel.y;
+      float z = msg->vel.z;
+      float yawRate = msg->yawRate;
+
+      m_cf.sendVelocityWorldSetpoint(
+        x, y, z, yawRate);
+      // m_sentSetpoint = true;
+      // ROS_INFO("set a velocity world setpoint");
+    // }
+  }
+
   void cmdStop(
     const std_msgs::Empty::ConstPtr& msg)
   {
@@ -707,6 +726,7 @@ private:
   ros::Subscriber m_subscribeCmdVel;
   ros::Subscriber m_subscribeCmdPosition;
   ros::Subscriber m_subscribeCmdFullState;
+  ros::Subscriber m_subscribeCmdVelocityWorld;
   ros::Subscriber m_subscribeCmdStop;
 
   std::vector<crazyflie_driver::LogBlock> m_logBlocks;
```

jpreiss, 2020-09-28:
This should be fixed by https://github.com/USC-ACTLab/crazyswarm/pull/202. Please reopen if you still have problems.

closed 2020-09-28 by jpreiss