USC-ACTLab/crazyswarm #213: How to record the flight log of each mav？

ucaszzw, 2020-05-28:
1. Real physical mav:
I hope that when using crazyswarm, I can record the flight data of each mav, can anyone tell me that this function can be achieved now?
2. Simulation data:
I hope to change the simulation parameters to realize the output of simulated flight data of mavs. I try to use the --wirtecsv command, for example: python example_cmd_pos.py --writecsv. However, it seems that nothing happens, the terminal's command line has been stuck there. I do n’t know if it ’s the wrong way to use it or the problem in the project itself. Has anyone tried it?

At the same time, when I use the "python example_cmd_pos.py --vis vispy" command, the terminal is also stuck and nothing appears, I can only use Ctrl-C to force close.

whoenig, 2020-05-28:
1. What kind of data do you need and what localization system do you use? In principle there are multiple options: a) if using mocap and you only need position information, you can collect the mocap data, for example by using `rosbag`; b) if you need higher-fidelity data, we recommend the uSD-card deck.

2. For the hanging of the code: Did you forget `--sim`?

ucaszzw, 2020-05-29:
First of all, thank you for such a quick reply. I am still not familiar enough with crazyswarm, so I have a little question and I want to consult you.
1. For physical mav, I will test it with rosbag as soon as possible.
2. For simulation, I still have some problems.
writecsv:
Are you saying something similar to this form? The command "python niceHover.py --sim --writecsv" can indeed simulate normally, but where should I check the output?
vis:
I tried the --vis command, for example: "`python niceHover.py --sim --vis vispy`" but there is an error message "`ImportError: No module named vispy`". Can you tell me how to solve this problem?

whoenig, 2020-05-29:
`python niceHover.py --sim --writecsv` was the intended usage, but it looks like the actual writing of the output files was accidentally removed in https://github.com/USC-ACTLab/crazyswarm/commit/321c2d23d44ff25146805c8d4c5409b7a86f764e#diff-67e9b799784a73db7ec9610467931051. Thus, this currently does not work. However, the numpy array is still being updated, if that helps for a quick workaround, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/pycrazyswarm/output.py.

Using vispy is optional, you can find instructions on how to install it at http://vispy.org/installation.html.

ucaszzw, 2020-05-29:
Thank you for your reply. I also realized that vispy is not installed. I am installing it. I will continue to test relevant functions.

closed 2021-02-20 by jpreiss