USC-ACTLab/crazyswarm #214: Switch to Python 3

jpreiss, 2020-05-29:
Python 2 support ended on January 1, 2020. In addition to security issues, it is becoming unpleasant to maintain Python 2 software. For example, [my py2-specific pull request for VisPy was rejected](https://github.com/vispy/vispy/pull/1840), so I had to insert a workaround in the crazyswarm code.

Long-term, we should try to transition to ROS Noetic, as it [supports Python 3](https://wiki.ros.org/UsingPython3). However, we may need to wait until all of our dependencies support Python 3.

Short-term, it should be possible to make the simulation-only path support Python 3, since it does not depend on ROS and [it's possible to write code that is valid as both Python 2 and 3](https://python-future.org/compatible_idioms.html). This may also require changes to the SWIG firmware wrapper build.

jpreiss, 2020-07-14:
@Zhenghao-Dai is going to work on this. First step is to install ROS Noetic on an Ubuntu 20 (virtual) machine and make sure we can compile all the C++ code.

I experimented with py3 in simulation (https://github.com/USC-ACTLab/crazyswarm/commit/761ff4be0fc7f20e7a999c22054c0a4bb79191af) and I think the main changes needed in our Python code are 1) the [new relative imports](https://www.python.org/dev/peps/pep-0328/) and 2) the [changed behavior of the `/` division operator](https://www.python.org/dev/peps/pep-0238/). The conversion tool `2to3` says it can [convert the imports automatically](https://docs.python.org/2/library/2to3.html#2to3fixer-import), but due to dynamic typing [it has no way to detect integer divisions](https://bugs.python.org/issue12831), so we will need to fix those manually on a case-by-case basis.

I propose that we drop Python 2 support once we can get Python 3 working.

whoenig, 2020-07-14:
Crazyflie_ros compiles fine with ROS Noetic, so I don't think we'll have a lot of trouble on the C++ side.

I don't think we can drop Python 2 support, since many people still use older ROS versions (that require Python 2). Ideally, we change the code such that it runs with Python >= 2.7.

jpreiss, 2020-07-14:
I was thinking that simultaneous support will make CI, testing, documentation, etc. more complicated, but I guess it's a necessary sacrifice to give users a smooth transition.

jpreiss, 2020-07-25:
@whoenig  Did you have to change anything to compile for Noetic? Did you also compile on Ubuntu 20? We are having some issues with compile errors in PCL.

jpreiss, 2020-07-25:
nevermind, problem is that PCL needs C++14.

jpreiss, 2020-08-04:
@whoenig some design decisions where you might have an opinion:

1. Should we support all possible combinations of Ubuntu and Python versions, or just `(<= 18, py2)` and `(20, py3)`?

2. In the firmware bindings `Makefile`, we need to run `setup.py` with the appropriate Python version. Two possible ways to make this happen:
    - require that the user aliases `python` to the desired version. This is fine in Ubuntu 20 because Python 2 is now only accessible as `python2`. There is no binary associated to `python` in a default install. However, it would be weird if we want to support Python 3 on older Ubuntus because they alias `python` to `python2` by default.
    - have an environment variable, so the Makefile can look something like `$CSW_PYTHON setup.py ...`.
    - some other way of getting input to the Makefile that is less global.

whoenig, 2020-08-04:
1. I think we can limit the support to (18, py2) and (20, py3). We can drop the support/testing for 16 and below.
2. I prefer an environment variable, or the approach used in the [official Makefile](https://github.com/bitcraze/crazyflie-firmware/blob/master/Makefile#L19). The later would require users to either edit the Makefile or a file similar to `config.mk` used in the official firmware.

jpreiss, 2020-08-04:
Sounds good. I think the environment variable is simpler; I don't anticipate we will need enough configuration for a file any time soon.

With the environment variable one can run

    CSW_PYTHON=python2 ./build.sh

so it does not actually require a global change to the system.

jpreiss, 2020-09-09:
@Zhenghao-Dai I am doing the conda setup on my machine and it appears that our instructions in the docs for setting the python version are wrong. When creating a conda environment from the yaml file, it is not possible to control the python version separately from the environment file. e.g. if you remove the python version from `conda_env.yaml` (as in your python 3 branch) and try, you will get something with Conda's default python version:

    $ conda env remove -n crazyswarm
    $ conda env create python=2.7 -f conda_env.yaml
    $ conda activate crazyswarm
    $ python
    Python 3.8.5 (default, Sep 4  2020, 07:30:14)
    ....

The `setup-miniconda` github action seems to get around this by first creating an empty environment with the right Python version  and then doing `conda env update -f` to get all the extra packages specified by the yaml file. You can see this in the CI logs for the "Conda" step.

See also the github issue https://github.com/conda/conda/issues/9506 confirming that this is a limitation of `conda env create -f`.

The following two-step version worked for me:

    $ conda env remove -n crazyswarm
    $ conda create --name crazyswarm python=2.7
    $ conda env update -f conda_env.yaml
    $ conda activate crazyswarm
    $ python
    Python 2.7.18 |Anaconda, Inc.| (default, Apr 23 2020, 22:42:48)
    ...

I think we can suggest this two-step version in our docs.

jpreiss, 2021-01-30:
closing, will leave #287 open until we verify ubuntu 20 / py3 on hardware.

closed 2021-01-30 by jpreiss