USC-ACTLab/crazyswarm #215: delete Matlab scripts

jpreiss, 2020-05-29:
We have a lot of Matlab code in the repo from the early days of Crazyswarm. Little of it is still useful.

- Prototyping code for our EKF, and a lot of `.mex` wrappers that validated the firmware implementations against the Matlab implementations. Since we now use the Bitcraze EKF, this is not needed.
- Prototypes of the "avoid-target" and ellipsed-based demo trajectories. These are no longer supported by the firmware. I wouldn't mind restoring the ellipse mode in the official firmware, but with the current maturity of our SWIG firmware wrapper and our simulator, there is no need to use Matlab as a prototyping tool.
- The code that derives the figure-8 piecewise polynomial. This is probably the most relevant thing. It would be nice to port this code to Python before we delete it. Otherwise `figure8.csv` will become magic numbers with no reasoning to back it up.

whoenig, 2020-05-29:
For the trajectory planning, there is https://github.com/whoenig/uav_trajectories, a simple wrapper around code from ETHZ, which I've used in the past to generate trajectories like the figure-8.

jpreiss, 2020-05-29:
I'm trying to figure out what https://github.com/USC-ACTLab/crazyswarm/blob/c659a7481d7b514b0014934288975a77d7d0855f/matlab/trajectorygen/fig8_traj.m is actually doing. It's longer than I expected. I think it does something manual instead of using optimization to determine the time intervals between waypoints.

I also recall some issues with making sure the figure 8 is symmetrical in space, that the velocity at the ends of the 8 are perpendicular to the "axis" of the 8, etc.

jpreiss, 2020-05-30:
[image]

our current `figure8.csv` is not even symmetrical...

whoenig, 2020-05-30:
The current figure8.csv might have been generated by me using the repo I mentioned earlier. For the original Crazyswarm, we had the figure8 (probably generated using your matlab script) built as part of the firmware, but this feature was removed as part of the integration into the official firmware. I personally don't care that much if the figure8 is perfectly symmetrical.

jpreiss, 2020-05-31:
agree, it is just a testing tool mostly.

jpreiss, 2020-06-12:
I looked at that ETHZ project and it's definitely more capable than anything we have in matlab.

Out of curiosity, I tried writing some code from scratch using `cvxpy` that plans a figure 8 enforcing symmetry while still leaving as much as possible open for optimization. I don't think the ETHZ code is capable of expressing that kind of constraint. However, it runs much faster and uses the more numerically stable way to represent polynomials.

jpreiss, 2020-06-12:
deleted e74999

closed 2020-06-12 by jpreiss