USC-ACTLab/crazyswarm #219: Need help troubleshooting hover.launch

jcampbellNP, 2020-06-11:
Hello,

I am unable to get the hover.launch script to run and get my CF to fly. I am using an OptriTrack system. The data is streaming from Motive, and I am able to get the mocap_helper.launch to work (it shows my rigid body data from Motive).

When I launch the hover.launch script RVIZ opens but there is nothing inside of it. I am not getting any errors. It just looks like it is not finding the CF.
I have followed all of the directions for the crazyswarm installation, preparation and configuration files. All of the crazyflie addresses are correct with the correct firmware from the directions.

Here is the hover.launch script and the logs for mocap_helper.launch and hover.launch. Any ideas what I should do to get my CF flying?

hover.launch
<?xml version="1.0"?>
<launch>
 <arg name="joy_dev" default="/dev/input/js0" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />
 <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
   <rosparam>
     world_frame: "/world"
     # Logging configuration (Use enable_logging to actually enable logging)
     genericLogTopics: ["log1"]
     genericLogTopicFrequencies: [10]
     genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
     # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
     # allCrazyflies.yaml to set per drone)
     firmwareParams:
       commander:
         enHighLevel: 1
       stabilizer:
         estimator: 2 # 1: complementary, 2: kalman
         controller: 2 # 1: PID, 2: mellinger
       ring:
         effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
         solidBlue: 255 # if set to solid color
         solidGreen: 0 # if set to solid color
         solidRed: 0 # if set to solid color
         headlightEnable: 0
       locSrv:
         extPosStdDev: 1e-3
         extQuatStdDev: 0.5e-1
       kalman:
         pNAcc_xy: 1.0
         pNAcc_z: 2.0
     # tracking
     motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
     object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
     send_position_only: False # set to False to send position+orientation; set to True to send position only
     # vicon_host_name: "vicon"
     # optitrack_host_name: "optitrack"
     optitrack_local_ip: "***.***.***.***"
     optitrack_server_ip: "***.***.***.***"
     # qualisys_host_name: "10.0.5.219"
     # qualisys_base_port: 22222
     # vrpn_host_name: "vicon"
     save_point_clouds: ~/pointCloud.ot
     print_latency: False
     write_csvs: False
     force_no_cache: False
     enable_parameters: True
     enable_logging: True
   </rosparam>
 </node>
 
 
 
 
 
 <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
   <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
   <param name="timescale" value="0.8" />
 </node>
[... 210 lines cut ...]
 * /numDynamicsConfigurations: 1
 * /numMarkerConfigurations: 4
 * /rosdistro: melodic
 * /rosversion: 1.14.5

NODES
  /
	crazyswarm_server (crazyswarm/crazyswarm_server)
	crazyswarm_teleop (crazyswarm/crazyswarm_teleop)
	rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [30030]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 40d252aa-ac0c-11ea-8e0d-64006a91f7f5
process[rosout-1]: started with pid [30041]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [30048]
process[crazyswarm_teleop-3]: started with pid [30049]
process[rviz-4]: started with pid [30054]
[ INFO] [1591897922.605681352]: Wait for services...
[ INFO] [1591897922.607034106]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1591897922.689557690]: waitForService: Service [/emergency] is now available.
[ INFO] [1591897922.691725150]: Manager ready.

jpreiss, 2020-06-11:
Does your `crazyflies.yaml` show your crazyflie with id `1` and the correct radio channel?

Another common culprit when using the mocap's object tracking is not giving the objects names like `cf1, cf2, ...` but it looks like you did that.

You are using ROS melodic and presumably Ubuntu 18.04, whereas we only support ROS kinetic on Ubuntu 16.04. However, if everything compiled correctly, I think it should work.

whoenig, 2020-06-11:
I use it with 18.04/Melodic and this works fine for me. Please share your `crazyflies.yaml`.

jpreiss, 2020-06-11:
Should we update the docs to show support for 18.04/melodic?

jcampbellNP, 2020-06-11:
Thank you for your responses .I have the address for my CF as 0/80/2M/E7E7E7E701
Here is the crazyflies.yaml. 
crazyflies:
  - id: 1
    channel: 80
    initialPosition: [0.0,0.0, 0.0]
    type: default
  - id: 2
    channel: 80
    initialPosition: [0.0,0.0, 0.0]
    type: default

whoenig, 2020-06-11:
This kind of process stall means that there is trouble with connecting to OptiTrack. From the logs, it seems `mocap_helper.launch` actually uses a different configuration (hostname: "optitrack" vs "10.0.13.184"). Also, double check your launch file. If you are on the latest version, the configuration parameter should be "optitrack_host_name", see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/src/crazyswarm_server.cpp#L1443-L1449 and https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L40. The parameters you set are not used in the latest version. If you used the Crazyswarm for a long time, make sure that you update all submodules, since the optitrack support improved dramatically about a year ago.

whoenig, 2020-06-11:
@jpreiss I just pushed a change to include 18.04 in the list of supported OS.

jpreiss, 2020-06-11:
@jcampbellNP the `crazyflies.yaml` should only contain the CFs that you are actively using. The intended workflow is to list all of the CFs you own in `allCrazyflies.yaml`, then use `chooser.py` to graphically select the subset that you want to use. `chooser.py` overwrites `crazyflies.yaml`.

jcampbellNP, 2020-06-12:
Thank you both for all of your help. It came down to setting "optitrack_host_name" to the correct host IP address and making sure that the crazyflies.yaml was also correct.

jpreiss, 2020-06-13:
Glad to hear it's working for you now!

closed 2020-06-13 by jpreiss