USC-ACTLab/crazyswarm #225: I haven´t communication with crazyflie using crazyswarm

eduardo-del, 2020-06-16:
Hi guys i have a little problem, I haven't communication when i try tu run python figure8_csv.py

this are the steps that i did:
`
cd ros_ws/src/crazyswarm/scripts
`
`
python chooser.py
`

then appears a window crazyflie Chosser, when i click to battery button this messages appear:

>../../../../scripts/battery.py:10: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
../../../../scripts/battery.py:12: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
timeout

2. The simulation with python figure8_csv.py --sim run correctly 
3. When i try run figure8_csv without --sim the crazyflie haven't communication 
I first did 

 `
source ros_ws/devel/setup.bash

`
`
roslaunch crazyswarm hover_swarm.launch
`
and in other terminal

`
python figure8_csv.py
`
 but the crazyflie isn't connected, i use optitrack and firmware 2020.06
the optitrack show me the frames of the crazyflie on rviz.

some one can help me with the communication
thank you.

whoenig, 2020-06-17:
What is the URI of your Crazyflie and how does your `allCrazyflies.yaml` look like?

eduardo-del, 2020-06-17:
the URI is 0/80/250K/E7E7E7E701

and 

` allCrazyflies.yaml`

`crazyflies:
  - id: 1
    channel: 80
    initialPosition: [0.0, 0.0, 0.0]
    type: medium
  - id: 2
    channel: 100
    initialPosition: [-1.0, -0.5, 0.0]
    type: default`

whoenig, 2020-06-17:
The datarate needs to be set to `2M` instead of `250K` for the Crazyswarm. You can change this setting using the official cfclient.

eduardo-del, 2020-06-17:
Okay, I will do it, thank you so much @whoenig , and also for the help on crazyflie_ros.

eduardo-del, 2020-06-17:
Hi @whoenig i had the same problem when i changed the URI on official cfclient, 
should I change other file or thing?

Thank you so much.

easy98, 2020-10-10:
Hello, I also meet the same problem, how do you solve it?

eduardo-del, 2020-10-11:
Hi @easy98, no i have the same problem really i didn't follow with crazyswarm.
I'm using crazyflie_ros, but if you want maybe we can help us with crazyswarm and make a solution.

easy98, 2020-10-12:
I solved my problem, since I forget to change the vicon's host name in the hover.launch file.  Now it works.

eduardo-del, 2020-10-12:
Good for you @easy98

closed 2021-02-20 by jpreiss