USC-ACTLab/crazyswarm #229: Get data from Motive

gavtech, 2020-07-08:
We are students and we are trying to connect/interface motive optitrack with Matlab in real time for crazyflie. We already have the crazyflie running. We are following your instructions in github.

We have one Windows system with Motive running and the other with Ubuntu 18.04 with ROS on the same network. We are trying to connect with Motive to get data. But, we are not being able to connect. Can you please advice how to connect both under the same network. Do we need some additional thing that we are missing.

Thank you.

whoenig, 2020-07-09:
- Did you enable data streaming in motive, selecting the correct interface? (typically your windows PC will have two networks: one for the cameras and one for your regular network; make sure in motive you select the second interface)
- Can you ping the motive PC from your Ubuntu PC?
- If the two PCs are not connected over a simple switch, make sure there is no firewall blocking the NatNet UDP ports (1510, 1511). The ideal connection between the two PCs is either direct, or using a switch only (also to minimize latency.)
- For troubleshooting, I recommend using the "pure" NatNetSDK, before using the Crazyswarm. In particular, make sure that the test applications in https://github.com/whoenig/NatNetSDKCrossplatform report data correctly

gavtech, 2020-07-16:
Dear Whoenig,

  Thank you for your response.
> Yes, we enabled data streaming in motive and also the rest that you mentioned.
> Yes, we can ping bot the systems and no Firewall is enabled.
> Now, we are trying to connect with vrpn client on Linux and Motive in Windows but once we start the roslaunch after a point it shows creating tracker and after that nothing. Below are the code and pic from terminal

<launch>

        <arg name="server" default="192.168.32.50"/>

        <node pkg="vrpn_client_ros" type="vrpn_client_node" name="vrpn_client_node"
      output="screen">
          <rosparam subst_value="true">
            server: $(arg server)
            port: 3883

            update_frequency: 100.0
            frame_id: world

            # Use the VRPN server's time, or the client's ROS time.
            use_server_time: false
            broadcast_tf: true

            # Must either specify refresh frequency > 0.0, or a list of trackers to create
            refresh_tracker_frequency: 1.0
            #trackers:
            #- FirstTracker
            #- SecondTracker
          </rosparam>
        </node>
</launch>

[image]

Thank you again.

Regards
Gavtech

whoenig, 2020-07-16:
For vrpn_client_ros this is the intended behavior. You'll need to use `rviz` or `rostopic echo` to look at the streamed mocap data.

Note that VRPN and NatNet are different protocols with different features. So getting VRPN to work just rules out physical connection issues, but there can be still configuration issues on both Windows and Linux PC with respect to NatNet. The two protocols also use different ports, so in theory firewalls might block NatNet but not VRPN traffic.

closed 2021-02-20 by jpreiss