USC-ACTLab/crazyswarm #235: hover_swarm.launch does not output Motive data

ThomasCarstens, 2020-07-17:
Hello
Datastreaming is not occurring on Ubuntu 18.04 machine (tried on two separate machines) and I think it is related to the Optitrack implementation of Natnet.

My Windows to Linux Datastreaming reaches the Ubuntu 18.04 machine and I can confirm this with a tcpdump. However the mocap handler does not output any data whether as ros /tf or in roslaunch. The Windows clients however do read the data correctly (whether local or remotely). The fact that it does work should rule out any issue with my datastreaming settings,but I have made sure to only stream rigidBodies for motionCapture, and have selected Datastreaming parameters in Motive as advised in multiple other issues. Image is here:

[image]

First here are the parameters for optitrack that I have followed
`motion_capture_type: "optitrack" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "motionCapture" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      optitrack_host_name: "172.21.72.129"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
`

First I checked your NatnetCrossplatform implementation: $ ./natnettest 172.21.72.129
`
1 279
NatNetVersion: 128.150.152.0
ServerVersion: 3.1.0.0
`
and $ ./mcr 172.21.72.129
`
NatNetVersion: 128.150.152.0
ServerVersion: 3.1.0.0
`
So data is not streaming. Using a Windows machine I can confirm that data streams but I have now used 2 Ubuntu 18.04 machines with the same effect. Can't seem to find my error.

Additionally here are the outputs of $catkin_make -DCMAKE_BUILD_TYPE=Debug and $roslaunch crazyswarm hover_swarm.launch

$ catkin_make -DCMAKE_BUILD_TYPE=Debug

`Base path: /home/thomas/crazyswarm/ros_ws
Source space: /home/thomas/crazyswarm/ros_ws/src
Build space: /home/thomas/crazyswarm/ros_ws/build
Devel space: /home/thomas/crazyswarm/ros_ws/devel
Install space: /home/thomas/crazyswarm/ros_ws/install
####
#### Running command: "cmake /home/thomas/crazyswarm/ros_ws/src -DCMAKE_BUILD_TYPE=Debug -DCATKIN_DEVEL_PREFIX=/home/thomas/crazyswarm/ros_ws/devel -DCMAKE_INSTALL_PREFIX=/home/thomas/crazyswarm/ros_ws/install -G Unix Makefiles" in "/home/thomas/crazyswarm/ros_ws/build"
####
-- Using CATKIN_DEVEL_PREFIX: /home/thomas/crazyswarm/ros_ws/devel
-- Using CMAKE_PREFIX_PATH: /home/thomas/crazyswarm/ros_ws/devel;/home/thomas/crazyflie_ws/devel;/home/thomas/xarm_ws/devel;/home/thomas/ros_project_dependencies_ws/devel;/opt/ros/melodic
-- This workspace overlays: /home/thomas/crazyswarm/ros_ws/devel;/home/thomas/crazyflie_ws/devel;/home/thomas/xarm_ws/devel;/home/thomas/ros_project_dependencies_ws/devel;/opt/ros/melodic
-- Found PythonInterp: /usr/bin/python2 (found suitable version "2.7.17", minimum required is "2") 
-- Using PYTHON_EXECUTABLE: /usr/bin/python2
-- Using Debian Python package layout
-- Using empy: /usr/bin/empy
-- Using CATKIN_ENABLE_TESTING: ON
-- Call enable_testing()
-- Using CATKIN_TEST_RESULTS_DIR: /home/thomas/crazyswarm/ros_ws/build/test_results
-- Found gtest sources under '/usr/src/googletest': gtests will be built
-- Found gmock sources under '/usr/src/googletest': gmock will be built
-- Found PythonInterp: /usr/bin/python2 (found version "2.7.17") 
-- Using Python nosetests: /usr/bin/nosetests-2.7
-- catkin 0.7.23
-- BUILD_SHARED_LIBS is on
-- BUILD_SHARED_LIBS is on
-- ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
-- ~~  traversing 11 packages in topological order:
-- ~~  - crazyflie (metapackage)
-- ~~  - crazyflie_cpp
-- ~~  - crazyflie_description
-- ~~  - crazyflie_tools
-- ~~  - libmotioncapture
-- ~~  - libobjecttracker
[... 764 lines cut ...]
    rviz (rviz/rviz)

auto-starting new master
process[master]: started with pid [17165]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 025879bc-c82e-11ea-8951-3ca0679047f5
process[rosout-1]: started with pid [17447]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [17509]
process[joy-3]: started with pid [17515]
process[crazyswarm_teleop-4]: started with pid [17522]
[ERROR] [1594991053.349968161]: Couldn't open joystick /dev/input/js0. Will retry every second.
process[rviz-5]: started with pid [17537]
[ INFO] [1594991053.374597925]: Wait for services...
[ INFO] [1594991053.377152892]: waitForService: Service [/emergency] has not been advertised, waiting...
process[rqt_plot_roll-6]: started with pid [17548]
`

with gdb debugging (as in issue #122 but in my case I have not modified optitrack.cpp)
`
[image]
As you see the thread stops in its tracks, and a tcpdump shows that position data is streamed for 5s (and command data indefinitely) but hover_swarm motion capture implementation does not receive it.

`

ThomasCarstens, 2020-07-17:
tcpdump: 

# tcpdump -vvv -XXX -n -i wlp2s0 'port 1510'
tcpdump: listening on wlp2s0, link-type EN10MB (Ethernet), capture size 262144 bytes
15:59:41.214639 IP (tos 0x0, ttl 64, id 28898, offset 0, flags [DF], proto UDP (17), length 32)
    172.21.72.166.40365 > 172.21.72.129.1510: [bad udp cksum 0xe96f -> 0x72f0!] UDP, length 4
	0x0000:  84a9 3e63 e471 3ca0 6790 47f5 0800 4500  ..>c.q<.g.G...E.
	0x0010:  0020 70e2 4000 4011 e098 ac15 48a6 ac15  ..p.@.@.....H...
	0x0020:  4881 9dad 05e6 000c e96f 0000 0000       H........o....
15:59:41.217759 IP (tos 0x0, ttl 128, id 33764, offset 0, flags [none], proto UDP (17), length 311)
    172.21.72.129.1510 > 172.21.72.166.40365: [udp sum ok] UDP, length 283
	0x0000:  3ca0 6790 47f5 84a9 3e63 e471 0800 4500  <.g.G...>c.q..E.
	0x0010:  0137 83e4 0000 8011 cc7f ac15 4881 ac15  .7..........H...
	0x0020:  48a6 05e6 9dad 0123 c840 0100 1701 4d6f  H......#.@....Mo
	0x0030:  7469 7665 0069 6200 3100 0100 0000 ffff  tive.ib.1.......
	0x0040:  ffff 0000 0000 0000 0000 0000 0000 0300  ................
	0x0050:  0000 b08c cd3d e95f f1b7 29ac 89bd a1a9  .....=._..).....
	0x0060:  4dbd ff59 8338 0a5e 89bd 4670 4dbd 0a04  M..Y.8.^..FpM...
	0x0070:  0eb8 2785 093e 0000 0000 0000 0000 0000  ..'..>..........
	0x0080:  0000 0000 0000 6366 3100 0300 0000 4d61  ......cf1.....Ma
	0x0090:  726b 6572 3100 4d61 726b 6572 3200 4d61  rker1.Marker2.Ma
	0x00a0:  726b 6572 3300 0000 0000 616c 6c00 0300  rker3.....all...
	0x00b0:  0000 6366 315f 3100 6366 315f 3200 6366  ..cf1_1.cf1_2.cf
	0x00c0:  315f 3300 0500 0000 466c 6578 3133 2023  1_3.....Flex13.#
	0x00d0:  3431 3039 3938 0079 1707 bff9 4fdb 3f2e  410998.y....O.?.
	0x00e0:  248f beda 683e 3e78 d953 3f0f 86db 3e72  $...h>>x.S?...>r
	0x00f0:  469f be05 0000 0046 6c65 7831 3320 2334  F......Flex13.#4
	0x0100:  3130 3939 3500 8a94 16bf b978 dd3f 8f41  10995......x.?.A
	0x0110:  893f 0730 ebbe 443d bcbe a7fd 33be ec0c  .?.0..D=....3...
	0x0120:  4a3f 0500 0000 466c 6578 3133 2023 0202  J?....Flex13.#..
	0x0130:  0000 0301 0000 8096 9800 0000 0000 e705  ................
	0x0140:  01ef ff2a 63                             ...*c
15:59:41.217900 IP (tos 0x0, ttl 64, id 28899, offset 0, flags [DF], proto UDP (17), length 32)
    172.21.72.166.40365 > 172.21.72.129.1510: [bad udp cksum 0xe96f -> 0x6ef0!] UDP, length 4
	0x0000:  84a9 3e63 e471 3ca0 6790 47f5 0800 4500  ..>c.q<.g.G...E.
	0x0010:  0020 70e3 4000 4011 e097 ac15 48a6 ac15  ..p.@.@.....H...
	0x0020:  4881 9dad 05e6 000c e96f 0400 0000       H........o....
15:59:41.220277 IP (tos 0x0, ttl 128, id 33765, offset 0, flags [none], proto UDP (17), length 370)
    172.21.72.129.1510 > 172.21.72.166.40365: [udp sum ok] UDP, length 342
	0x0000:  3ca0 6790 47f5 84a9 3e63 e471 0800 4500  <.g.G...>c.q..E.
	0x0010:  0172 83e5 0000 8011 cc43 ac15 4881 ac15  .r.......C..H...
	0x0020:  48a6 05e6 9dad 015e 5c87 0500 5201 0700  H......^\...R...
	0x0030:  0000 0100 0000 6366 3100 0100 0000 ffff  ......cf1.......
	0x0040:  ffff 0000 0000 0000 0000 0000 0000 0300  ................
	0x0050:  0000 b08c cd3d e95f f1b7 29ac 89bd a1a9  .....=._..).....
	0x0060:  4dbd ff59 8338 0a5e 89bd 4670 4dbd 0a04  M..Y.8.^..FpM...
	0x0070:  0eb8 2785 093e 0000 0000 0000 0000 0000  ..'..>..........
	0x0080:  0000 0000 0000 6366 3100 0300 0000 4d61  ......cf1.....Ma
	0x0090:  726b 6572 3100 4d61 726b 6572 3200 4d61  rker1.Marker2.Ma
	0x00a0:  726b 6572 3300 0000 0000 616c 6c00 0300  rker3.....all...
	0x00b0:  0000 6366 315f 3100 6366 315f 3200 6366  ..cf1_1.cf1_2.cf
	0x00c0:  315f 3300 0500 0000 466c 6578 3133 2023  1_3.....Flex13.#
	0x00d0:  3431 3039 3938 0079 1707 bff9 4fdb 3f2e  410998.y....O.?.
	0x00e0:  248f beda 683e 3e78 d953 3f0f 86db 3e72  $...h>>x.S?...>r
	0x00f0:  469f be05 0000 0046 6c65 7831 3320 2334  F......Flex13.#4
	0x0100:  3130 3939 3500 8a94 16bf b978 dd3f 8f41  10995......x.?.A
	0x0110:  893f 0730 ebbe 443d bcbe a7fd 33be ec0c  .?.0..D=....3...
	0x0120:  4a3f 0500 0000 466c 6578 3133 2023 3431  J?....Flex13.#41
	0x0130:  3039 3937 002b 0b7e 3fdb c9dc 3fec 4cba  0997.+.~?...?.L.
	0x0140:  3fe0 f1e4 beac 7b88 3e36 602a 3e8d 6256  ?.....{.>6`*>.bV
	0x0150:  3f05 0000 0046 6c65 7831 3320 2334 3130  ?....Flex13.#410
	0x0160:  3939 3600 aac7 8a3f d1ad dc3f 47eb 35be  996....?...?G.5.
	0x0170:  e97a 25be 8254 4c3f 3350 db3e a97d c83e  .z%..TL?3P.>.}.>
^C
4 packets captured
4 packets received by filter
0 packets dropped by kernel

whoenig, 2020-07-17:
Thanks for the detailed report. This seems to be an issue with `NatNet` and not really with the Crazyswarm usage of NatNet, so we can focus on debugging NatNetSDKCrossplatform first (see https://github.com/whoenig/NatNetSDKCrossplatform). 

From your report, it looks like the command port (1510) works fine, but you don't receive any data (since `mcr` does not report anything). Note that the port for the data is different (1511 by default) and that it uses a different protocol, so you would need to check if you can receive multicast data on 239.255.42.99:1511.

I have used Ubuntu 18.04 with Motive before, so I don't think its an OS issue. Which Motive version are you running? Also, please double check the settings in Motive for streaming - perhaps the default multicast address/port have been changed in Motive.

ThomasCarstens, 2020-07-20:
Okay, I do receive data on  the port as a multicast but the ./mcr test will only give me server version and natnet version (although it does, in fact, trigger entry of data). In fact something seems to be interrupting the Natnet service? I have joined a strace ./mcr but maybe you will see something I do not?

# tcpdump -vvv -XXX -n -i wlp2s0 'port 1511'

18:22:16.076519 IP (tos 0x0, ttl 2, id 25138, offset 0, flags [none], proto UDP (17), length 110)
    172.21.72.126.1511 > 239.255.42.99.1511: [udp sum ok] UDP, length 82
	0x0000:  3ca0 6790 47f5 f406 690f 1f7c 0800 4500  <.g.G...i..|..E.
	0x0010:  006e 6232 0000 0211 4757 ac15 487e efff  .nb2....GW..H~..
	0x0020:  2a63 05e7 05e7 005a 6ef5 0700 4e00 1693  *c.....Zn...N...
	0x0030:  0c00 0000 0000 0000 0000 0000 0000 0000  ................
	0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0050:  0000 0000 0000 40ce ead3 30bf 8240 0000  ......@...0..@..
	0x0060:  0000 0000 0000 cdbb de58 2800 0000 44dd  .........X(...D.
	0x0070:  de58 2800 0000 0000 0000 0000            .X(.........
18:22:16.085838 IP (tos 0x0, ttl 2, id 25139, offset 0, flags [none], proto UDP (17), length 110)
    172.21.72.126.1511 > 239.255.42.99.1511: [udp sum ok] UDP, length 82
	0x0000:  3ca0 6790 47f5 f406 690f 1f7c 0800 4500  <.g.G...i..|..E.
	0x0010:  006e 6233 0000 0211 4756 ac15 487e efff  .nb3....GV..H~..
	0x0020:  2a63 05e7 05e7 005a 8c3b 0700 4e00 1793  *c.....Z.;..N...
	0x0030:  0c00 0000 0000 0000 0000 0000 0000 0000  ................
	0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0050:  0000 0000 0000 7186 2c1c 42bf 8240 0000  ......q.,.B..@..
	0x0060:  0000 0000 0000 ad06 e058 2800 0000 bf4b  .........X(....K
	0x0070:  e058 2800 0000 0000 0000 0000            .X(.........
18:22:16.094041 IP (tos 0x0, ttl 2, id 25140, offset 0, flags [none], proto UDP (17), length 110)
    172.21.72.126.1511 > 239.255.42.99.1511: [udp sum ok] UDP, length 82
	0x0000:  3ca0 6790 47f5 f406 690f 1f7c 0800 4500  <.g.G...i..|..E.
	0x0010:  006e 6234 0000 0211 4755 ac15 487e efff  .nb4....GU..H~..
	0x0020:  2a63 05e7 05e7 005a e5ca 0700 4e00 1893  *c.....Z....N...
	0x0030:  0c00 0000 0000 0000 0000 0000 0000 0000  ................
	0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0050:  0000 0000 0000 415b 1833 53bf 8240 0000  ......A[.3S..@..
	0x0060:  0000 0000 0000 5a4c e158 2800 0000 e88a  ......ZL.X(.....
	0x0070:  e158 2800 0000 0000 0000 0000            .X(.........
18:22:16.102284 IP (tos 0x0, ttl 2, id 25141, offset 0, flags [none], proto UDP (17), length 110)
    172.21.72.126.1511 > 239.255.42.99.1511: [udp sum ok] UDP, length 82
	0x0000:  3ca0 6790 47f5 f406 690f 1f7c 0800 4500  <.g.G...i..|..E.
	0x0010:  006e 6235 0000 0211 4754 ac15 487e efff  .nb5....GT..H~..
	0x0020:  2a63 05e7 05e7 005a a640 0700 4e00 1993  *c.....Z.@..N...
	0x0030:  0c00 0000 0000 0000 0000 0000 0000 0000  ................
	0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0050:  0000 0000 0000 dc68 5b28 64bf 8240 0000  .......h[(d..@..
	0x0060:  0000 0000 0000 d690 e258 2800 0000 b9cd  .........X(.....
	0x0070:  e258 2800 0000 0000 0000 0000            .X(.........
18:22:16.109810 IP (tos 0x0, ttl 2, id 25142, offset 0, flags [none], proto UDP (17), length 110)
    172.21.72.126.1511 > 239.255.42.99.1511: [udp sum ok] UDP, length 82
	0x0000:  3ca0 6790 47f5 f406 690f 1f7c 0800 4500  <.g.G...i..|..E.
	0x0010:  006e 6236 0000 0211 4753 ac15 487e efff  .nb6....GS..H~..
	0x0020:  2a63 05e7 05e7 005a 96a7 0700 4e00 1a93  *c.....Z....N...
	0x0030:  0c00 0000 0000 0000 0000 0000 0000 0000  ................
	0x0040:  0000 0000 0000 0000 0000 0000 0000 0000  ................
	0x0050:  0000 0000 0000 94b6 5d0d 75bf 8240 0000  ........].u..@..
	0x0060:  0000 0000 0000 0fd1 e358 2800 0000 c1f3  .........X(.....
	0x0070:  e358 2800 0000 0000 0000 0000            .X(.........
^C
1243 packets captured
1555 packets received by filter
312 packets dropped by kernel

$ strace ./mcr 172.21.72.126
execve("./mcr", ["./mcr", "172.21.72.126"], 0x7ffc0d6d7688 /* 85 vars */) = 0
brk(NULL)                               = 0x55aa787d6000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7fc033704000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls/haswell/x86_64", 0x7ffdf4e98140) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls/haswell", 0x7ffdf4e98140) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... 222 lines cut ...]
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
[... cut ...]
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8, 0x7ffdf4e98150, 128, -1)  = -1 EINTR (Interrupted system call)
--- SIGWINCH {si_signo=SIGWINCH, si_code=SI_KERNEL} ---
epoll_wait(8,

whoenig, 2020-07-20:
I just pushed a [change](https://github.com/whoenig/NatNetSDKCrossplatform/commit/8253ce147bcd611ed4f872a193d9f46d03c9c69c) to NatNetSDKCrossplatform with some additional error reporting. It looks boost asio is causing some trouble. Your strace error is pretty low level, so I hope that the boost asio one might be a bit more informative. Perhaps you could run that on your end?

ThomasCarstens, 2020-07-20:
I run ./mcr and there's the same output (after pulling the change and compiling again). I'm not familiar with boost asio, I assume you mean another strace?

$ strace ./mcr 172.21.72.126
execve("./mcr", ["./mcr", "172.21.72.126"], 0x7ffd457269c8 /* 85 vars */) = 0
brk(NULL)                               = 0x5616aa020000
access("/etc/ld.so.nohwcap", F_OK)      = -1 ENOENT (No such file or directory)
mmap(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x7f2c50a4c000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls/haswell/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls/haswell", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/tls/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/tls", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/haswell/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/haswell", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/crazyflie_ws/devel/lib/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/crazyflie_ws/devel/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/tls/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/tls/haswell/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/tls/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/tls/haswell", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/tls/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/tls/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/tls/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/tls", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/haswell/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/haswell", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/xarm_ws/devel/lib/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/xarm_ws/devel/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}) = 0
openat(AT_FDCWD, "/home/thomas/ros_project_dependencies_ws/devel/lib/tls/haswell/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/ros_project_dependencies_ws/devel/lib/tls/haswell/x86_64", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/ros_project_dependencies_ws/devel/lib/tls/haswell/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
stat("/home/thomas/ros_project_dependencies_ws/devel/lib/tls/haswell", 0x7ffe97d3bb90) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/home/thomas/ros_project_dependencies_ws/devel/lib/tls/x86_64/libboost_system.so.1.65.1", O_RDONLY|O_CLOEXEC) = -1 ENOENT (No such file or directory)
[... cut ...]
epoll_ctl(8, EPOLL_CTL_ADD, 10, {EPOLLIN|EPOLLPRI|EPOLLERR|EPOLLHUP|EPOLLET, {u32=2852455744, u64=94655341724992}}) = 0
setsockopt(10, SOL_SOCKET, SO_REUSEADDR, [1], 4) = 0
bind(10, {sa_family=AF_INET, sin_port=htons(1511), sin_addr=inet_addr("0.0.0.0")}, 16) = 0
setsockopt(10, SOL_IP, IP_ADD_MEMBERSHIP, {imr_multiaddr=inet_addr("239.255.42.99"), imr_interface=inet_addr("0.0.0.0")}, 8) = 0
ioctl(10, FIONBIO, [1])                 = 0
recvmsg(10, {msg_namelen=28}, 0)        = -1 EAGAIN (Resource temporarily unavailable)
epoll_wait(8, [{EPOLLIN, {u32=2852435624, u64=94655341704872}}], 128, -1) = 1
epoll_wait(8,

whoenig, 2020-07-21:
Boost asio is a C++ library we rely on for the networking component (i.e., receiving multi-cast packets). I read up on some very old issues and it can be related to a machine that has multiple network cards and a missing route (see https://svn.boost.org/trac10/ticket/6446, first comment). The easiest fix seems to be to join the multicast group by specifying your local IP directly (see https://stackoverflow.com/questions/18491877/boostasio-multicast-listen-address).

There is a new branch with those changes for you to test (https://github.com/whoenig/NatNetSDKCrossplatform/tree/bug-recv-issue). There is no need for strace, but please run `netstat -g` if there are still issues. Note that this new version requires two arguments: `mcr <optitrack-ip> <local-interface-ip>`.

ThomasCarstens, 2020-07-21:
$ ./mcr 172.21.72.126 192.168.1.101
NatNetVersion: 128.150.152.0
ServerVersion: 3.1.0.0

$ netstat -g
IPv6/IPv4 Group Memberships
Interface       RefCnt Group
--------------- ------ ---------------------
lo              1      224.0.0.251
lo              1      all-systems.mcast.net
eno1            1      239.255.42.99
eno1            1      224.0.0.251
eno1            1      all-systems.mcast.net
lo              1      ff02::fb
lo              1      ip6-allnodes
lo              1      ff01::1
eno1            1      ff02::fb
eno1            1      ff02::1:ffd6:f7d1
eno1            1      ip6-allnodes
eno1            1      ff01::1

Group does exist on eno1 which you can see in ifconfig.

ThomasCarstens, 2020-07-21:
I have noticed that a Linux SampleClient from NaturalPoint also does not output data. The code is unsupported at this stage , it does give a log of errors from the NatNetlib API that could help here. It shows descriptions being read and somehow the connection is dropped. I suspect that the receiver buffer is too large but I have not found how to change the size of the buffer. Your own code seems to attribute a size dynamically, is that correct?

NatNet Sample Client (NatNet ver. 3.1.0.0)

[NatNetLib]  [INFO]: [Client] Initializing using ethernet address: 192.168.1.101

[NatNetLib]  [INFO]: [Client] Initializing command listener.

[NatNetLib] [ERROR]: [Client] Receive Buffer Size: 425984

[NatNetLib]  [INFO]: [Client] Host information received. (HostApp: Motive  HostVer: 2.2.0.0  HostIP: 172.21.72.126 ()

[NatNetLib]  [INFO]: [Client] Initializing data listener.

[NatNetLib]  [WARN]: [HELPER] ReceiveBuffer size = 425984

[NatNetLib] [ERROR]: [Client] Unable to get host information for 172.21.72.126.

[NatNetLib]  [INFO]: [Client] Connected.

[SampleClient] Server application info:
Application: Motive (ver. 2.2.0.0)
NatNet Version: 3.1.0.0
Client IP:(null)
Server IP:172.21.72.126
Server Name:
Mocap Framerate : 120.00
Analog Samples Per Mocap Frame : 0
Client initialized and ready.
[SampleClient] Sending Test Request

[SampleClient] Requesting Data Descriptions...[SampleClient] Received 7 Data Descriptions:
Data Description # 0 (type=1)
RigidBody Name : RigidBody
RigidBody ID : 1
RigidBody Parent ID : -1
Parent Offset : 0.00,0.00,0.00
	Marker #0:
		Position: 0.10, 0.00, -0.07
	Marker #1:
		Position: -0.04, -0.00, 0.14
	Marker #2:
		Position: -0.05, 0.00, -0.06
Data Description # 1 (type=0)
MarkerSet Name : RigidBody
Marker1
Marker2
Marker3
Data Description # 2 (type=0)
MarkerSet Name : all
RigidBody_1
RigidBody_2
RigidBody_3
Data Description # 3 (type=5)
Unknown data type.Data Description # 4 (type=5)
Unknown data type.Data Description # 5 (type=5)
Unknown data type.Data Description # 6 (type=5)
Unknown data type.
Client is connected to server and listening for data...

[NatNetLib] [ERROR]: Socket error 107: Transport endpoint is not connected

[NatNetLib] [ERROR]: Socket error 107: Transport endpoint is not connected

whoenig, 2020-07-22:
1. So it looks like `mcr` does the right thing now, at least according to `netstat` and `ifconfig`. Did you double check that `tcpdump` still receives data while mcr is running and not showing any output? Also I noticed that the interface seems to have changed - in your previous report your interface name was `wlp2s0`, but now it is `eno1`?
2. Good to know that there is a Linux Sample Client now. Too bad that it is still not officially supported. From your output, it actually looks like it is the same problem: the data channel works fine (the data description query seems to be done on the data channel), but your multicast doesn't (getting that socket error 107). The `Unknown data type.Data Description` is probably not a problem: perhaps your Motive 2.2 added some new features that are not in the SDK, yet. If it is an option, you might want to downgrade to an older Motive version and try again.
3. Perhaps you could try a simple multicast receiver in another language? Here is an [example for Python](https://stackoverflow.com/questions/603852/how-do-you-udp-multicast-in-python). If there is an error or any semantic difference with respect to the OS calls being made, that might help to investigate further.

ThomasCarstens, 2020-09-10:
To get back to you on this, I now have Optitrack streaming to Linux. There were three key issues:
Multicast had to be enabled on Linux,
The firewall had to accept streaming info on the ports, on top of being disabled 
The machines had to ping to eachother for an unknown reason before multicast being triggered
That's it, glad to send more if needed

whoenig, 2020-09-11:
Thanks for the update - that's great. I'd like to improve the documentation with this information. Would you be able to add some more details? Especially the first point is unclear to me. Also, is the ping required just once? Before or after the receiver application is launched?

whoenig, 2021-09-24:
Closing this due to inactivity.

closed 2021-09-24 by whoenig