USC-ACTLab/crazyswarm #236: How to calculate the size of the group in the Streaming setpoint modes

ucaszzw, 2020-07-20:
I hope to control the crazyflies group in Streaming setpoint modes. I have a problem now. I don't know what the maximum size of the group I can control.

**1. My first question is how to calculate the maximum amount I can control based on the communication bandwidth.**
> Streaming setpoint modes simplify the process of developing new high-level planners, since the planner can run on the PC where it has access to more computational resources. However, they require more radio bandwidth, so they do not scale as well to large numbers of robots. Currently, we have not validated anything larger than 7 robots on 3 radios.
I saw such a sentence on the wiki, but I want to know, at what frequency is it achieved?
**1.1. Can I reduce the frequency of command transmission to achieve a larger cluster?
1.2. If I have more USB Radio, can I control a larger swarm?**

**2. My second question is, how realistic is the simulation in crazyswarm?**
Before using Streaming setpoint modes, I need to use simulation to verify my algorithm. In crazyswarm, we can add` --sim `before running the program to realize the simulation of the program. For example, when sending the speed vector command to the crazyflie, the PID algorithm has a relaxation time, and it is impossible for the drone to reach the specified speed instantly. I want to know whether the simulation function in crazyswarm can truly simulate the characteristics of drones.

jpreiss, 2020-07-20:
@whoenig knows more about communication, but roughly the calculation for 1.1 would be like 

                        USB poll rate * packets per USB request
    robots per radio = -----------------------------------------
                          Total packets needed * desired rate

I believe the peak USB poll rate is 1000 Hz, and we can send 2 Crazyradio packets per USB request. The packets needed will be around 1.5 per robot - one for the setpoint, and motion capture measurements fit 2 in a single packet.

So if you want an update rate of 100 Hz, in theory you could control up to 13 robots on a single radio. Adding a fudge factor of 2x for USB congestion and all the other unmodeled problems, would suggest around 6-7 Crazyflies per radio. This seems reasonable to me.

1.1 Yes. Your Python script controls the rate of streaming setpoints directly. The rate of motion capture updates is configured in the mocap software, e.g. Vicon Tracker. I think you can go much lower than 100 Hz safely.

1.2 Yes.

2. Not realistic at all. The simulated Crazyflie does whatever you ask, even if it is physically impossible.

ucaszzw, 2020-07-20:
@jpreiss Thank you for such a quick response. I am trying to control 15-20 crazyflies using streaming setpoint command. I was particularly anxious when I saw that there are currently no more than 7 verifications. Thanks for your answer, I will try as soon as possible.

whoenig-overleaf, 2020-07-20:
Great answer already, just some comments:

1. What James calculated makes sense to me also. As pointed out in 1.2, you can safely reduce the frequency of setpoints to get a larger number. Reducing the mocap updates works better if you off-load more to the on-board EKF in the first place, i.e. use single-marker tracking. For 1.2, remember to use different channels for each Crazyradio.

2. If you want physics simulation, take a look at [CrazyS](https://github.com/gsilano/CrazyS).

ucaszzw, 2020-07-20:
Thank you for your reply. Your answer is very helpful to me.

jpreiss, 2020-07-20:
@ucaszzw please let us know what you find out so we can update the documentation. As far as I know, nobody has ever tried to explicitly maximize `robots / radio` by tuning the rate and checking stability. It would be great to have an experimentally determined upper bound.

The 7 robots / 3 radios is simply the largest streaming-setpoint experiment I have tried so far. Our PC always has 3 radios plugged in, and I never tried using streaming setpoints with more than 7 Crazyflies.

ucaszzw, 2020-07-20:
@jpreiss Of course, I will. This is my main task for the next two months. I just checked it on Google, and it seems that the default value of ubuntu's usb poll rate is 125HZ. When you are doing 7 crazyflies test, did you change it to 1000HZ?

whoenig, 2020-07-20:
There is no need to change any OS setting. This estimate is based on how frequently the user code using libUSB can exchange data over USB (active polling). You can measure this on your hardware using `rosrun crazyflie_tools comCheck`.

ucaszzw, 2020-07-21:
@whoenig Thanks for your reply, but I don’t know if I fully understand it. In other words, the USB poll rate does not need to be adjusted manually, it will change automatically according to the program. I can manually adjust the rate of streaming setpoints and the rate of motion capture updates in the program.

whoenig, 2020-07-21:
Yes, no need to change anything on your side.

closed 2021-02-20 by jpreiss