USC-ACTLab/crazyswarm #237: Autonomous flight using the ROS packages and Flow Deck

santiagorg2401, 2020-07-20:
Hi,
I followed the steps in https://crazyswarm.readthedocs.io/en/latest/usage.html, but I have a problem, by the moment I am using Flow Deck but the goal is to use LPS, so after tagging the Crazyflies with 1,2,3,4,..., A, B, and writing them in allCrazyflies.yaml
crazyflies:1
...
crazyflies : 10
...
and so on
The GUI chooser.py shows me this error when clicking any button:
[rospack] Error: package 'crazyflie_tools' not found
Exception in Tkinter callback
Traceback (most recent call last):
File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1550, in __call__
return self.func(*args)
File "chooser.py", line 213, in checkBattery
voltage = subprocess.check_output(["rosrun crazyflie_tools battery --uri " + uri], shell=True)
File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['rosrun crazyflie_tools battery --uri radio://0/80/2M/E7E7E7E708']' returned non-zero exit status 2

Then, when I try to fly the drone with (for example) niceHover.py, first I run the crayswarm server
auto-starting new master
process[master]: started with pid [9403]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to d6b7ebd0-c611-11ea-bc79-181deab367da
process[rosout-1]: started with pid [9414]
started core service [/rosout]
process[crazyswarm_server-2]: started with pid [9421]
ERROR: cannot launch node of type [joy/joy_node]: joy
ROS path [0]=/opt/ros/melodic/share/ros
ROS path [1]=/home/santiagorg2401/crazyflie/crazyswarm/ros_ws/src
ROS path [2]=/opt/ros/melodic/share
process[crazyswarm_teleop-4]: started with pid [9422]
process[rviz-5]: started with pid [9423]
[ INFO] [1594759051.849131804]: Wait for services...
[ INFO] [1594759051.851109082]: waitForService: Service [/emergency] has not been advertised, waiting...
[ INFO] [1594759051.918606844]: waitForService: Service [/emergency] is now available.
ch: 1
[ INFO] [1594759051.932975640]: Manager ready.
terminate called after throwing an instance of 'XmlRpc::XmlRpcException'
[crazyswarm_server-2] process has died [pid 9421, exit code -6, cmd /home/santiagorg2401/crazyflie/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/santiagorg2401/.ros/log/d6b7ebd0-c611-11ea-bc79-181deab367da/crazyswarm_server-2.log].
log file: /home/santiagorg2401/.ros/log/d6b7ebd0-c611-11ea-bc79-181deab367da/crazyswarm_server-2*.log

And finally in another terminal window, the niceHover.py
Traceback (most recent call last):
File "niceHover.py", line 9, in <module>
swarm = Crazyswarm()
File "/home/santiagorg2401/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyswarm.py", line 43, in __init__
import crazyflie
File "/home/santiagorg2401/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py", line 12, in <module>
from crazyflie_driver.srv import *
ImportError: No module named crazyflie_driver.srv

I do not really know why is those errors happening.

whoenig, 2020-07-21:
For your first problem, make sure you source your ROS workspace (`source ~/crazyswarm/ros_ws/source/devel.sh`) in every terminal that you use. For your second problem, there seems to be an issue with the formatting of your configuration files (either `hover_swarm.launch` or `crazyflies.yaml`) -- please post/attach the files here.

santiagorg2401, 2020-07-22:
crazyflies.yaml
crazyflies:
- channel: 80
  id: 8
  initialPosition: [0, 0, 0.0]
  type: default

-hover_swarm.launch

<?xml version="1.0"?>
<launch>
  <arg name="joy_dev" default="/dev/input/js0" />

  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflieTypes.yaml" />
  <rosparam command="load" file="$(find crazyswarm)/launch/crazyflies.yaml" />

  <node pkg="crazyswarm" type="crazyswarm_server" name="crazyswarm_server" output="screen" >
    <rosparam>
      world_frame: "/world"
      # Logging configuration (Use enable_logging to actually enable logging)
      genericLogTopics: ["log1"]
      genericLogTopicFrequencies: [10]
      genericLogTopic_log1_Variables: ["stateEstimate.x", "ctrltarget.x"]
      # firmware parameters for all drones (use crazyflieTypes.yaml to set per type, or
      # allCrazyflies.yaml to set per drone)
      firmwareParams:
        commander:
          enHighLevel: 1
        stabilizer:
          estimator: 2 # 1: complementary, 2: kalman
          controller: 2 # 1: PID, 2: mellinger
        ring:
          effect: 16 # 6: double spinner, 7: solid color, 16: packetRate
          solidBlue: 255 # if set to solid color
          solidGreen: 0 # if set to solid color
          solidRed: 0 # if set to solid color
          headlightEnable: 0
        locSrv:
          extPosStdDev: 1e-3
          extQuatStdDev: 0.5e-1
        kalman:
          resetEstimation: 1
      # tracking
      motion_capture_type: "none" # one of none,vicon,optitrack,qualisys,vrpn
      object_tracking_type: "libobjecttracker" # one of motionCapture,libobjecttracker
      send_position_only: False # set to False to send position+orientation; set to True to send position only
      vicon_host_name: "vicon"
      # optitrack_host_name: "optitrack"
      # qualisys_host_name: "10.0.5.219"
      # qualisys_base_port: 22222
      # vrpn_host_name: "vicon"
      save_point_clouds: ~/pointCloud.ot
      print_latency: False
      write_csvs: False
      force_no_cache: False
      enable_parameters: True
      enable_logging: False
    </rosparam>
  </node>

  <node name="joy" pkg="joy" type="joy_node" output="screen">
    <param name="dev" value="$(arg joy_dev)" />
  </node>

  <node pkg="crazyswarm" type="crazyswarm_teleop" name="crazyswarm_teleop" output="screen">
    <param name="csv_file" value="$(find crazyswarm)/launch/figure8_smooth.csv" />
    <param name="timescale" value="0.8" />
  </node>

  <node name="rviz" pkg="rviz" type="rviz" args="-d $(find crazyswarm)/launch/test.rviz"/>

  
  

</launch>

whoenig, 2020-07-22:
Please use "insert code" in the toolbar, or upload the files somewhere and share the link. The configuration is whitespace sensitive, so I can't tell if there is an issue from your previous comment.

santiagorg2401, 2020-07-22:
[Files.zip](https://github.com/USC-ACTLab/crazyswarm/files/4962874/Files.zip)

whoenig, 2020-07-23:
The only thing I spot is the initialPosition in `crazyflies.yaml`. I remember there being a parsing bug in ROS that requires this to be a floating point number, i.e., `[0.0, 0.0, 0.0]` rather than `[0, 0, 0.0]`.

santiagorg2401, 2020-07-24:
[image]
I did it setting the zeros as 0.0, and the source command and the chooser.py is still not working

santiagorg2401, 2020-07-24:
[image]
Terminal 1.
`santiagorg2401@santiagorg2401-G7-7588:~$ source crazyflie/crazyswarm/ros_ws/devel/setup.bash 
santiagorg2401@santiagorg2401-G7-7588:~$ python crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/chooser.py 
timeout
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1550, in __call__
    return self.func(*args)
  File "crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/chooser.py", line 213, in checkBattery
    voltage = subprocess.check_output(["rosrun crazyflie_tools battery --uri " + uri], shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['rosrun crazyflie_tools battery --uri radio://0/80/2M/E7E7E7E706']' returned non-zero exit status 1
6
timeout
timeout
Exception in Tkinter callback
Traceback (most recent call last):
  File "/usr/lib/python2.7/lib-tk/Tkinter.py", line 1550, in __call__
    return self.func(*args)
  File "crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/chooser.py", line 213, in checkBattery
    voltage = subprocess.check_output(["rosrun crazyflie_tools battery --uri " + uri], shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 223, in check_output
    raise CalledProcessError(retcode, cmd, output=output)
CalledProcessError: Command '['rosrun crazyflie_tools battery --uri radio://0/80/2M/E7E7E7E706']' returned non-zero exit status 1
Flash STM32 FW to radio://0/80/2M/E7E7E7E706
timeout
santiagorg2401@santiagorg2401-G7-7588:~$ python crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/chooser.py 
`
Terminal 2.
`santiagorg2401@santiagorg2401-G7-7588:~$ source crazyflie/crazyswarm/ros_ws/devel/setup.bash 
santiagorg2401@santiagorg2401-G7-7588:~$ roslaunch crazyswarm hover_swarm.launch 
... logging to /home/santiagorg2401/.ros/log/3456f9ce-cdbb-11ea-ab55-181deab367da/roslaunch-santiagorg2401-G7-7588-5535.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://santiagorg2401-G7-7588:46631/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/CF21SingleMarker/bigQuad: False
 * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
 * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
[... 179 lines cut ...]
[ INFO] [1595601452.206348997]: waitForService: Service [/emergency] is now available.
[ INFO] [1595601452.221361957]: Manager ready.
ch: 1
[ INFO] [1595601452.235945328]: Adding CF: cf6 (radio://0/80/2M/E7E7E7E706, cf6)...
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[crazyswarm_server-2] process has died [pid 5565, exit code -6, cmd /home/santiagorg2401/crazyflie/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/santiagorg2401/.ros/log/3456f9ce-cdbb-11ea-ab55-181deab367da/crazyswarm_server-2.log].
log file: /home/santiagorg2401/.ros/log/3456f9ce-cdbb-11ea-ab55-181deab367da/crazyswarm_server-2*.log
`
Terminal 3
`santiagorg2401@santiagorg2401-G7-7588:~$ source crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/niceHover.py 
import-im6.q16: not authorized `np' @ error/constitute.c/WriteImage/1037.
from: can't read /var/mail/pycrazyswarm
Z: orden no encontrada
bash: crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/niceHover.py: línea 9: error sintáctico cerca del elemento inesperado `('
bash: crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/niceHover.py: línea 9: `    swarm = Crazyswarm()'
santiagorg2401@santiagorg2401-G7-7588:~$ source crazyflie/crazyswarm/ros_ws/devel/setup.bash 
santiagorg2401@santiagorg2401-G7-7588:~$ python crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/niceHover.py 
Traceback (most recent call last):
  File "crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/niceHover.py", line 9, in <module>
    swarm = Crazyswarm()
  File "/home/santiagorg2401/crazyflie/crazyswarm/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyswarm.py", line 36, in __init__
    crazyflies_yaml = open(crazyflies_yaml, 'r').read()
IOError: [Errno 2] No such file or directory: '../launch/crazyflies.yaml'
`

whoenig, 2020-07-25:
- The timeout error indicates that you can't connect to the CF. Check your URI `radio://0/80/2M/E7E7E7E706` and that you can connect to that CF with other means, such as the `cfclient`.
- The niceHover.py script needs to be launched from the scripts folder, not the home-folder, otherwise the relative paths don't work.

santiagorg2401, 2020-07-28:
I will try again when being in the lab and I will post the results.
My brother is covid-19 suspect, so I have to wait for his results.
Thank you for answering Mr. Whönig

santiagorg2401, 2020-08-05:
Hi Mr Whönig,
I ran the following terminal windows but I got this
[image]

I checked with cfclient if the Crazyflie is able to connect and it is
[image]

santiagorg2401, 2020-08-05:
Hi, I re-installed all the packages, and this is what happens:
[image]
Checking the address witch cfclient:
[image]

santiagorg2401, 2020-08-05:
Update:
The battery button is still not working but I could make them fly, but there are 2 problems, sometimes Rviz does not connect to the Crazyflie and it takes several tries until it connects, and many times when the Crazyflie starts flying, it just goes anywhere out the trajectory and crashes

whoenig, 2020-08-06:
- For the battery, make sure you use a recent NRF51 firmware and crazyswarm. In particular, make sure you picked up [this fix](https://github.com/bitcraze/crazyflie2-nrf-firmware/commit/eaffde62d30358668647ac05bacc6a02f2241d1c).
- For the random fly-aways, make sure that the state estimation is correct. The typical approach is to enable logging, log `stateEstimate.x`, `stateEstimate.y`, `stateEstimate.z`, run `hover_swarm.launch` (but do not attempt to fly), and move the Crazyflie manually in the space. If you use `rosecho /cf8/log1`, you should see the correct coordinates. Also keep in mind that there seem to be problems with the state estimator, see [this issue](https://github.com/bitcraze/crazyflie-firmware/issues/555).
- For the connection issue, please state what the error is when there is a failure. Perhaps you are running into another [firmware bug](https://github.com/USC-ACTLab/crazyswarm/issues/182)? 

Also, I hope both you and your brother are fine and Covid-free!

santiagorg2401, 2020-08-14:
Hi,
For the firmware updating I am using the cfclient GUI app, is that ok?
For the estimators, do I have to type them on the Python code in crazyswarm/ros_ws/src/crazyswarm/scripts ?
I will attach a screenshot (of the Rviz issue) when being in the laboratory.
Thank you for your hopes, and yes, we are covid free.

closed 2021-02-20 by jpreiss