USC-ACTLab/crazyswarm #238: Low link quality and high latency

swarm5, 2020-07-22:
I used a radio station and tested a crazyflie. There is a problem with my communication link and it says: "link quality low"&"latency too high!".
I want to know what is the reason for this problem? There are no interfering devices such as WiFi in my room.

swarm5, 2020-07-22:
[image]

jpreiss, 2020-07-22:
It is normal to see a few of those messages at startup. It's only a problem if the message is being continually printed on the console.

swarm5, 2020-07-22:
@jpreiss Is this a normal phenomenon when it is just started up? I tested figure8.py, the plane flew very unstable, and finally lost control and hit the wall

swarm5, 2020-07-22:
@whoenig @jpreiss I want to use crazyswarm to control multiple crazyflies, but when I only control one, the connection quality has been shown to be very low. How can I improve link quality?

swarm5, 2020-07-22:
I tried it again, and unfortunately, it still hits the wall. The following is the prompt message:
[image]

> [ INFO] [1595423517.376228426]: [cf1] Update parameters
> [ INFO] [1595423517.466289558]: [cf1] ESTIMATOR: Using Kalman (2) estimator
> [ INFO] [1595423517.511347072]: Update params: 0.152460 s
> [ INFO] [1595423517.511488289]: Started 1 threads
> [ WARN] [1595423517.558222733]: Latency too high! Is 0.046575 s.
> [ INFO] [1595423548.348406384]: [cf1] Upload trajectory
> [ INFO] [1595423548.462612558]: [cf1] Uploaded trajectory
> [ WARN] [1595423548.463068943]: Latency too high! Is 0.109697 s.
> [ INFO] [1595423548.468084909]: Takeoff!
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423550.076004642]: ICP did not converge! for object cf1
> [ WARN] [1595423550.076083188]: No updated pose for CF cf1 for 0.000786 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423550.081392513]: ICP did not converge! for object cf1
> [ WARN] [1595423550.081413635]: No updated pose for CF cf1 for 0.006301 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423550.091665297]: ICP did not converge! for object cf1
> [ WARN] [1595423550.091710257]: No updated pose for CF cf1 for 0.016503 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423550.101344318]: ICP did not converge! for object cf1
> [ WARN] [1595423550.101363242]: No updated pose for CF cf1 for 0.026231 s.
> [ INFO] [1595423550.994229568]: [cf1] GoTo
> [ INFO] [1595423553.506020874]: Start trajectory!
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423554.275744814]: ICP did not converge! for object cf1
> [ WARN] [1595423554.275843332]: No updated pose for CF cf1 for 0.000547 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.155229949]: ICP did not converge! for object cf1
> [ WARN] [1595423557.155264100]: No updated pose for CF cf1 for 0.000815 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.161124833]: ICP did not converge! for object cf1
> [ WARN] [1595423557.161162573]: No updated pose for CF cf1 for 0.006686 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.171338431]: ICP did not converge! for object cf1
> [ WARN] [1595423557.171383252]: No updated pose for CF cf1 for 0.016885 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.181077630]: ICP did not converge! for object cf1
> [ WARN] [1595423557.181110640]: No updated pose for CF cf1 for 0.026655 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.976796528]: ICP did not converge! for object cf1
> [ WARN] [1595423557.976832274]: No updated pose for CF cf1 for 0.002540 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423557.982692346]: ICP did not converge! for object cf1
> [ WARN] [1595423557.982712365]: No updated pose for CF cf1 for 0.008489 s.
> [ WARN] [1595423558.025034146]: Dynamic check failed for object cf1
> wroll: -35.2388 >= 20
> wyaw: -18.1076 >= 10
> roll: -1.80571 >= 1.4
> 
> [ WARN] [1595423558.025079834]: No updated pose for CF cf1 for 0.050743 s.
> [ WARN] [1595423558.025326861]: Dynamic check failed for object cf1
> wroll: -35.1711 >= 20
> wyaw: -18.0196 >= 10
> roll: -1.81369 >= 1.4
> 
> [ WARN] [1595423558.025358673]: No updated pose for CF cf1 for 0.051068 s.
> [ WARN] [1595423558.031088405]: Dynamic check failed for object cf1
> wroll: -31.1804 >= 20
> wyaw: -15.9723 >= 10
> roll: -1.78999 >= 1.4
> 
> [ WARN] [1595423558.031113716]: No updated pose for CF cf1 for 0.056844 s.
> [ WARN] [1595423558.041104054]: Dynamic check failed for object cf1
> wroll: -26.4622 >= 20
> wyaw: -13.5125 >= 10
> roll: -1.78623 >= 1.4
> 
[... 838 lines cut ...]
> 
> [ WARN] [1595423561.673443702]: No updated pose for CF cf1 for 0.080441 s.
> [ WARN] [1595423561.682835468]: Dynamic check failed for object cf1
> wroll: 28.5773 >= 20
> roll: 1.70627 >= 1.4
> fitness: 0.00122157 >= 0.001
> 
> [ WARN] [1595423561.682862979]: No updated pose for CF cf1 for 0.089860 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423561.692480228]: ICP did not converge! for object cf1
> [ WARN] [1595423561.692500313]: No updated pose for CF cf1 for 0.099560 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423561.702323926]: ICP did not converge! for object cf1
> [ WARN] [1595423561.702345068]: No updated pose for CF cf1 for 0.109415 s.
[... cut ...]
> [ WARN] [1595423561.722271688]: ICP did not converge! for object cf1
> [ WARN] [1595423561.722292463]: No updated pose for CF cf1 for 0.129363 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [ WARN] [1595423561.732540037]: ICP did not converge! for object cf1
> [ WARN] [1595423561.732569379]: No updated pose for CF cf1 for 0.139624 s.
> [pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
>

whoenig, 2020-07-22:
The connection warning means you have some interference or bad Crazyradio placement. If you can, disable 2.4 Ghz Wifi; however, in general this is not a problem and just a warning.

For your other issue it looks like your mocap has trouble tracking the 4 markers. If you don't want to fly very aggressive maneuvers, it is much more robust to use single-marker tracking (where you put a single, larger marker on top of each Crazyflie). Otherwise, make sure your mocap is calibrated, the markers are correctly placed and well visible by multiple cameras during the flight.

ucaszzw, 2020-07-22:
I still have a few questions that are not very clear. I would like to ask you:
1. I just finished calibrating my motion capture system. At the same time, I run` roslaunch crazyswarm mocap_helper.launch`. In this case, I put cf in the center of the field. At this time, the indicator light of cf is always green but sometimes it jumps to red. I ran the script command and cf gradually lost control and flew out of the field. I think the reason seems to lie in the communication link. However, I don’t know what these prompts mean:
> [Pcl::IterativeClosestPoint::computeTransformation] Not enough correspondences found. Relax your threshold parameters.
> [WARN] [1595423550.076004642]: ICP did not converge! for object cf1
> [WARN] [1595423550.076083188]: No updated pose for CF cf1 for 0.000786 s."
2. "however, in general this is not a problem and just a warning." Do you think that the communication link is not enough to cause the loss of control of my cf? The main reason is the motion capture system? If there is 2.4G interference, is there any way to change the radio, such as changing its channel?
3. "If you don't want to fly very aggressive maneuvers, it is much more robust to use single-marker tracking (where you put a single, larger marker on top of each Crazyflie). "Marker for single-point tracking Point, what is its minimum size? I hope to run a lot of cfs later, whether there will be occlusion problems in single-point tracking.

ucaszzw, 2020-07-22:
I am not in the laboratory now, I have a guess:
Will the problem be: I assigned cf to the wrong channel? How to check the radio channel?

whoenig, 2020-07-22:
1. When you run `mocap_helper.launch`, there is no communication with the CF, so the light changes you see are independent of the Crazyswarm. What you describe sounds like your broadcasts are not working properly. Did you update your Crazyradio firmware? Did you update your NRF51 firmware? Also, please [enable logging](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L48), and look at the `/cf1/log1` topic if you just manually move around your Crazyflie. It should update correctly.

2. In my experience you can still get good flight performance when having a link that has a warning around 0.3. You can change channels etc., but I don't think this warning is related to your fly-off issue.

3. Single-point tracking is more robust to occlusion problems: i) the marker on top is typically visible by more/all cameras; ii) the re-assignment after a marker became invisible is much more robust since it uses global assignment rather than greedy assignment at every timestep. The minimum size depends on your mocap: it should be at least 3 pixels in the furthest camera, but 5-10 pixels are generally better. We typically use 7 - 9mm markers, but you can go bigger too.

swarm5, 2020-07-23:
@whoenig I hope to upgrade the firmwares of Crazyflies with the provided firmwares again. I choose to Upload the precompiled firmware.
But the following situation occurred，How can I solve it?：
`$ rosrun crazyflie_tools flash --target nrf51 --filename prebuilt/cf2_nrf.bin`

> No matching USB Device with devid = 0 found! Do you have a Crazyradio dongle connected?

swarm5, 2020-07-23:
@whoenig I have a problem that I have been unable to understand:
Different channels need to be allocated for Crazyflies. The document states "For example, cf1 is assigned to channel 80, cf2 is assigned to channel 90, cf3 is assigned to channel 100, cf4 is assigned to channel 80 and so on."
My question is, how are these values determined for 80, 90, 100?

swarm5, 2020-07-23:
When I use cfclient to connect, the quality of the connection is very high.
But when I use crazyswarm to connect, there is always a problem with the quality of the link
[image]
[image]

whoenig, 2020-07-23:
Your Crazyswarm screenshot does not show any link quality issues. In general, you have to find good channels empirically; a tool is available [here](https://github.com/bitcraze/crazyflie-lib-python/blob/master/examples/radio-test.py). When you execute it, make sure to place the CF roughly where they would actually be flying.

jpreiss, 2020-07-24:
Are you running `cfclient` at the same time as crazyswarm, to see the link quality?

If so, then that is probably the source of the degraded link quality.

swarm5, 2020-07-29:
@jpreiss @whoenig I changed a computer and performed another channel test. I still have a few questions:
1. **When the frequency is 2M, how far is the communication distance?** I found a phenomenon, when I test the distance is 4m, there will be a phenomenon that the rssi average is nan. For details, please see:https://github.com/bitcraze/crazyflie-lib-python/issues/161. However, I took cf to the computer and tried many times, but there was no corresponding situation.
2. **Is the channel recommended by radio-test reasonable?** The figure below is the experimental result of my three tests. I don’t know if the understanding is correct. An excellent channel requires a stable ack rate and a high rssi.
[image]
[image]
[image]
The program recommends many peak points of ack rate, but the ack rate of the channels around it is 0. Judging from the picture, whether we choose the 82-100 section of the channel will be more secure?
3. **How to set up crazyswarm channel reasonably?** If my optimal channel is 61, 62, 63..., can I set it as the channel of my crazyswarm radio? Or in crazyswarm, each channel must have a large numerical difference?

whoenig, 2020-07-30:
1. We had no issues with 15m indoors.
2. When in doubt, use the channels with higher ack rates. For us usually 80-100 work best. From your plots, it looks like the test script has some bug (there should be no nans).
3. You need a difference of at least 2, when using the 2M mode, because the frequencies overlap slightly. E.g., 80, 82, 84 would be fine.

closed 2020-09-10 by swarm5