USC-ACTLab/crazyswarm #239: Access real-time quad position

pietropier1, 2020-07-25:
First of all, thank you for the great project. It was very easy to set the all thing up.

I'm in the process of modifying the existing "figure8" script in order to have the quads executing a custom controller. In particular, I need to have access to the real-time quads position (ideally the entire state, but for now R3 position vector is good enough). I'm able to print the content of `cf.state.pos `, but when trying to access its elements (either by `cf.state.pos[i]` or `cf.state.pos.__getitem__(i)`, with 0<i<3 ), the following error appears: 

`AttributeError: 'module' object has no attribute 'vindex'`

I tried to backtrack where the vindex method is defined for the class _cffirmare but I got stuck. Is there an alternative/better way to get the state of the quads in order to build a position feedback controller?

Thanks in advance for your help,

whoenig, 2020-07-25:
- If you need the entire state or do not use an external tracking system, you need to [enable logging](https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/launch/hover_swarm.launch#L48) and add another variable `enable_logging_pose` that you set to true there. Then, you can use a ROS subscriber to topic `/cf<id>/pose` to get the data (no Python API support yet). This will stream the data from your Crazyflie, so there is an additional latency of a few ms. Whenever you use logging, you can only control fewer CFs per radio (around 5, depending on your desired update rate).
- If you use a mocap, you should use `cf.position()`, which returns a numpy array, see https://github.com/USC-ACTLab/crazyswarm/blob/master/ros_ws/src/crazyswarm/scripts/pycrazyswarm/crazyflie.py#L259-L273. Depending on how many markers you use, this could be extended to give you orientation as well.

jpreiss, 2020-07-25:
This is a bug. I implemented `__getitem__` against a newer firmware version and ported it back to master, where our `crazyflie-firmware` submodule still points to an old version that doesn't have `vindex`.

However, you should use `cf.position()` as @whoenig suggests. `cf.state` only exists in simulation. Any script that reads `cf.state` will not work on real hardware. We should probably name it `cf._state`.

pietropier1, 2020-08-07:
Using cf.position() did work. Thank you for your help.

whoenig, 2020-08-07:
Thanks for the update. Closing.

closed 2020-08-07 by whoenig