USC-ACTLab/crazyswarm #240: hover_swarm failing with multiple CFs (crtp assert failing)

PartiallyObservable, 2020-08-04:
Hi, we are having some issues with reliably getting hover_swarm to launch with more than 1-2 crazyflies. This is unexpected as we have previously been able to fly several (several months/a year ago). 

What are we trying to do:

- Install and test crazyswarm and VICON using swap6v script (6 crazyflies doing the hexagon swap routine) 

Problem we are experiencing:

- hover_swarm.launch will throw a std::runtime_error when adding multiple crazyflies. We cannot get hover_swarm.launch to reliably launch with more than 3 CFs, and it will sometimes fail even with 1CF. 
- When it fails, the CF (or several CFs) the parameter update fails on also seems to go into a panic mode (solid red light) and needs to be restarted manually. 
- We have been able to successfully fly 3 CFs and they appear stable and have a good connection. Note that chooser.py seems to have no issues communicating to CFs (battery check and reboot seem to be fine).

Information about our installation:

- Ubuntu 18.04, ROS melodic, installed using instructions on readthedocs. 
- Crazyswarm version is current as of last week. 
- We have flashed the crazyradio with the firmware in crazyradio-firmware
- We have flashed the crazyflies using the python tool that displays the upload progress. We have tried the firmwares in prebuilt, and the firmwares that are built with the build.sh script (in crazyflie-firmware / crazyflie2-nrf-firmware). The python tool does not report any errors, and the crazyflies seem to accept the firmware update. 
- We have seen this issue using both a single radio, and using multiple radios. 

hover_swarm.launch output on a failed start:

```
liam@holmes:~/drones/crazyswarm/ros_ws/src/crazyswarm/scripts$ roslaunch crazyswarm hover_swarm.launch 
... logging to /home/liam/.ros/log/e2e2e8ca-d69b-11ea-b975-7cdd90b11476/roslaunch-holmes-31682.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://holmes:36449/

SUMMARY
========

PARAMETERS
 * /crazyflieTypes/CF21SingleMarker/batteryVoltageWarning: 3.8
 * /crazyflieTypes/CF21SingleMarker/batteryVoltateCritical: 3.7
 * /crazyflieTypes/CF21SingleMarker/bigQuad: False
 * /crazyflieTypes/CF21SingleMarker/dynamicsConfiguration: 0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kR_z: 60000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_omega_rp: 200
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_xy: 0.2
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kd_z: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_m_z: 500
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_xy: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/ki_z: 0.05
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_xy: 0.4
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kp_z: 1.25
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_xy: 20000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/kw_z: 12000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/mass: 0.032
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/ctrlMel/massThrust: 132000
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_rollpitch: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/mNGyro_yaw: 0.1
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_xy: 1.0
 * /crazyflieTypes/CF21SingleMarker/firmwareParams/kalman/pNAcc_z: 2.0
 * /crazyflieTypes/CF21SingleMarker/markerConfiguration: 3
 * /crazyflieTypes/default/batteryVoltageWarning: 3.8
 * /crazyflieTypes/default/batteryVoltateCritical: 3.7
 * /crazyflieTypes/default/bigQuad: False
 * /crazyflieTypes/default/dynamicsConfiguration: 0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_m_z: 1500
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_xy: 2.0
 * /crazyflieTypes/default/firmwareParams/ctrlMel/i_range_z: 0.4
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_xy: 70000
 * /crazyflieTypes/default/firmwareParams/ctrlMel/kR_z: 60000
[... 234 lines cut ...]
[ INFO] [1596577563.440505771]: [cf17] LPS25H I2C connection [OK].
[ INFO] [1596577563.443228260]: [cf17] ESTIMATOR: Using Complementary (1) estimator
[ INFO] [1596577563.445602042]: [cf17] CONTROLLER: Using PID (1) controller
[ INFO] [1596577563.448072601]: [cf17] MTR-DRV: Using brushed motor driver
[ INFO] [1596577563.937740584]: [cf17] EEPROM: I2C connection [OK].
[ INFO] [1596577564.173272935]: [cf17] AK8963: Self test [OK].
[ WARN] [1596577564.792814766]: [cf17] Link Quality low (0.700000)
[ INFO] [1596577564.973907218]: [cf17] SYS: The system resumed after watchdog timeout [WARNING]
[ INFO] [1596577564.976724732]: [cf17] SYS: Assert failed at src/modules/src/crtp.c:186
^C[rviz-4] killing on exit
[crazyswarm_teleop-3] killing on exit
[crazyswarm_server-2] killing on exit
terminate called after throwing an instance of 'std::runtime_error'
  what():  timeout
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

I attempted to look at "src/modules/src/crtp.c:186" where the assert fails according to the [INFO] output. It appears to be a FreeRTOS function (186: if (uxQueueMessagesWaiting(queues[p.port]) > 0) ), but I wasn't sure what I should do with that information. 

Thank you for any help

PartiallyObservable, 2020-08-05:
We just found this issue https://github.com/USC-ACTLab/crazyswarm/issues/182 and attempted that solution. It appears to have "fixed" it? I will close this issue if this continues to be a good fix (only had time to test it a few times).

whoenig, 2020-08-05:
This is still a "good" fix that I am using in practice as well (I recently tested with 16 CFs). There is no issue with correctness, as long as you use the crazyswarm, and likely no issue with the official Python library. Getting a proper fix in the official firmware will take some time (but we are working with Bitcraze on that).

Other than that, I usually use the latest official firmware STM32/NRF51 these days, especially if you hit issues regarding a firmware reboot which might be caused by firmware bugs that are fixed in the latest version.

Flashing with chooser.py or a JTAG debugger is the preferred method, since the official GUI has (surprisingly) no verification step, and I have had issues that flashing was shown as success, but some pages were written improperly in the flash. Chooser.py uses the crazyflie_cpp library under the hood, which always verifies the flashed firmware. If you flash using a JTAG debugger, verification is executed also.

PartiallyObservable, 2020-08-05:
I see, I think this will be our working solution then. Thank you for your confirmation and input about firmware flashing. We will use chooser.py in the future to make sure we don't have sneaky firmware issues come up.

closed 2020-08-05 by PartiallyObservable