USC-ACTLab/crazyswarm #243: [crazyswarm_server-2] process has died

864419409, 2020-08-27:
When I run the following command
`zzw@zzw-ubuntu:~/crazyswarm0525/crazyswarm $ roslaunch crazyswarm hover_swarm.launch`
The following error occurred。Could you tell me how to deal with this？

> [crazyswarm_server-2] process has died [pid 9633, exit code -6, cmd /home/zzw/crazyswarm0525/crazyswarm/ros_ws/devel/lib/crazyswarm/crazyswarm_server __name:=crazyswarm_server __log:=/home/zzw/.ros/log/48a3edc4-e81e-11ea-92d8-200db010542d/crazyswarm_server-2.log].
> log file: /home/zzw/.ros/log/48a3edc4-e81e-11ea-92d8-200db010542d/crazyswarm_server-2*.log

whoenig, 2020-08-28:
This usually means that there was a segmentation fault, most frequently caused by a misconfiguration. Perhaps https://github.com/USC-ACTLab/crazyswarm/issues/147 will help?

864419409, 2020-09-04:
感谢您，问题已经解决了

closed 2020-09-04 by 864419409